thesofproject/sof #3340: [BUG][BSW] unable to install params when aplay on PCM0 in S24_LE/S32_LE

fredoh9, 2020-08-24:
**Describe the bug**
aplay -Dhw:0,0 -r 48000 -c 2 -f S32_LE -d 10 /dev/zero

`aplay: set_params:1407: Unable to install hw params:`

https://sof-ci.01.org/sofpr/PR3324/build6974/devicetest/

from etrace, looks buffer size is the problem.

```
[306878054.218750] ( 67879896.000000) c0 mixer        1.2          src/audio/mixer.c:202  ERROR mixer_params(): sink buffer size is insufficient
[306878063.489583] (        9.270833) c0 pipe         3.21      src/audio/pipeline.c:491  ERROR pipeline_params(): ret = -12, host->comp.id = 17
[306878074.843750] (       11.354167) c0 ipc                       src/ipc/handler.c:274  ERROR ipc: pipe 3 comp 17 param
```

**To Reproduce**
aplay -Dhw:0,0 -r 48000 -c 2 -f S32_LE -d 10 /dev/zero
aplay -Dhw:0,0 -r 48000 -c 2 -f S24_LE -d 10 /dev/zero 

**Reproduction Rate**
100%

**Expected behavior**
aplay should play and return no error

aiChaoSONG, 2020-08-25:
This is a general issue on every platform.  Some pipelines are created with S16_LE, but the template we use just hard code the PCM capabilities. we may need a discussion.

plbossart, 2021-06-10:
@fredoh9 @aiChaoSONG does this problem still exist? I think it was a duplicate of other problems with BYT/CHT mixers. please close if this isn't a problem any longer.

aiChaoSONG, 2021-06-11:
fixed by https://github.com/thesofproject/sof/commit/e4fa878, close

closed 2021-06-11 by aiChaoSONG