thesofproject/sof #3344: GLK: sof-logger issue?

plbossart, 2020-08-25:
tracking https://sof-ci.01.org/linuxpr/PR2378/build4363/devicetest/

GLK tests show this for sof-logger:
"invalid number of parameters"?

This may be an SOF, sof-test or kernel issues, we'll transfer the issue as needed.

@fredoh9 we'll need your help on this one.

fredoh9, 2020-08-25:
I'm trying to reproduce on my GLK bobba, but this is not 100% reproducible. In this test, it was `check-runtime-pm-status.sh`.

Looking at around logger/convert.c,
TRACE_MAX_PARAMS_COUNT is 4. If it return error there, I think entry.header.params_num might be corrupted or garbaged already.
```
        /* fetching entry params from dma dump */
        if (entry.header.params_num > TRACE_MAX_PARAMS_COUNT) {
                log_err(config->out_fd,
                        "Invalid number of parameters.\n");
                ret = -EINVAL;
                goto out;
        }
```

paulstelian97, 2020-08-31:
I think the parameters number should be limited in firmware at compile time, so sounds like the ldc file is either corrupt or misaligned for some reason (not from the same build).

plbossart, 2021-06-10:
@marc-hb @fredoh9 @lgirdwood I am not sure if this bug is current, I haven't noticed this error in a very long time. Close?

marc-hb, 2021-06-10:
The DMA can sometimes stop in the middle of a log statement which can explain this.

I made a number of logger fixes recently, so I recommend closing and re-opening with more details if this is seen again. Commits 96f6a7b385bd, 1a79de1c1669, d54a9281f884, 1a79de1c1669 should provide better error messages. See also 73378e269f78.

More changes:
https://github.com/thesofproject/sof/commits/73df64444b7528/tools/logger
https://github.com/thesofproject/sof/commits/73df64444b7528/src/trace

marc-hb, 2021-06-10:
> And I notice one more thing. slogger's TIMESTAMP is not linearly increase. There are abrupt jump in back and force. Is this expected?

I have added a sequence number in my workspace and it shows that two consecutive log statements are sometimes swapped.

plbossart, 2021-06-10:
ok let's close and file a new issue if applicable. Thanks @marc-hb

closed 2021-06-10 by plbossart