thesofproject/sof #3345: [Regression] No sound on kernel 5.8 series

Valerio1474, 2020-08-25:
**Describe the bug**
Hello,

I don't know what might be changed from kernel 5.7.x to kernel 5.8.x but the sound has stopped working under the newest kernel. I thought that the 1.5.1 firmware could fix the problem, but to no avail.

**To Reproduce**
Compile and boot any 5.8.x kernel with same options I have on 5.7.x

**Reproduction Rate**
Always

**Expected behavior**
The expected behavior is to seamlessly install and use a new kernel.

**Impact**
No sound at all.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.8.x
    * SOF: 1.5.1
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: Debian/Sid/amd64/Cannon Lake
[5.7.0_dmesg.txt](https://github.com/thesofproject/sof/files/5122491/5.7.0_dmesg.txt)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

Valerio1474, 2020-08-25:
[5.8.3_dmesg.txt](https://github.com/thesofproject/sof/files/5122508/5.8.3_dmesg.txt)
Here the dmesg from the newer 5.8 kernel

lgirdwood, 2020-08-25:
@kv2019i @plbossart fyi

plbossart, 2020-08-25:
Looks like an HDMI issue to me:

````
[    2.243044] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: error at snd_soc_card_add_dai_link on sof-hda-dsp: -22
````

@Valerio1474 I know it's painful but do you have a means to bisect between the old and new kernel?

Valerio1474, 2020-08-25:
> Looks like an HDMI issue to me:
> 
> ```
> [    2.243044] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: error at snd_soc_card_add_dai_link on sof-hda-dsp: -22
> ```
> 
> @Valerio1474 I know it's painful but do you have a means to bisect between the old and new kernel?

Hi @plbossart, in the past I've tried to bisect the kernel to help to solve a couple of bugs: sometime I've been successful and sometime I've been not. Before that I start doing something I'm not so comfortable with, can you exclude that I'm missing a firmware file, a symbolic link, a configuration file (those things I've set up for my previous bug [https://github.com/thesofproject/linux/issues/2133](url)? If I should keep going the bisect way, do you have specific instructions for me?

kv2019i, 2020-08-25:
@Valerio1474 Hmm, I think it's caused by this regression:

https://github.com/thesofproject/linux/issues/2261

Fixed in upstream with
"ASoC: intel/skl/hda - fix probe regression on systems without i915"
https://lore.kernel.org/r/20200714132804.3638221-1-kai.vehmanen@linux.intel.com

The fix is in 5.9-rc2, but not yet backported to any 5.8 kernel.

Valerio1474, 2020-08-25:
Hi @kv2019i, if you are guessing right it would be a lot of spared time, so I'm compiling the 5.9-rc2 kernel

Valerio1474, 2020-08-25:
Hi @plbossart @kv2019i, I was finally able to compile and boot the 5.9-rc2 kernel and I can test the soundcard with:
> speaker-test -c 2
and hear the white noise coming from my speakers.

Should I wait for the backport of the relative fix to 5.8 kernel before closing this bug?

Valerio1474, 2020-08-25:
Hi @plbossart @kv2019i, I was finally able to compile and boot the 5.9-rc2 kernel and I can test the soundcard with:
> speaker-test -c 2
and hear the white noise from my speakers.

Should I wait for the backport of the relative fix to 5.8 kernel before closing this bug? 
[image]

plbossart, 2020-08-25:
@Valerio1474 thanks for testing, please keep this open until you see this sorted out in -stable.