thesofproject/sof #3359: [BUG][CML][CI] ERROR FW ABI 0x3011000 DBG ABI 0x5001000

aiChaoSONG, 2020-08-28:
**Describe the bug**
On CML Helios,  `[ 6357.443608] sof-audio-pci 0000:00:1f.3: status = 0x00000000 panic = 0x00000000` is observerd in dmesg,
`ERROR FW ABI 0x3011000 DBG ABI 0x5001000` is observed in sof-logger error trace when run `check-runtime-pm-status.sh`

**Impact**
No negative impact, only noisy error message.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: topic/sof-dev {33fd4f99}
    * SOF: master {78202631}
2) Name of the topology file
    * Topology: {sof-cml-rt1011-rt5682.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {CML Helios with Ubuntu 20.04}

mengdonglin, 2020-08-31:
@aiChaoSONG Can you please share more dmesg and clarify why kernel thinks this is is an error "ERROR FW ABI 0x3011000 DBG ABI 0x5001000"?

aiChaoSONG, 2020-08-31:
close for invalid.

closed 2020-08-31 by aiChaoSONG