thesofproject/sof #3371: [BUG][JSL] KD: Unable to create KD topology with 24/32 sample width.

zrombel, 2020-09-01:
**Describe the bug**
On JSL RVP, when trying to build KD topology with audio format 16kHz 24/32 2ch, and sending StreamPmcParams IPC to host component, FW replies with ERROR -22. In TraceLog Error can be found that states: 
`ERROR kpb: host_buffer_size (537600) must be at least 768000`
After changing buffer size to indicated value FW still replies with ERROR -22 to StreamPmcParams IPC but TraceLog shows different error: 
`ERROR 9 blocks needed for allocation but only 0 blocks are remaining`

Bug introduced with PR:
#3186

**Topology**
```
        +----------+   +------------------+
        |    OS    |-->|   AUDIO DEVICE   |-----------------------+
        +----------+   +------------------+                       |
                                                                  |
                   +-------------------------+                    |
        +--------+ | +---+   +-----+   +---+ |                    |
        |KD Dummy|<->|Buf|<->|ChSel|<->|Buf|<->+                  |
        +--------+ | +---+   +-----+   +---+ | |  pipe_cap        |
                   +-------------------------+ |                  |
                                               |                  |
                             +-----------------+                  |
        +--------------------|----------------------------------+ |
        |+----+  +------+  +---+  +---+  +---+  +---+  +------+ | |
        ||Host|<-|Buffer|<-|Kpb|<-|Buf|<-|Vol|<-|Buf|<-| DMIC |---+
        |+----+  +------+  +---+  +---+  +---+  +---+  +------+ |
        +-------------------------------------------------------+
```

**To Reproduce**
Run python tests with Slim Driver on JSL RVP: 
07_04_TestKdDmic16000Hz24b32b2ch

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash
    * SOF: branch: master, hash: 35d9034090f443b21c3d493d6ad16e6d97e38560
    * Python Tests: branch: master, hash: a3e2b33c32
2) Name of the platform(s) on which the bug is observed.
    * JSL RVP

**LOGS**
[JSL KD ERROR - original buffer size.zip](https://github.com/thesofproject/sof/files/5155902/JSL.KD.ERROR.-.original.buffer.size.zip)
[JSL KD ERROR - bigger buffer size.zip](https://github.com/thesofproject/sof/files/5155903/JSL.KD.ERROR.-.bigger.buffer.size.zip)

slawblauciak, 2020-09-04:
@zrombel as per earlier discussion, the host must comply with the new history buffer requirement.
The host buffer needs to be twice the size the history buffer.

slawblauciak, 2020-09-04:
I was able to reproduce this issue when changing the host buffer size myself. Looking into this right now.

slawblauciak, 2020-09-10:
@zrombel please try this https://github.com/thesofproject/sof/pull/3399

slawblauciak, 2020-09-15:
Fix confirmed effective.
Another problem has surfaced however, should probably have a separate issue for it.

zrombel, 2020-09-15:
I will open an issue for newly found bug. This issue can be closed once #3399 is merged to master.

lgirdwood, 2020-09-17:
@zrombel can we close ? #3399 now merged.

zrombel, 2020-09-17:
Of course. Closing issue.

closed 2020-09-17 by zrombel