thesofproject/sof #3378: [FEATURE] Processing pipeline macros with volume need to get their PGA name from caller macro

singalsu, 2020-09-02:
**Is your feature request related to a problem? Please describe.**
Yes, currently the general looking macros like pipe-eq-iir-volume.m4 are in practice usable only with Dmic0 endpoint. We need the same macros for HDA and SDW endpoints to avoid duplication. The old style PGAx.y like names cannot be controlled by UCM in a maintainable way. There's same issue with with playback pipeline macros.

**Describe the solution you'd like**
Minimum set of processing pipeline macros to maintain, no duplication need with just different PGA_NAME defines. The higher level macros  like sof-hda-generic.m4 needs to define PGA_NAME before instantiating echo pipeline a undefine PGA_NAME after it. There might be other ways but this simple way should work.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

lgirdwood, 2020-09-07:
@singalsu so IIUC all we need here is to pass in kcontrol names from the callers ? @juimonen can you take a quick look here and report on effort. Thanks !

lgirdwood, 2021-01-12:
@juimonen @singalsu any update here ? can we close ?

lgirdwood, 2021-01-25:
@singalsu lets review for v1.8 as topology 2.0 means we probably don't need this.

singalsu, 2021-01-25:
Probably no need, though I wonder topology 2.0 is going to handle all of what m4 does (e.g. pass correct filter blobs for processing pipelines, define processing for endpoints). For now need to do all pipeline processing changes from top level CMakeList.txt with a lot of care since pipelines are not as symmetrical as they initially look like. The PGAs names is one of the hard-coded pitfalls.

lgirdwood, 2021-02-05:
@juimonen @ranj063 fyi - any comment here for topology 2.0

singalsu, 2021-02-12:
It would not hurt to fix this for 1.x since it's broken. As example when I did for my PR #3841 the example topology that uses various -DxPROCs I could use sof-pipe-eq-iir-volume only for DMIC capture. The topology graph gets messed up if I attempt to use it for headset microphone capture (where it also makes sense).

I'll do a fix to intel-generic-dmic.

singalsu, 2021-02-12:
The problems looks like this:

[image]

lgirdwood, 2021-02-12:
@singalsu can we do some temporary duplication here (and state this in big comments at the top of each file) - I dont wont to waste any extra effort on stuff that is deprecated. @plbossart @ranj063 any comments ?

singalsu, 2021-02-12:
@lgirdwood I made the fix PR #3844, it's quite simple. With that included the pipelines that contained hard coded PGA names can be used more generally.

ranj063, 2021-02-12:
> @juimonen @ranj063 fyi - any comment here for topology 2.0

@lgirdwood @singalsu please have a look at https://github.com/ranj063/sof/blob/topic/tplg2/tools/topology2/. I've updated the branch to add the HDA generic topology and it should give you a decent idea of how naming works there. But in general, a volume component in a pipeline will have a unique name defined by the pipeine ID and volume comp index. So there should be no clashes at the top-level. 

As for controls and data associated with them, https://github.com/ranj063/sof/blob/topic/tplg2/tools/topology2/include/pipelines/pipeline-highpass-capture.conf should give you an idea. Its easy to see what controls components are associated with and what data they have

singalsu, 2021-02-12:
Off-topic for this but thanks for examples! I think there's in 2.0 example some cleanup work that has been done for m4 topologies missing. Though I guess I named the first version of processing pipeline as high-pass capture but later I've started prefer names like pipe-eq-iir-volume-capture where we know exactly what components are instantiated.  Also it's not sensible to duplicate the .conf for every filter coefficients variation. It's better to pass the include files names for processing configuration blobs. We cam create a new suitable syntax. It's easy at least for my developed algos since everything is generated without hand-edits. The m4 system now allows to create topology variants for endpoint processing pipelines and alter the the configuration blobs with minimal amount of duplication.

There's still weaknesses in m4 processing macros like some pipelines with volume define the mute switch and some not. It's not really visible from the macro name. Also I don't like the duplication of similar pipelines for different sample rates. And I still keep finding old bugs from there because flat m4 is tricky :(

ranj063, 2021-02-12:
Thanks for the feedback @singalsu. Agree that the filter-coefficients needs to be included from the top-level. I will change this. Please do take a look at the other classes and let me know if you have any suggestions for improvements.

juimonen, 2021-02-15:
@ranj063 @singalsu in topology2 we need to be able to give explicit names to the components&controls. So we need to separate class names from instance names. This was one thing making topology1 messy -> for example some component and controls names are "locked" in UCM -> we can't compose component names from pipeline, pcm id's etc. as they might change depending on the topology/platform (or how you combine existing pipelines).

PR #3844 (2021-02-26): Topology: Get PGA_NAME from pipeline caller macro
This patch fixes the error that happens with the hard-coded PGA
names Dmic0 and Dmic1 in pipelines pipe-eq-iir-volume-capture.m4
and pipe-eq-iir-volume-capture-16khz.m4. Pipelines for beamformer
TDFB have the same hard-coded PGA names and are fixed too.

If impacted pipelines are used for any other purpose than DMIC endpoints
the topology graph gets messed up. With this change the caller macro,
e.g. intel-generic-dmic.m4 needs to set the PGA_NAME macro when the
respective pipeline is instantiated. If it is not set the default
name via N_PGA() m [... cut]
commit 8bb8787c (2021-02-26): Topology: Get PGA_NAME from pipeline caller macro

This patch fixes the error that happens with the hard-coded PGA
names Dmic0 and Dmic1 in pipelines pipe-eq-iir-volume-capture.m4
and pipe-eq-iir-volume-capture-16khz.m4. Pipelines for beamformer
TDFB have the same hard-coded PGA names and are fixed too.

If impacted pipelines are used for any other purpose than DMIC endpoints
the topology graph gets messed up. With this change the caller macro,
e.g. intel-generic-dmic.m4 needs to set the PGA_NAME macro when the
respective pipeline is instantiated. If it is not set the default
name via N_PGA()  [... cut]
closed 2021-02-26 by lgirdwood