thesofproject/sof #3387: [BUG] firmware boot failed on TGL device

plbossart, 2020-09-03:
**Describe the bug**
firmware authentication/boot fails with TGL 1.5.1 firmware signed with the production key.
This is an HDaudio platform with no dmic.
The HDaudio legacy works fine.

**To Reproduce**
install latest SOF driver and boot
use 1.5.1 firmware from sof-bin/lib/firmware/intel/sof/v1.5.1/intel-signed/sof-tgl-v1.5.1-58b36ed.ri

**Reproduction Rate**
100%

**Expected behavior**
firmware released by Intel boots

**Impact**
no SOF support

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: a34de7786e1e21a395385ba25d6975eedde14e5c
    * SOF:  sof-bin/lib/firmware/intel/sof/v1.5.1/intel-signed/sof-tgl-v1.5.1-58b36ed.ri
2) Name of the topology file
    * Topology: irrelevant, firmware does not boot.
3) Name of the platform(s) on which the bug is observed.
    * Platform: <ask me privately>

**Screenshots or console output**

````
[    3.419454] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.420680] fbcon: i915drmfb (fb0) is primary device
[    4.568084] Console: switching to colour frame buffer device 240x67
[    4.590463] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[    4.611562] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0xc00
[    4.611565] sof-audio-pci 0000:00:1f.3: found ML capability at 0xc00
[    4.611566] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x800
[    4.611570] sof-audio-pci 0000:00:1f.3: found DSP capability at 0x800
[    4.611570] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x500
[    4.611575] sof-audio-pci 0000:00:1f.3: found GTS capability at 0x500
[    4.611576] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x1f00
[    4.611580] sof-audio-pci 0000:00:1f.3: found DRSM capability at 0x1f00
[    4.611581] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x700
[    4.611585] sof-audio-pci 0000:00:1f.3: found SPIB capability at 0x700
[    4.611610] sof-audio-pci 0000:00:1f.3: DMA mask is 64 bit
[    4.611616] sof-audio-pci 0000:00:1f.3: hda global caps = 0x1009701
[    4.611617] sof-audio-pci 0000:00:1f.3: detected 9 playback and 7 capture streams
[    4.611696] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    4.611696] sof-audio-pci 0000:00:1f.3: using IPC IRQ 164
[    4.611756] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[    4.611757] sof-audio-pci 0000:00:1f.3: PP capability, will probe DSP later.
[    4.616346] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000040 at  00000048
[    4.616350] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[    4.620646] sof-audio-pci 0000:00:1f.3: codec_mask = 0x5
[    4.620660] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[    4.620662] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000040 at  00000048
[    4.622171] sof-audio-pci 0000:00:1f.3: skipping SoundWire, no links enabled
[    4.622266] sof-audio-pci 0000:00:1f.3: HDA codec #0 probed OK: response: 10ec0236
[    4.623022] hdaudio ehdaudio0D0: loading codec module: hdaudio:v10EC0236r00100002a01
[    4.627720] sof-audio-pci 0000:00:1f.3: HDA codec #2 probed OK: response: 80862812
[    4.628257] hdaudio ehdaudio0D2: loading codec module: hdaudio:v80862812r00100000a01
[    4.631095] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    4.631096] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    4.631099] sof-audio-pci 0000:00:1f.3: DMICs detected in NHLT tables: 0
[    4.631105] sof-audio-pci 0000:00:1f.3: loading firmware
[    4.632336] sof-audio-pci 0000:00:1f.3: request_firmware intel/sof/sof-tgl.ri successful
[    4.632337] sof-audio-pci 0000:00:1f.3: firmware doesn't contain extended manifest
[    4.632345] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000002 at  00000048
[    4.632350] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[    4.632356] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[    4.632360] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000004 at  00000044
[    4.632360] sof-audio-pci 0000:00:1f.3: booting DSP firmware
[    4.632394] sof-audio-pci 0000:00:1f.3: period_bytes:0x0
[    4.632394] sof-audio-pci 0000:00:1f.3: periods:1
[    4.632438] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[    4.632454] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[    4.632455] sof-audio-pci 0000:00:1f.3: period_bytes:0x0
[    4.632456] sof-audio-pci 0000:00:1f.3: periods:1
[    4.632482] sof-audio-pci 0000:00:1f.3: Attempting iteration 0 of Core En/ROM load...
[    4.632485] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf010f0f successful
[    4.632496] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf010f0e successful
[    4.632498] sof-audio-pci 0000:00:1f.3: unstall/run core: core_mask = 1
[    4.632500] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[    4.633472] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000000 successful
[    4.633476] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf010e0e successful
[    4.710065] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5000001 successful
[    7.710564] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000012 timedout
[    7.710567] sof-audio-pci 0000:00:1f.3: error: cl_copy_fw: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[    7.711091] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x140000 successful
[    7.711095] sof-audio-pci 0000:00:1f.3: error: load fw failed ret: -110
[    7.711118] sof-audio-pci 0000:00:1f.3: unknown ROM status value 80000012
[    7.711122] sof-audio-pci 0000:00:1f.3: status = 0x0000002c panic = 0x00000000
[    7.711137] sof-audio-pci 0000:00:1f.3: extended rom status:  0x80000012 0x2c 0x0 0x0 0x0 0x0 0x25101c2 0x0
[    7.711139] sof-audio-pci 0000:00:1f.3: unknown ROM status value 80000012
[    7.711154] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[    7.711156] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110
[    7.761558] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1d003c timedout
[    7.761561] sof-audio-pci 0000:00:1f.3: error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read
[    7.761564] sof-audio-pci 0000:00:1f.3: error: dsp core reset failed: core_mask 1
[    7.761792] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110
````

keqiaozhang, 2020-09-04:
@plbossart , please try the debug key signed firmware for your TGL platform.
http://sof-ci-storage.sh.intel.com/daily/20200903/sof-7650398d/fw/debug/

kv2019i, 2020-09-04:
@plbossart Have the same. It appears these devices don't have any key for the DSP:
[    7.711122] sof-audio-pci 0000:00:1f.3: status = 0x0000002c panic = 0x00000000
 
0x2c -> 44 -> ADSP_CSE_VALIDATION_FAILED   ( https://thesofproject.github.io/latest/architectures/dsp/intel/cavs-boot/apollolake/apl-boot-rom.html?highlight=rom )

mengdonglin, 2020-09-08:
@plbossart @kv2019i We got confirmation that this TGL platform doesn't have DMIC to PCH and SOF support is not required. Can we close this issue as won't fix?

slawblauciak, 2020-09-08:
This platform is not supported, legacy audio driver should be used.

closed 2020-09-08 by slawblauciak

plbossart, 2020-09-08:
I think it's a real bug in the platform configuration.
The platform indicates the DSP is supported, but the keys are wrong.
Either the DSP should NOT be indicated as supported, or the keys should be set correctly.