thesofproject/sof #3388: [FEATURE] Add fuzzer to oss-fuzz

cujomalainey, 2020-09-03:
**Is your feature request related to a problem? Please describe.**
Have fuzzing be an automated task so no one needs to do anything other than fix the bugs

**Describe the solution you'd like**
get project approved then integrate related infra

**Describe alternatives you've considered**
manual fuzzing is tedious

**Additional context**
Fuzzing found a fairly large number of bugs in a short period of time and has prove to be worth the time to integrate.

cujomalainey, 2020-09-03:
@ranj063 can you give me a google account email that I can use to add you.
@xiulipan if we get approved we will need to start having travis uploading builds to a bucket and will probably need your help

xiulipan, 2020-09-04:
@cujomalainey Do you mean we will have a google cloud server or bucket to store build binary?

cujomalainey, 2020-09-04:
That is the idea from what I'm reading in the docs. CI would build what they need then they would run the fuzzers for us.

cujomalainey, 2020-09-15:
@ranj063 friendly ping so I can send the PR

ranj063, 2020-09-17:
@cujomalainey  sorry for the delay. here it is

ranjani.sridharan@intel.corp-partner.google.com

cujomalainey, 2020-09-21:
@xiulipan I was reading old documentation, CI is not needed

cujomalainey, 2020-09-22:
Project has been accepted into OSS-fuzz. I am working on a trimmed docker file (removing all the xtensa stuff) and a build script for fuzzing

cujomalainey, 2020-09-23:
So got it to a buildable stage, but this is going to be a challenge. oss-fuzz builds with clang and all our infra is built around gcc. Will need to make some changes.

cujomalainey, 2020-09-23:
So, buried deep within the closed issues of oss-fuzz is what I feared. It does not support the default method of fuzzing we are doing. It only supports persistent fuzzing using shims. To continue using the current fuzzing method we will have to do our own ad hoc system. As for oss-fuzz I am working to write a shim that it can work with.

cujomalainey, 2020-10-28:
oss-fuzz is going well, patches posted for all fuzzer bugs, few infra bugs to post and seed corpus is still missing. Once those are complete we can begin automated fuzzing.

https://github.com/thesofproject/sof/pull/3562
https://github.com/thesofproject/sof/pull/3561
https://github.com/thesofproject/sof/pull/3559
https://github.com/thesofproject/sof/pull/3558

cujomalainey, 2020-12-28:
this is done, but fuzzer scope is limited. Need cmake changes to expand it. mainly #3617

closed 2020-12-28 by cujomalainey