thesofproject/sof #3395: [BUG] FW boot fails occasionally on ICL

kv2019i, 2020-09-08:
**Describe the bug**
The ICL FW boot fails occasionally. Example case here:
https://sof-ci.01.org/linuxpr/PR2413/build4423/devicetest/

The driver has had retry logic since the initial upstream version and there has been multiple patches to adjust the trace output so that error-level traces are emitted only if all FW boot attempts fail.

On ICL the first-boot failure seems to happen with fairly high reproduction rate. This could be a symptom of gaps in implementation of ICCMAX flow -> https://github.com/thesofproject/linux/issues/2310 and https://github.com/thesofproject/sof/issues/1739 , but this has not been confirmed.

**To Reproduce**
Triggered by standard SOF CI test set

**Reproduction Rate**
Seen regularly in CI.

**Expected behavior**
First boot should be successful.

**Impact**
No end-user impact so thus P3. 

**Environment**
Not specific to any recent version, but for example triggers on:
Linux: 7e7294e230bb .travis.yml: fixup options, cores and clang support, bump to focal
SOF: 35d90340 build-tools.sh: fix misnamed locals DO_BUILD_topology and

ICL HDA configuration with generic topology.

**Screenshots or console output**
https://sof-ci.01.org/linuxpr/PR2413/build4423/devicetest/

FYI @marc-hb @xiulipan @ranj063 @fredoh9 @plbossart

marc-hb, 2020-09-09:
Just found https://github.com/thesofproject/sof-test/issues/340 which seems like a (partial?) duplicate, @fredoh9

kv2019i, 2020-09-09:
Did a stress test on one ICL RVP with a loop putting system to suspend, resuming, booting the DSP, and then repeating the sequence. 

Bug occurence rate: 0.6% (out of 13000 iterations)

With no system suspend inbetween, I did not see boots fail.

Test script:
```
while true; do rtcwake -m mem -s 5 ; sleep 5 ; done
```

marc-hb, 2020-09-09:
Thanks to @kv2019i 's indications here are some relevant pointers in the kernel source. As of version 5.8, the retry logic is located in `hda_dsp_cl_boot_firmware()` in `sound/soc/sof/intel/hda-loader.c`; this is the function logging some of the relevant messages. HDA_FW_BOOT_ATTEMPTS is currently 3

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/sound/soc/sof/intel/hda-loader.c?h=v5.8

`TIMEOUT_DEFAULT_BOOT_MS` is 2000 yet the interval between `Attempting iteration 0` and `Attempting iteration 1` at https://sof-ci.01.org/linuxpr/PR2413/build4423/devicetest/ is exactly 300 ms, @kv2019i ?

marc-hb, 2020-09-10:
From @aiChaoSONG 
> This also happened (internally) to CML_HEL_RT5682, check the failure on CML_HEL_RT5682 in https://sof-ci.sh.intel.com/#/result/planresultdetail/319

The message is exactly the same (even the PCI ID) and the timeout 300ms too. Only one retry. This has been much, much rarer than on ICL.

I just added `cml` to https://github.com/thesofproject/sof-test/pull/372

marc-hb, 2020-09-23:
@ranj063 thinks https://github.com/thesofproject/linux/issues/2434 might be related.

xiulipan, 2020-10-21:
Update the bug title as the issue is actually DSP reset before load FW to DSP.
Also add comments about other platforms, as the issue now can be seen on most platforms but with lower rate.

kv2019i, 2020-11-02:
WIP version of ICCMAX flow implementation now available: https://github.com/thesofproject/linux/pull/2548  This could potentially help with this bug.

fredoh9, 2020-11-02:
Trying to reproducing without/with linux#2548. To test this, sof-test 61b077be should be reverted first, otherwise it ignores this error.

fredoh9, 2020-11-03:
I'm still able to reproduce same problem with PR#2548. Hum... I doubled checked without the problem is reproduced. This is NOT 100% but when I tried this command it's high chance to see the problem.

```
$ ./check-runtime-pm-status.sh -l 50
```

without/with PR#2548, I'm seeing same error pattern.
```
[ 1263.646048] sof-audio-pci 0000:00:1f.3: booting DSP firmware
...
[ 1263.646371] sof-audio-pci 0000:00:1f.3: Attempting iteration 0 of Core En/ROM load...
[ 1263.646832] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0f successful
[ 1263.646848] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0e successful
[ 1263.646853] sof-audio-pci 0000:00:1f.3: unstall/run core: core_mask = 1
[ 1263.646860] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[ 1263.647396] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000000 successful
[ 1263.647416] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf010e0e successful
**[ 1263.947820] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x6000010 timedout
[ 1263.947831] sof-audio-pci 0000:00:1f.3: unknown ROM status value 06000010
[ 1263.947837] sof-audio-pci 0000:00:1f.3: status = 0x00000000 panic = 0x00000000**
[ 1263.947853] sof-audio-pci 0000:00:1f.3: extended rom status:  0x6000010 0x0 0x0 0x0 0x0 0x0 0x2001182 0x0
[ 1263.947858] sof-audio-pci 0000:00:1f.3: unknown ROM status value 06000010
[ 1263.947866] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[ 1263.947874] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000f0f successful
[ 1263.947876] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
```

plbossart, 2021-06-10:
@kv2019i @fredoh9 @marc-hb is this issue still occurring, if not can we close?

kv2019i, 2021-06-10:
I believe yes, we are hiding this with the multiple DSP boot retries. The boot-from-IMR code @keyonjie might actually help to this.
That's another question whether we want to keep a bug open.

marc-hb, 2021-06-10:
I think we could easily test this with a small, `[DO NOT UPSTREAM]` kernel patch that prints an error after the 1st or 2nd boot retry.

fredoh9, 2021-06-11:
`[DO NOT UPSTREAM]` patch is up to maintainers, but I don't think there is no point to keep this issue open as we are hiding this issue already and set P3 for this. I will close this if no one has objection in 24 hours.

closed 2021-06-14 by fredoh9

marc-hb, 2021-12-06:
Important, November 2021 refactoring: https://github.com/thesofproject/linux/pull/3250