thesofproject/sof #3409: [BUG][TGL][DSM] feedback data for DSM should be voltage information only

RDharageswari, 2020-09-14:
DSM has 2 streams, playback and a reference capture stream.
For 2 spk solution, the expected feedback is 2 channel from the MUXDEMUX11.0 DEMUX to the Host. That is Voltage data of each stream, but today we get 4 channel output(VIVI)- voltage and current information of each stream and hence the feedback is 4 channel capture.
Ideally for 2 speakers: pb will be of 2 channel, capture will also be 2 channel voltage information to the host, while the 4 channel feedback information(VIVI) will be given to the DSM in pb pipeline for quality output.
and for 4 speakers: pb will be of 4 channel, capture will also be 4 channel voltage information to the host, while the 8 channel feedback information(VIVIVIVI) will be given to the DSM module in pb pipeline.

lgirdwood, 2020-09-14:
@RDharageswari do you think a topology change is needed here in demux or do you suspect a FW issue ?

RDharageswari, 2020-09-14:
i see there is a mux_configuration available sof_mux_config and the right slot mapping in the topology file will actually help here.
numid=165,iface=MIXER,name='MUXDEMUX11.0 DEMUX'

RDharageswari, 2020-09-14:
Attaching the topology for reference:
[image]

ranj063, 2020-09-14:
@lgirdwood it looks like the ask is for the DEMUX component to provide 4-ch data to the DSM module and only provide 2-ch to the HOST component. Do you think we need a splitter to extract just the 2 channels from the 4-ch feedback data?

keyonjie, 2020-09-15:
We have already designed to use macros for all of this, @RDharageswari just try to specify different values for them: e.g.

For 2 speakers:

```
define(`SMART_PB_CH_NUM', 2)
define(`SMART_TX_CHANNELS', 2)
define(`SMART_RX_CHANNELS', 4)
define(`SMART_FB_CHANNELS', 4)
define(`SMART_REF_CH_NUM', 2)

```

For 4 speakers:

```
define(`SMART_PB_CH_NUM', 4)
define(`SMART_TX_CHANNELS', 4)
define(`SMART_RX_CHANNELS', 8)
define(`SMART_FB_CHANNELS', 8)
define(`SMART_REF_CH_NUM', 4)

```

keyonjie, 2020-09-15:
@RDharageswari feel free to experiment and submit tplg changes after verified.

sathya-nujella, 2020-09-15:
> We have already designed to use macros for all of this, @RDharageswari just try to specify different values for them: e.g.
>
> For 2 speakers:
>
> ```
> define(`SMART_PB_CH_NUM', 2) [...]

SMART_FB_CHANNELS=4  (Let's say this is - VIVI )
SMART_REF_CH_NUM=2  
Which of the 2 channels out of 4 are given as REF channels?

keyonjie, 2020-09-15:
> SMART_FB_CHANNELS=4 (Let's say this is - VIVI )
> SMART_REF_CH_NUM=2
> Which of the 2 channels out of 4 are given as REF channels?

That is also programmable, @RDharageswari is doing some similar changes for the FB channels, it worth to add macros for channel mask for both FB and REF if needed:

```
CONTROLBYTES_PRIV(DEMUX_priv,
`       bytes "0x53,0x4f,0x46,0x00,0x00,0x00,0x00,0x00,'
`       0x28,0x00,0x00,0x00,0x00,0x60,0x00,0x03,'
`       0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,'
`       0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,'
`       COMP_FORMAT_VALUE(PIPELINE_FORMAT),0x00,DEC2HEX(SMART_RX_CHANNELS),0x00,0x02,0x00,0x00,0x00,'
`       DEC2HEX(SMART_PB_PPL_ID),0x00,0x00,0x00,DEC2HEX(SMART_FB_CHANNELS),0x01,0x02,0x04,'
`       0x08,0x10,0x20,0x40,0x80,0x00,0x00,0x00,'
`       DEC2HEX(SMART_REF_PPL_ID),0x00,0x00,0x00,DEC2HEX(SMART_REF_CH_NUM),0x01,0x02,0x04,'
`       0x08,0x00,0x00,0x00,0x00,0x00,0x00,0x00"'
)

```

RDharageswari, 2020-09-16:
Hi All,
I have submitted the patches here:
https://github.com/thesofproject/sof/pull/3421

lgirdwood, 2020-09-17:
@RDharageswari cane we close now ? PR merged.

closed 2020-09-17 by RDharageswari