thesofproject/sof #3419: [BUG] Formalize definition of reset callback

cujomalainey, 2020-09-15:
**Describe the bug**
Reset callback definition is not clear on the depth of a reset

```
/**                                                                                                                                                                                
 * Resets component.                                                                                                                                                               
 * @param dev Component device.                                                                                                                                                    
 */   
```

I have noticed many components treat this callback very differently

For example:
 * DCB does nothing (maintains filter state)
 * IIR removes everything including filter config
 * volume does nothing (maintains config)

cujomalainey, 2020-09-15:
@johnylin76 @singalsu FYI

lgirdwood, 2020-09-16:
@singalsu can you take a look at this after v1.6.

bkokoszx, 2021-01-27:
Hi @singalsu I'm just checking SOF 1.7 status. Were you able to take a look at this bug?

@lgirdwood Do we need it in 1.7?

closed 2021-01-27 by lgirdwood