thesofproject/sof #3428: [BUG] [TGL] [DSM] [I2S] No feedback data captured at times after SNDRV_PCM_TRIGGER_STOP/START events on pb stream

RDharageswari, 2020-09-17:
Steps to repro the issue:
1. Do playback on speakers
2.Start stop Start stop Start the pb stream [This start and stop refers to SNDRV_PCM_TRIGGER_STOP/START events]
3.capture feedback data for 8 seconds
4.Start stop Start stop Start the pb stream
5.capture feedback data for 8 seconds
Repeat the above for 5 times, in one of the feedback captured file, the Voltage/current data is missing.
The result is not consistent across all the captured files

fw branch:
tgl-009-drop-stable

RDharageswari, 2020-09-18:
unable to provide more information due to this: https://github.com/thesofproject/sof/issues/3432

lgirdwood, 2020-09-18:
@RDharageswari can you confirm if the exception is also on master ? Is this with real/test DSM ?

RDharageswari, 2020-09-18:
This issue can be reproduced on master as well. master uses the actual DSMcomponent.i don't have data to prove this issue can be reproduced without maxim DSM as the default smart_amp throws exception with feedback stream itself as in https://github.com/thesofproject/sof/issues/3432

lgirdwood, 2020-09-18:
@RDharageswari have you tried running tools/coredumper, this will help shed light on if its related to DSM module or SOF code (it should be able to take us to the line of code) which is doing the unaligned memory read..

keyonjie, 2020-09-23:
I can reproduce this with nocodec mode, @bkokoszx I can reproduce this with below steps:
1. run aplay in one terminal, e.g. 'aplay -Dhw:0,2 -f s16_le -r48000 -c2 /dev/random -vvv -i'
2. run arecord in another terminal, e.g. 'arecord -Dhw:0,2 -f s16_le -r48000 -c2 tmp.wav -vvv'. you should see the arecord level same with aplay (e.g. aplay terminal is 99, so is the arecord terminal).
3. stop the arecord, and restart it, check if arecord level still correct.
4. do step 3 several times, you will see sometimes the arecord level is 0, which means the capture/feedback data are 0s.

bkokoszx, 2020-09-24:
@keyonjie @RDharageswari @lgirdwood 
I can reproduce issue on my RVP nocodec platform with smart amplifier test component. I can see that copying process looks fine - I mean capture pipeline (even in case when arecord level is 0) processes the right amount of data. I suspect that the data is equal to 0 - I think that maybe there is some more generic problem with SSP. I will debug it.

lgirdwood, 2020-09-24:
@bkokoszx thanks, I think we will miss v1.6- rc1 but we can hit rc final for this one.

RDharageswari, 2020-09-24:
Thanks @bkokoszx. Can you also check the PCM_START & PCM_STOP sequence once, to confirm if it is the same issue.
[image]

keyonjie, 2020-09-25:
Looks this is a SSP issue, @RDharageswari @bkokoszx can you try if below change help:
```
diff --git a/src/drivers/intel/ssp/ssp.c b/src/drivers/intel/ssp/ssp.c
index 28a267f90..1891d4324 100644
--- a/src/drivers/intel/ssp/ssp.c
+++ b/src/drivers/intel/ssp/ssp.c
@@ -730,7 +730,7 @@ static void ssp_stop(struct dai *dai, int direction)
            ssp->state[SOF_IPC_STREAM_CAPTURE] != COMP_STATE_PREPARE) {
                ssp_update_bits(dai, SSCR1, SSCR1_RSRE, 0);
                ssp_update_bits(dai, SSRSA, SSRSA_RXEN, 0);
-               ssp_empty_rx_fifo(dai);
+//             ssp_empty_rx_fifo(dai);
                ssp->state[SOF_IPC_STREAM_CAPTURE] = COMP_STATE_PREPARE;
                dai_info(dai, "ssp_stop(), RX stop");
        }

```

bkokoszx, 2020-09-25:
@keyonjie @lgirdwood 
With above change I also did not have reproduction. It looks like more generic issue. Did we hit something similar earlier on another pipelines?

RDharageswari, 2020-09-26:
Thank @keyonjie: This works for me as well

keyonjie, 2020-09-27:
Thanks for debugging and verification, @bkokoszx and @RDharageswari .

I create a PR https://github.com/thesofproject/sof/pull/3476 to fix this, please help to review.

keyonjie, 2020-10-12:
@RDharageswari shall we close this now?

closed 2020-10-19 by RDharageswari