thesofproject/sof #3430: [BUG] Dell Latitude 3410 no sound

J-Bu, 2020-09-17:
**Describe the bug**
We try to switch to sof sound handling for our own linux distribution and are not able to make it work on the Dell Latitude 3410 wit the official released versions from sof-bin.
The root cause is, that the device /dev/pcmC0D6c is not created when using the topology files from sof-bin we tried every version from v1.4.2 - v1.5.1. Thsi device is referenced in the alsa ucm2 files and if it is not present pulseaudio refuse to load the profile and skips the card completely.
We know that Ubuntu 18.04 oem osp1 has working sof support on that device and after replacing the sof-hda-generic.tplg with the one from Ubunut the device appeared and sound started working.

**To Reproduce**
Use the files released from sof-bin on the Dell Latitude 3410

**Reproduction Rate**
100%

**Expected behavior**
All needed devices are created and sound is working when using the files from sof-bin

**Impact**
Showstopper

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.9-rc5 {856deb866d16e29bd65952e0289066f6078af773}
    * SOF: All files from sof-bin (v.1.4.2, v1.4.3, v1.5, v1.5.1)
2) Name of the topology file
    * Topology: sof-hda-generic.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: Dell Latitude 3410

**Screenshots or console output**

output with sof-bin v1.5.1
[dmesg_sof_v1.5.1.txt](https://github.com/thesofproject/sof/files/5238067/dmesg_sof_v1.5.1.txt)
```
root@eLux1580-2855-2549:~# dmesg | grep sof
[    0.447023] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[    0.447025] software IO TLB: mapped [mem 0x7852c000-0x7c52c000] (64MB)
[   11.176261] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[   11.177569] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   11.178886] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[   11.182264] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   11.183697] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_globals_exit [i915])
[   11.200412] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   11.273240] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[   11.274979] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   11.276304] sof-audio-pci 0000:00:1f.3: DMICs detected in NHLT tables: 0
[   11.384722] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:5:1-88707
[   11.386099] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:16:0 Kernel ABI 3:16:0
[   11.387752] sof-audio-pci 0000:00:1f.3: warning: unknown ext header type 3 size 0x1c
[   11.389138] sof-audio-pci 0000:00:1f.3: warning: unknown ext header type 4 size 0x10
[   11.481561] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:16:0 Kernel ABI 3:16:0
[   11.482680] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[   11.486458] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[   11.487618] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[   11.493630] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[   11.494767] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[   11.499695] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[   11.501388] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[   11.506461] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[   11.507854] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[   11.507857] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[   11.507858] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[   11.507860] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[   11.507861] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[   11.508674] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[   11.508676] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[   11.509480] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[   11.509482] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[   11.510278] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[   11.517281] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[   11.585011] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC3204: line_outs=1 (0x14/0x0/0x0/0x0/0x0) type:speaker
[   11.586003] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   11.586943] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[... 69 lines cut ...]
[   11.673303] snd_hda_codec_realtek ehdaudio0D0:      Headset Mic=0x19
[   11.674291] snd_hda_codec_realtek ehdaudio0D0:      Headphone Mic=0x1a
[   11.736040] snd_hda_codec_realtek ehdaudio0D0: ASoC: sink widget AIF1TX overwritten
[   11.736957] snd_hda_codec_realtek ehdaudio0D0: ASoC: source widget AIF1RX overwritten
[   11.738045] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi3 overwritten
[   11.739025] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi2 overwritten
[   11.739847] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi1 overwritten
[   11.740687] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Codec Output Pin1 overwritten
[   11.741494] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Codec Input Pin1 overwritten
[   11.742570] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Analog Codec Playback overwritten
[   11.743357] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Digital Codec Playback overwritten
[   11.744091] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Alt Analog Codec Playback overwritten
[   11.745012] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Analog Codec Capture overwritten
[   11.746083] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Digital Codec Capture overwritten
[   11.746895] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Alt Analog Codec Capture overwritten
[   11.762525] input: sof-hda-dsp Headphone Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input12
[   11.763478] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input13
[   11.764433] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input14
[   11.765354] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input15
root@eLux1580-2855-2549:~# ls /dev/snd/
by-path    controlC0  hwC0D0     hwC0D2     pcmC0D0c   pcmC0D0p   pcmC0D1c   pcmC0D1p   pcmC0D3p   pcmC0D4p   pcmC0D5p   pcmC0D6c   pcmC0D7c   seq        timer
root@eLux1580-2855-2549:~# 
```

Tried to get logs from sof-logger but it only outputs the header and no other information. Do I need to enable any debug flags for ir to work.

lgirdwood, 2020-09-17:
@J-Bu sof-logger should work, can you paste the command line here.  I guess you only need a topology update that adds a PCM capture device  /dev/pcmC0D6c in order for pulseaudio to work ? 
@kv2019i @juimonen any comments ? My understanding is that pulseaudio should be good now but I suspect this needs a recent version of pulseaudio and alsa-lib (found in Ubuntu 20.04 onwards) ?

plbossart, 2020-09-17:
@J-Bu can you provide information on which versions of alsa-lib, alsa-ucm-conf and pulseaudio you are using in your distro?

also did you actually install the topology files? The fact that you are missing a capture file is likely a difference with topology. I really don't get how you are missing a capture device if you don't have any dmics. The only capture device on your platform is likely the HDaudio codec.

kv2019i, 2020-09-17:
@J-Bu Hmm, PCM6 and PCM7 are for DMIC and these are indeed required by the upstream UCM files for SOF HDA driver. But when I look at the dmesg, I see this:
[   11.276304] sof-audio-pci 0000:00:1f.3: DMICs detected in NHLT tables: 0

So this reports there are no digital mics connected to PCH. But a bit earlier, there's this:
[   11.177569] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver

.. so we are getting conflicting data from BIOS, and end up using SOF for a configuration we do not support in our upstream UCM files. @plbossart ever seen this?

plbossart, 2020-09-17:
@kv2019i the SOF driver would be selected only if there is a kernel parameter, e.g

options snd_intel_dspcfg dsp_driver=3

or a quirk for that device.

In this case indeed things would work fine, but we probably need to make the DMIC devices conditional in the UCM files.

Now that I think of it the TGL device we both have also doesn't have a dmic but things work fine when you force the use of the SOF driver, the only requirement is that the ucm2 stuff be used.

kv2019i, 2020-09-17:
@plbossart 
No, I don't think that's the case here, and rather dsp_driver=0. But:
[ 11.177569] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver

So snd_intel_dsp_check_dmic(pci)) returns true. But when we load topology, we query NHLT tables for dmic numbers and get zero back. This is not a config we support in UCM.

kv2019i, 2020-09-17:
@J-Bu You can experiment by overriding the topology:
sudo cp /lib/firmware/intel/sof-tplg/sof-hda-generic-2ch.tplg /lib/firmware/intel/sof-tplg/sof-hda-generic.tplg

... i.e. use the 2 dmics topology file in case BIOS reports 0 dmics. Not a real fix, but might get you the PCM6+7.

plbossart, 2020-09-17:
We recently added a patch to the kernel to allow for single mics, before we only allowed for 2 and 4.

ASoC: SOF: intel: hda: support also devices with 1 and 3 dmics
3dca35e35b42b3405ddad7ee95c02a2d8cf28592

as well as topologies that will use the relevant names.

juimonen, 2020-09-17:
@J-Bu @plbossart @kv2019i I checked the topologies and the ubuntu 18.04 oem sof-hda-generic.tplg attached to this PR has the dmic definition and the sof 1.5 sof-hda-generic.tplg doesn't. so @J-Bu that's clear case why you don't have the dmic devices...

So is is now so that the topology selection mechanism in kernel wasn't in 18.04 oem, and ubuntu just copied the 2ch as generic? And is is that the selcetion mechanism is broken in this kernel as it is not choosing the 2ch topology?

J-Bu, 2020-09-18:
First of all thanks for the quick reply.

**environment**
Currently we are using following versions:
alsa-lib: 1.2.2
alsa-ucm-conf: 1.2.2
pulseaudio: 12

But we have also tried latest releases versions but it did not change anything
alsa-lib: 1.2.3
alsa-ucm-conf: 1.2.3
pulseaudio: 13

**sof-logger**
I am nit quite sure which lcd file I need to use so I tried multiple:
 * sof-cnl-v1.5.1.ldc
 * sof-icl-v1.5.1.ldc
 * sof-tgl-v1.5.1-58b36ed.ldc

All lcd files let to the same result
```
root@eLux1580-2855-2549:~# ./sof-logger -l /lib/firmware/intel/sof/v1.5.1/sof-cnl-v1.5.1.ldc  
         TIMESTAMP              DELTA C# COMPONENT          LOCATION                       CONTENT
```
When running sof-logger following line is printed to dmesg
```
[  381.922711] sof-audio-pci 0000:00:1f.3: error: debugfs entry cannot be read in DSP D3
```

**topology**
I copied sof-hda-generic-2ch.tplg to sof-hda-generic.tplg and the dmics are detected and sound is working.
@kv2019i 
Does this have any bad impact on other devices?

**kernel**
I applied the mentioned patch to head of 5.9-rc5 and used topology files from sof master

ASoC: SOF: intel: hda: support also devices with 1 and 3 dmics
3dca35e35b42b3405ddad7ee95c02a2d8cf28592

Sound is working and dmesg now returns that NHLT tables contains 1 dmic:
```
[   12.224702] sof-audio-pci 0000:00:1f.3: DMICs detected in NHLT tables: 1
```

juimonen, 2020-09-18:
@J-Bu ok, so then the proper topology would be sof-hda-generic-1ch.tplg, so you should get "dual mono" from 1 dmic. But I think this is only officially released with sof 1.6, so in coming weeks. 

This might work with sof-hda-generic-2ch, but not sure how the channels go then? could you try to record from the dmic with arecord or similar (stereo file) and see the actual wav file, do you have audio in both channels? If not then the 1ch topology should be used.

juimonen, 2020-09-18:
@J-Bu and now when reading more carefully your previous post, If you copied topologies from sof master the 1 channel topology should be chosen automatically, and thus you should have "dual mono" dmic. So the copy of 2ch over generic should not have any effect.

J-Bu, 2020-09-18:
I think my response was a bit too quick.

When using topology files from sof-master the devices are created but no sound is played back. Maybe newer alsa-lib, alsa-ucm-conf and pulseaudio are needed.

If I use the pacthed kernel and copy the sof-hda-generic-2ch.tplg from v1.5.1 sof-hda-generic-1ch.tplg sound is working. I think this is the way for us to go for the time beeing and if sof 1.6 is released we will try again with updated alsa-lib and alsa-ucm-conf.

lgirdwood, 2020-09-18:
@slawblauciak any comments on sof-logger ?

slawblauciak, 2020-09-18:
From my very basic understanding the firmware log can only be obtained in D0 power state.
I don't see much indication in the dmesg that firmware is even booted. So, I don't find it surprising the log output is empty and the D3 error is visible in dmesg.

lgirdwood, 2020-09-18:
@slawblauciak ok, so we probably want to have sof-logger checking driver PM state (in generic sysfs) and report a warning e.g. "DSP in D3 state so log may be not be available" as a future update.