thesofproject/sof #3432: [BUG] [TGL] [DSM] [I2S] Echo Reference stream throws FW exception with  sof-tgl-max98373-rt5682.tplg

RDharageswari, 2020-09-18:
**Describe the bug**
Echo Reference stream throws FW exception and DSP panics, with tgl-009-drop-stable branch: 0.9 release
The sof-tgl-max98373-rt5682.tplg uses smart_amp_test component.

**Reproduction Rate**
All the time

**Expected behavior**
Echo Reference stream should capture the feedback data from codec

**Impact**
showstopper

**Environment**
tgl-009-drop-stable
sof-tgl-max98373-rt5682.tplg

**Screenshots or console output**
localhost ~ # aplay -Dhw:0,0 Titanium_low1.wav -c2 -r48000 -fs24_le &
[1] 2860
localhost ~ # Warning: format is changed to S16_LE
Playing WAVE 'Titanium_low1.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
Warning: rate is not accurate (requested = 44100Hz, got = 48000Hz)
         please, try the plug plugin

localhost ~ # arecord -Dhw:0,0 try_chk.wav  -d 10 -c4 -r48000 -f s24_le
Recording WAVE 'try_chk.wav' : Signed 24 bit Little Endian, Rate 48000 Hz, Channels 4
arecord: set_params:1405: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S24_LE
SUBFORMAT:  STD
SAMPLE_BITS: 32
FRAME_BITS: 128
CHANNELS: 4
RATE: 48000
PERIOD_TIME: 21250
PERIOD_SIZE: 1020
PERIOD_BYTES: 16320
PERIODS: (4 5)
BUFFER_TIME: 85250
BUFFER_SIZE: 4092
BUFFER_BYTES: 65472
TICK_TIME: 0
localhost ~ # aplay: pcm_write:2059: write error: Input/output error

kernel logs:
[   93.555847] sof-audio-pci 0000:00:1f.3: error : DSP panic!
[   93.555850] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   93.555951] sof-audio-pci 0000:00:1f.3: error: runtime exception
[   93.555951] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[   93.555952] sof-audio-pci 0000:00:1f.3: error: panic at :0
[   93.555953] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[   93.555954] sof-audio-pci 0000:00:1f.3: error: Exception Cause: LoadStoreAlignmentCause, Load or store to an unaligned address
[dmesg_logs.txt](https://github.com/thesofproject/sof/files/5242918/dmesg_logs.txt)

RDharageswari, 2020-09-18:
Tried with the following formats also:
s24_le playback + s32_le capture still the same issue

lgirdwood, 2020-09-18:
@RDharageswari does the sof-logger error log show any data ? Are you using the test DSM or real DSM ?

RDharageswari, 2020-09-18:
This issue is not on master, as the master uses maxim DSM component

keyonjie, 2020-09-23:
Hi @RDharageswari can you upload the .wav file you mentioned, looks it is S16_LE format?

keyonjie, 2020-09-23:
Hi @bkokoszx I can reproduce this issue, once we run aplay and arecord with different format specified, the DSP will panic.

bkokoszx, 2020-09-24:
@keyonjie @RDharageswari 
#3464 should fix DSP panic. With this PR and different formats on playback & capture we get an error:
```
pipe_err(current->pipeline, "pipeline_comp_params_neg(): params conflict with existed active pipeline!");
```

keyonjie, 2020-09-24:
cool @bkokoszx thanks for fixing this!

lgirdwood, 2020-09-25:
@RDharageswari #3464 now merged, can we close ?

closed 2020-10-19 by RDharageswari