thesofproject/sof #3434: [BUG][CI][TGL-010] failed to set/get ctrl ipc comp for codec adacpter component after module reload

aiChaoSONG, 2020-09-18:
**Describe the bug**
In module load/reload test case, after module reloading, topology will be reloaded and component parameters will be restored. 
in restoring parameters to codec adapter component, the below error showed in dmesg.
```
kernel: [ 7034.950621] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50020000: GLB_COMP_MSG: GET_VALUE
kernel: [ 7034.950625] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50040000: GLB_COMP_MSG: GET_DATA
kernel: [ 7034.950703] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x50040000 size 124
kernel: [ 7034.950704] sof-audio-pci 0000:00:1f.3: error: set/get ctrl ipc comp 2
kernel: [ 7034.950705] sof-audio-pci 0000:00:1f.3: error: kcontrol value get for widget: 2
kernel: [ 7034.950711] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50040000: GLB_COMP_MSG: GET_DATA
kernel: [ 7034.950793] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x50040000 size 216
kernel: [ 7034.950794] sof-audio-pci 0000:00:1f.3: error: set/get ctrl ipc comp 2
kernel: [ 7034.950794] sof-audio-pci 0000:00:1f.3: error: kcontrol value get for widget: 2
``` 

**To Reproduce**
run test case in sof-test repo: 
- for TGL_RVP_HDA, run
`TPLG=sof-hda-generic-4ch-kwd.tplg ~/sof-test/test-case/check-kmod-load-unload.sh -l 50`
- for TGL_RVP_NOCODEC , run
`TPLG=sof-tgl-nocodec-ci.tplg ~/sof-test/test-case/check-kmod-load-unload.sh -l 50`

**Reproduction Rate**
reporduced in the first iteration, should be high

**Impact**
No negative functional impact

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {topic/sof-dev 3334de38}
    * SOF: {tgl-010-drop-stable e0d648b5}
2) Name of the topology file
    * Topology: {sof-tgl-nocodec-ci.tplg for TGL_RVP_NOCODEC, sof-hda-generic-4ch-kwd.tplg for TGL_RVP_HDA}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {TGL_RVP}

[dmesg.txt](https://github.com/thesofproject/sof/files/5243435/dmesg.txt)

lgirdwood, 2020-09-18:
@aiChaoSONG is there any sof-logger trace or error log here ?

aiChaoSONG, 2020-09-18:
@lgirdwood Sorry for forgetting to mention that @keyonjie  provide a fix on tgl-010-drop-stable branch: https://github.com/thesofproject/sof/commit/6d3e2ee834eff5946d057bee07cf7c3920a9fa46

slawblauciak, 2020-09-22:
@aiChaoSONG could you please confirm if the change in question is effective?

keyonjie, 2020-09-23:
This is fixed on tgl-010-drop-stable branch, the fix will be upstream together with post processing implementation, close the issue here.

closed 2020-09-23 by keyonjie