thesofproject/sof #3435: [BUG][APL] Firmware debug build failed for APL platform.

jdrzaszcz, 2020-09-18:
**Describe the bug**
Firmware debug build has failed for APL platform since commit fc5d74fccf690b20eb53daa66400e6ef85935eae.

Debug options enabled:
echo "CONFIG_DEBUG=y" >> override.config
echo "CONFIG_TRACEV=y" >> override.config
echo "CONFIG_PROBE=y" >> override.config
echo "CONFIG_PROBE_POINTS_MAX=16" >> override.config
echo "CONFIG_PROBE_DMA_MAX=4" >> override.config
echo "CONFIG_GDB_DEBUG=y" >> override.config
echo "CONFIG_DEBUG_HEAP=y" >> override.config
echo "CONFIG_DEBUG_LOCKS=y" >> override.config
echo "CONFIG_DEBUG_IPC_COUNTERS=y" >> override.config
echo "CONFIG_PERFORMANCE_COUNTERS=y" >> override.config

Build failed due to `.bss' will not fit in region `sof_fw'.

/home/audiodsp/SOF/xtensa/APL/RG-2017.8-linux/XtensaTools/bin/xt-ld: sof section `.bss' will not fit in region `sof_fw'
CMakeFiles/sof.dir/build.make:2910: recipe for target 'sof' failed
make[3]: *** [sof] Error 2
make[3]: Leaving directory '/home/audiodsp/SOF/jdrzasz/sof/build_cnl'
CMakeFiles/Makefile2:400: recipe for target 'CMakeFiles/sof.dir/all' failed
make[2]: *** [CMakeFiles/sof.dir/all] Error 2
make[2]: Leaving directory '/home/audiodsp/SOF/jdrzasz/sof/build_cnl'
CMakeFiles/Makefile2:1589: recipe for target 'src/arch/xtensa/CMakeFiles/bin.dir/rule' failed
make[1]: *** [src/arch/xtensa/CMakeFiles/bin.dir/rule] Error 2
make[1]: Leaving directory '/home/audiodsp/SOF/jdrzasz/sof/build_cnl'
Makefile:661: recipe for target 'bin' failed
make: *** [bin] Error 2

**To Reproduce**
Enable mentioned debug options and build FW with SOF master commit 646e5623990871b4c1beb142974d42ae06332462.

**Reproduction Rate**
100 %
[apl_build.txt](https://github.com/thesofproject/sof/files/5243798/apl_build.txt)

**Expected behavior**
Build should be done without warning/error

**Environment**
SOF master commit: 646e5623990871b4c1beb142974d42ae06332462

**Screenshots or console output**
APL build log: 
[apl_build.txt](https://github.com/thesofproject/sof/files/5243803/apl_build.txt)

lgirdwood, 2020-09-18:
@jdrzaszcz it looks like we are hitting the memory limits here form APL. I guess this all works if you remove other items like some audio processing components ?

jdrzaszcz, 2020-09-18:
@lgirdwood Yes that's probably this problem. When I removed two audio components, firmware compilation is complete.

lgirdwood, 2020-09-18:
@jdrzaszcz shall we close ?

jdrzaszcz, 2020-09-19:
@lgirdwood Yes, we can close it. We will have to change config in CI for APL debug build.

closed 2020-09-19 by jdrzaszcz