thesofproject/sof #3438: check-alsabat-headset-playback.sh frequency failures found on BYT_MB_NOCODEC, others.

plbossart, 2020-09-19:
https://sof-ci.01.org/linuxpr/PR2452/build4513/devicetest/

check-alsabat-headset-playback.sh for BYT_MB_NOCODEC

````
BAT analysis: signal has 65536 frames at 44100 Hz, 1 channels, 2 bytes per sample.

Channel 1 - Checking for target frequency 997.00 Hz
Amplitude: 17958.7; Percentage: [54]
Detected peak at 988.51 Hz of 20.67 dB
 Total 20.7 dB from 988.51 to 988.51 Hz
Detected peak at 997.26 Hz of 36.36 dB
 Total 41.7 dB from 990.53 to 1002.64 Hz
 PASS: Peak detected at target frequency
Detected peak at 1003.99 Hz of 20.69 dB
 Total 41.8 dB from 1003.99 to 1003.99 Hz
Detected peak at 1006.00 Hz of 20.30 dB
 Total 41.8 dB from 1006.00 to 1006.00 Hz
Detected at least 4 signal(s) in total

Return value is -1003
2020-09-18 16:50:01 UTC [REMOTE_ERROR] Starting func_exit_handler(), exit status=1, FUNCNAME stack:
2020-09-18 16:50:01 UTC [REMOTE_ERROR]  main()  @  /home/ubuntu/sof-test/test-case/check-alsabat.sh
2020-09-18 16:50:02 UTC [REMOTE_INFO] Starting /usr/bin/sof-logger  -l /etc/sof/sof-byt.ldc -o /home/ubuntu/sof-test/logs/check-alsabat/2020-09-18-16:49:54-26390/etrace.txt
error: in logger_read(), fread(..., /sys/kernel/debug/sof/etrace) failed: Inappropriate ioctl for device(25)
alsa-utils version 1.2.2
````
@fredoh9 @aiChaoSONG @marc-hb does this ring a bell?

aiChaoSONG, 2020-09-21:
I think we have two issues there, one is https://github.com/thesofproject/sof-test/issues/379, I working on this but delayed due to TGL release, the other is `error: in logger_read(), fread(..., /sys/kernel/debug/sof/etrace) failed: Inappropriate ioctl for device`, I think it is sof-logger or SOF issue.

aiChaoSONG, 2020-09-21:
> Not sure what is contributing the frequency difference time to time

@fredoh9 It's the ramp (fade in) which contributing to the frequency.

lgirdwood, 2020-09-21:
Fwiw, I do manually check all alsa-bat failures by downloading the WAV and listening to the output as well as viewing in audacity. I also remember discussing with @singalsu a while back that we do need to make the pass band window larger.
@singalsu can you comment on the alsa-bat pass band window size do @marc-hb can fix if necessary.

marc-hb, 2020-09-21:
> the other is error: in logger_read(), fread(..., /sys/kernel/debug/sof/etrace) failed: Inappropriate ioctl for device, I think it is sof-logger or SOF issue.

I shared a fix for this in the comment immediately before yours.

aiChaoSONG, 2020-09-22:
confirmed two issues are here,  create new ones: 
- alsabat failure: https://github.com/thesofproject/sof-test/issues/379
- sof-logger failure: https://github.com/thesofproject/sof/issues/3446

close this compond issue

closed 2020-09-22 by aiChaoSONG

marc-hb, 2020-09-22:
> create new ones:
> alsabat failure: thesofproject/sof-test#379

I didn't understand the purpose of creating a new one instead of keeping this one but 379 is actually an old issue, thanks for closing this duplicate.

aiChaoSONG, 2020-09-23:
@marc-hb This issue is actually two issues with different code, so I split it