thesofproject/sof #3444: [FEATURE]Add coverity to CI

cujomalainey, 2020-09-21:
**Is your feature request related to a problem? Please describe.**
Static analysis is an excellent way to catch bugs, its free lets use it

**Describe the solution you'd like**
Coverity easily integrates into Travis https://scan.coverity.com/travis_ci

**Describe alternatives you've considered**
cppcheck

**Additional context**
I can hopefully try this out in next week or so

cujomalainey, 2020-09-21:
@plbossart if you want we can investigate this for kernel builds as well. Its already running upstream, just need to scope the failure to our related code https://scan.coverity.com/projects/linux

plbossart, 2020-09-21:
@cujomalainey I used to run coverity a long time ago, but their free scans are not meant to be run for each PR. IIRC they wanted to have e.g. a branch and a check run on a weekly basis. Sometimes the scans could take up to 24 hours to complete, and there are lots of false positives. It's the same as cppcheck, it's hard to use in CI since it's extremely verbose.

plbossart, 2020-09-21:
@cujomalainey see the info from your link: https://scan.coverity.com/travis_ci

````
Build Submission Frequency
It's probably overkill to run static analysis on each and every commit of your project. To increase availability of the free service to more projects, the addon is designed by default to run analysis on a per-branch basis. We recommend you create a branch named coverity_scan, which you can merge into whenever you would like to trigger analysis. See the FAQ for information about build submission frequency.
````

cujomalainey, 2020-09-21:
Maybe it worth running coverity on release branches and making cppcheck on main build? Fixing as many errors as possible before adding to CI?

plbossart, 2020-09-21:
You don't have to convince me @cujomalainey, I run cppcheck, make W=1 and sparse on the kernel driver code. On the firmware side things can be improved....

````
Checking src/audio/crossover/crossover.c ...
src/audio/crossover/crossover.c:225:10: style: Variable 'err' is assigned a value that is never used. [unreadVariable]
 int err = 0;
         ^
30/191 files checked 18% done
Checking src/audio/crossover/crossover_generic.c ...
Checking src/audio/crossover/crossover_generic.c: CONFIG_FORMAT_S16LE...
src/audio/crossover/crossover_generic.c:150:10: style: Variable 'idx' is assigned a value that is never used. [unreadVariable]
 int idx = 0;
         ^
Checking src/audio/crossover/crossover_generic.c: CONFIG_FORMAT_S24LE...
src/audio/crossover/crossover_generic.c:189:10: style: Variable 'idx' is assigned a value that is never used. [unreadVariable]
 int idx = 0;
         ^
Checking src/audio/crossover/crossover_generic.c: CONFIG_FORMAT_S32LE...
src/audio/crossover/crossover_generic.c:228:10: style: Variable 'idx' is assigned a value that is never used. [unreadVariable]
 int idx = 0;
         ^
Checking src/audio/kpb.c ...
src/audio/kpb.c:1319:9: style: Unused variable: i [unusedVariable]
 size_t i;
        ^
src/audio/kpb.c:1321:9: style: Unused variable: channel [unusedVariable]
 size_t channel;
        ^
Checking src/audio/pipeline.c ...
src/audio/pipeline.c:551:13: style: Variable 'data.start' is reassigned a value before the old one has been used. [redundantAssignment]
 data.start = host;
            ^
src/audio/pipeline.c:539:13: note: data.start is assigned
 data.start = host;
            ^
src/audio/pipeline.c:551:13: note: data.start is overwritten
 data.start = host;
            ^
src/audio/pipeline.c:550:14: style: Variable 'data.params' is reassigned a value before the old one has been used. [redundantAssignment]
 data.params = params;
             ^
src/audio/pipeline.c:540:14: note: data.params is assigned
 data.params = &hw_params;
             ^
src/audio/pipeline.c:550:14: note: data.params is overwritten
 data.params = params;
             ^
Checking src/audio/smart_amp/smart_amp_maxim_dsm.c ...
src/audio/smart_amp/smart_amp_maxim_dsm.c:214:43: style: Clarify calculation precedence for '&' and '?'. [clarifyCalculation]
   ch = param->param.id & DSM_CH1_BITMASK ? 0 : 1;
                                          ^
src/audio/smart_amp/smart_amp_maxim_dsm.c:134:10: style: Variable 'ret' is assigned a value that is never used. [unreadVariable]
 int ret = 0;
         ^
Checking src/audio/smart_amp_test.c ...
src/audio/smart_amp_test.c:69:2: warning: Either the condition 'if(sad)' is redundant or there is possible null pointer dereference: sad. [nullPointerRedundantCheck]
 sad->model_handler = comp_data_blob_handler_new(dev);
 ^
src/audio/smart_amp_test.c:77:6: note: Assuming that condition 'if(sad)' is not redundant
  if (sad)
     ^
src/audio/smart_amp_test.c:69:2: note: Null pointer dereference
 sad->model_handler = comp_data_blob_handler_new(dev);
 ^
Checking src/drivers/imx/edma.c ...
src/drivers/imx/edma.c:368:13: style: Variable 'handshake' is assigned a value that is never used. [unreadVariable]
  handshake = config->dest_dev;
            ^
src/drivers/imx/edma.c:373:13: style: Variable 'handshake' is assigned a value that is never used. [unreadVariable]
  handshake = config->src_dev;
[... 133 lines cut ...]
   if (ret < 0)
           ^
src/probe/probe.c:905:10: style: Checking if unsigned expression 'ret' is less than zero. [unsignedLessThanZero]
 if (ret < 0)
         ^
Checking src/trace/trace.c ...
src/trace/trace.c:216:13: error: Comparing pointers that point to different objects [comparePointers]
 while (ptr < end) {
            ^
src/trace/trace.c:209:15: note: Variable declared here.
 extern void *_trace_ctx_start;
              ^
src/trace/trace.c:211:40: note: Address of variable taken here.
 struct tr_ctx *ptr = (struct tr_ctx *)&_trace_ctx_start;
                                       ^
src/trace/trace.c:210:15: note: Variable declared here.
 extern void *_trace_ctx_end;
              ^
src/trace/trace.c:212:40: note: Address of variable taken here.
 struct tr_ctx *end = (struct tr_ctx *)&_trace_ctx_end;
                                       ^
src/trace/trace.c:216:13: note: Comparing pointers that point to different objects
 while (ptr < end) {
            ^
````

cujomalainey, 2020-09-21:
Are you running this manually? Because we could move this to CI for you?

plbossart, 2020-09-21:
> Are you running this manually? Because we could move this to CI for you?

yes, the log is way to verbose for a CI pass/fail report. It's got to be a maintainer check done from time to time, with developers strongly encouraged to do the same.

If you look at the kernel level Colin King runs coverity but he only detects deltas from his 'known' state to the new errors. It's quite hard to automate, someone needs to look at the results.

plbossart, 2020-09-22:
command used for reference

cppcheck --platform=unix32 --force --max-configs=1024 --inconclusive --enable=all --suppress="variableScope" src

cujomalainey, 2020-09-22:
I mean, we could run an informational CI on only the files diffed in the PR to help infer if someone is introducing new errors.