thesofproject/sof #3446: [BUG][logger] 'Inappropriate ioctl for device' when start logger to acquire error log

aiChaoSONG, 2020-09-22:
**Describe the bug**
Once our test failed, we will start sof-logger like this `/usr/bin/sof-logger  -l /etc/sof/sof-byt.ldc -o /home/ubuntu/sof-test/logs/check-alsabat/2020-09-18-16:49:54-26390/etrace.txt` to acquire error log. and the logger conplained `error: in logger_read(), fread(..., /sys/kernel/debug/sof/etrace) failed: Inappropriate ioctl for device(25)`

check the failure in https://sof-ci.01.org/linuxpr/PR2452/build4513/devicetest/. We also observed this sof-logger failure with other test case failure.

**Reproduction Rate**
once test case failed, quite high rate to see this sof-logger issue.

**Expected behavior**
sof-logger should be able to collect error log.

**Environment**
Linux: 337674140e88 ASoC: SOF: topology: remove const in sizeof()
SOF: 646e5623 travis: remove duplicate and deprecated host-build-all.sh

marc-hb, 2020-09-22:
This happens when a second, concurrent sof-logger is started and can be reproduced very easily with a one-line change, see example in https://github.com/thesofproject/sof-test/pull/386

I think it has been like this since forever, it's just no one cared about sof-logger issues before: most errors were ignored and messages were silenced, see #3402 , https://github.com/thesofproject/sof-test/pull/373 etc.

lgirdwood, 2020-09-23:
@slawblauciak could be a race or etrace is empty ?

slawblauciak, 2020-09-23:
@lgirdwood that looks like a kernel issue to me

ktrzcinx, 2020-09-23:
It doesn't look like like race, after quick reading the code, I looks rather like reading empty file error, but it's not fully fit bug description.

ktrzcinx, 2020-09-24:
@aiChaoSONG could you check if this patch could help: https://github.com/thesofproject/sof/pull/3463/files ?

slawblauciak, 2020-11-04:
@aiChaoSONG any updates?

marc-hb, 2020-11-04:
This could be a different issue but I noticed trying to access the `etrace` in suspend produces a confusing error message. More investigation / stress-testing is on my TODO list but keeps getting de-prioritized :-(

aiChaoSONG, 2020-11-05:
@marc-hb If you mean the below error message, then we are all good, this message has nothing to do with any error, it is used to avoid an empty error trace, which will help us to detect if sof-logger error trace works. It is printed on purpose. check https://github.com/thesofproject/sof/issues/3265

```
[103725963721.614594] (             nan) c0 dma-trace             src/trace/dma-trace.c:326  ERROR FW ABI 0x3011000 DBG ABI 0x5001000 tag v1.5-rc1-666-g46e49cc3b452 src hash 0x19c2e0e4 (ldc hash 0x19c2e0e4)
[103725963721.614594] (        0.000000) c0 dma-trace             src/trace/dma-trace.c:326  ERROR FW ABI 0x3011000 DBG ABI 0x5001000 tag v1.5-rc1-666-g46e49cc3b452 src hash 0x19c2e0e4 (ldc hash 0x19c2e0e4)
```

marc-hb, 2020-11-05:
No I meant some `ioctl` error message.

aiChaoSONG, 2022-01-11:
This is not reproducible any more, let's close

closed 2022-01-11 by aiChaoSONG