thesofproject/sof #3449: [BUG][TGL][POST PROCESSING] tgl-010-drop-stable: Stream Stall in multicore post processing for specific audio formats

zrombel, 2020-09-22:
**Describe the bug**
Issue concerns all audio formats except audio format: 48000Hz 32b 32b 2ch.
In topology where all pipelines are on primary core and Post Processing is on secondary core, playback pipeline doesn't copy data on playback pipeline after trigger start is send.

**Topology**
```
                +------------------------------------+
    +------+    | +---+  +-------+  +---+  +-------+ |
    | Host |----->|Buf|->|GenProc|->|Buf|->|SSP Dai|---+
    +------+    | +---+  +-------+  +---+  +-------+ | |
                +------------------------------------+ |
                                                       |
                +------------------+                   |
    +------+    | +---+  +-------+ |                   |
    | Host |<-----|Buf|<-|SSP Dai|<--------------------+
    +------+    | +---+  +-------+ |
                +------------------+
```

**To Reproduce**
Run python tests with slim driver:
TestGenericProcessorMultiCorePlb48000Hz16b16b2ch
TestGenericProcessorMultiCorePlb48000Hz16b16b4ch
TestGenericProcessorMultiCorePlb48000Hz16b16b8ch
TestGenericProcessorMultiCorePlb48000Hz32b32b4ch
TestGenericProcessorMultiCorePlb48000Hz32b32b8ch

with option:
--playback_option=10

**Reproduction Rate**
~1/10

**Environment**
1) SOF: branch: tgl-010-drop-stable, hash: 251f8de
2) Python tests: branch: master, hash: 3c251ce9839d0e547fdb500edd624c1f5ae8d1e6

**Logs**
[tgl-010-postprocessing-mc-stream-stall.zip](https://github.com/thesofproject/sof/files/5262604/tgl-010-postprocessing-mc-stream-stall.zip)

jdrzaszcz, 2020-09-25:
I rerun the multicore postprocessing tests with tgl-010-drop-stable (hash: 251f8de33eb4f455299f7e22cb83f3d8303576ff) + https://github.com/thesofproject/sof/pull/3464, and it looks like this PR fixes this issue.

slawblauciak, 2020-09-29:
@jdrzaszcz please verify if the fix is effective on latest tgl branches, thanks

jdrzaszcz, 2020-09-30:
@slawblauciak Rerun  multicore post processing tests with playback_iterations=100 on top of tgl-010-drop-stable, hash: 0ba4010e5564120080786ac0b43d574d00bc4e2e and there was no Stream Stall for any format.

closed 2020-09-30 by jdrzaszcz