thesofproject/sof #3466: [BUG] LPRO ext manifest is not part of firmware

fredoh9, 2020-09-25:
**Describe the bug**
I built with cnl firmware with CONFIG_CAVS_LPRO_ONLY=y. There is no LPRO information in ext manifest.

```
 .fw_metadata   0x220001a0      0x128 src/init/libext_manifest.a(ext_manifest.c.o)
                0x220001a0                ext_man_fw_ver
                0x220001f0                ext_man_cc_ver
                0x22000260                ext_man_probe
                0x22000290                ext_man_dbg_info
                0x220002b0                ext_man_config
                0x220002d0                . = ALIGN (_EXT_MAN_ALIGN_)
 *fill*         0x220002c8        0x8 
```

form map file, this is only hit
```
LOAD src/platform/intel/cavs/libcavs_ext_manifest.a
```

**To Reproduce**

This fix is required
https://github.com/thesofproject/sof/pull/3465

./scripts/xtensa-build-all.sh -d -c cnl
then select Platform->Use low power ring oscillator always
then check sof.map for .fw_metadata section

**Reproduction Rate**
100%

**Expected behavior**
LPRO information as cavs platform specifc data (EXT_MAN_ELEM_PLATFORM_CONFIG_DATA) in ext manifest

**Impact**
ICCMAX implementation for ICL is blocked

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel:40e44b3a8841e918493a4eb80a76c45007263ac9 (any of topic/sof-dev)
    * SOF: c59b513eaf19e5551ca6afbde2bd3a6d90cc8063
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: I'm using cml and icl

fredoh9, 2020-09-25:
Looks this the data I need to parse. Let me see what I can do.
```
 .fw_metadata   0x22000000      0x198 CMakeFiles/sof.dir/src/platform/intel/cavs/platform.c.o
 *fill*         0x22000198        0x8
```

abonislawski, 2020-09-25:
I will prepare a fix, looks like LPRO define was not updated for ext_manifest

slawblauciak, 2020-09-29:
@fredoh9 the PR has been merged, good to close?

abonislawski, 2020-09-29:
@slawblauciak #3473 is also needed

fredoh9, 2020-09-30:
Final piece is https://github.com/thesofproject/sof/pull/2957, but we can close this.

closed 2020-09-30 by fredoh9