thesofproject/sof #3467: [FEATURE] move debug and optimization to global configs in cmake

cujomalainey, 2020-09-25:
**Is your feature request related to a problem? Please describe.**
Allow optimizations to be globally configurable, as they should be, we should not be enforcing anything on the developer

**Describe the solution you'd like**
move to cmake vars/env vars

**Describe alternatives you've considered**
none

**Additional context**
https://github.com/thesofproject/sof/pull/3458#discussion_r494675193

lgirdwood, 2020-09-25:
@jajanusz fyi.

lgirdwood, 2021-01-12:
@cujomalainey I guess this is still WIP since Janusz has left Intel ? IIRC I've seen one or two PRs to help here but assume we need more ?

cujomalainey, 2021-01-13:
I think we can do more work on this once we stop running Cmakes both with and without kconfig. If it's only with kconfig values set then we can adjust things as necessary

lgirdwood, 2021-01-14:
@cujomalainey I think we have the global debug/optimisation configs in kconfig today but there may be one or tow places where it's hard coded in cmake. Would using only kconfig work here for you to set the cmake vars ?

cujomalainey, 2021-01-15:
Thanks, once #3617 is complete i can go around and fixup hardcoded flags where i find them

lgirdwood, 2021-01-18:
@marc-hb do you think you can fix #3617 this week ?

marc-hb, 2021-01-19:
I don't think so, sorry. I had a very quick look a long time ago and it seemed to require significant CMake surgery. I'm familiar with CMake but not with our CMakeLists.txt files.

bkokoszx, 2021-01-27:
@cujomalainey Regarding https://github.com/thesofproject/sof/issues/3467#issuecomment-760560305 comment - will you be able to take a look at this after #3617 gets merged?

cujomalainey, 2021-01-27:
Yes, this should be much easier

lgirdwood, 2021-05-10:
Aim is to align with how Zephyr set configuration. More of this will align in v1.9

marc-hb, 2021-05-10:
Just for the record: Zephyr does not use CMAKE_BUILD_TYPE.

(and .config change #3617 has been merged for some time)