thesofproject/sof #3474: [FEATURE] Can't find a way to create sof-tgl.ri file in stable-v1.6 branch

KaiChuan-Hsieh, 2020-09-25:
I am trying to study how to utilize the SDK to create required bin files for TGL platform.
Here are steps I've done:
=> Download repo
$ mkdir -p ~/.bin
$ PATH="${HOME}/.bin:${PATH}"
$ curl https://storage.googleapis.com/git-repo-downloads/repo > ~/.bin/repo
$ chmod a+rx ~/.bin/repo
=> Download manifest
$ mkdir sof
$ cd sof
$ repo init -u git@github.com:thesofproject/sof-manifests.git
$ repo sync
=> Build toolchain and sof firmware(ex. sof-cnl.ri)
$ ./build-script/build-sof.sh
=> Build tplg
$ cd sof/sof
$ ./scripts/build-tools.sh (need to use latest alsa-lib libatopology.so.2.0.0 to replace in /usr/lib/x86_64-linux-gnu/libatopology.so.2.0.0)

However, I can't find a way to create sof-tgl.ri, I try to checkout to stable-v1.6 branch, but it shows it requires MEU_TOOL in xtensa-build-all.sh for signing the firmware, and I try to use "make bin" under build_tgl_gcc folder, but it shows

rimage: /home/u/Workspace/sof/sof/sof/rimage/src/rimage.c:185: main: Assertion `image.adsp->write_firmware' failed.
Aborted (core dumped)
make[3]: *** [src/arch/xtensa/CMakeFiles/run_rimage.dir/build.make:57: src/arch/xtensa/CMakeFiles/run_rimage] Error 134
make[2]: *** [CMakeFiles/Makefile2:2073: src/arch/xtensa/CMakeFiles/run_rimage.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:2189: src/arch/xtensa/CMakeFiles/bin.dir/rule] Error 2
make: *** [Makefile:736: bin] Error 2 

Is there any more explicit instruction I can find in the project or else where?

lgirdwood, 2020-09-25:
@ranj063 @marc-hb @juimonen can someone clarify.
@KaiChuan-Hsieh please note the crash is logged in the GH issue tracker for fixing.

marc-hb, 2020-09-25:
At the moment TGL requires the MEU_TOOL. I'm afraid the MEU_TOOL is available only under NDA, if correct please contact your Intel representative.

> Assertion `image.adsp->write_firmware' failed.

This is because rimage doesn't support signing TGL. I added this assert in https://github.com/thesofproject/rimage/commit/2156dcb00d2eb05a8 a while back because it was more "user-friendly" (ahem) than dereferencing a null or uninitialized pointer.

plbossart, 2020-09-25:
@KaiChuan-Hsieh I hope you are aware that even if you manage to sign the firmware yourself, the generated binary will not run on platforms designed for Windows since they use a 'production key' that Intel does not release externally. The signature tools provided with MEU will only work on Chromebooks and open-platforms, but for TGL I am afraid there are no platforms available commercially at the moment.

KaiChuan-Hsieh, 2020-09-26:
@plbossart @marc-hb thanks for your kindly reply.

I am running Ubuntu 20.04 on a TGL platform, and I am seeking a way for signing it with Community key as CNL platform. Is there any way to achieve this goal without signing NDA with Intel?

lgirdwood, 2020-09-26:
@KaiChuan-Hsieh do you work for OEM/ODM if so you should be able to update the OEM key manifest with your own key and sign with this ?

KaiChuan-Hsieh, 2020-09-26:
@lgirdwood in the project, there is otc_private_key_3k.pem, I thought it is the community key for signing the firmware, I don't have any OEM key in hand, since I am not working for OEM/ODM. Is there any command instruction like "rimage -o sof-cnl.ri -c /home/u/Workspace/sof/sof/sof/rimage/config/cnl.toml -k /home/u/Workspace/sof/sof/sof/keys/otc_private_key.pem -i 3 -f 1.6 -b 1 bootloader-cnl sof-cnl" for knowing how to create sof-tgl.ri, since I already try rimage, and it is not used by TGL.

lgirdwood, 2020-10-06:
@KaiChuan-Hsieh any update ? Btw your TGL platform must be programmed with an OEM signing key (that you have access to) in order for you to sign and execute the code (by default it will use the Intel signing key which means only Intel can sign).

KaiChuan-Hsieh, 2020-10-06:
@lgirdwood I still can't find a way to create sof-tgl.ri, but the sof-bin stable-v1.6 contain the rimage for TGL platform, so it is not urgent now. However, it is still helpful to have the rimage signed by the community key if someone would like to hacking the project.

lgirdwood, 2020-10-06:
@KaiChuan-Hsieh agreed, there is a FEATURE for rimage TGL signing. Not sure when it will be done though (unless you have time). OpenSSL does the signing and is invoked by rimage, the difference for TGL is a larger key size (384 bytes) vs key size of 256 bytes for the others.

KaiChuan-Hsieh, 2020-10-06:
@lgirdwood thanks for your hint. I'll find sometime to dig into the project, hope that I can contribute it. Link the issue to #2158, and close it.

closed 2020-10-06 by KaiChuan-Hsieh

marc-hb, 2020-10-06:
> there is a FEATURE for rimage TGL signing.

#1833