thesofproject/sof #3480: [BUG] Reduce footprint by putting large items under kconfig options

lgirdwood, 2020-09-29:
**Describe the bug**
Improve footprint by making more data and text items optional under kconfig. Platforms with limited memories must have a smaller set of features (e.g. less SRC conversion ratios, less module, less debug etc) compared to recent platforms with larger memories.

**To Reproduce**
```sh
for i in $( find build_*/sof ); do
	echo $i
	xtensa-apl-elf-nm -S --size-sort --reverse-sort $i | head -n 20
	echo
done
```
Low hanging fruit is 

- [ ]  SRC coefficients.
- [ ] Trig tables.
- [ ] primary core 
- [ ] Heap tables.
```
build_apl_gcc/sof
be0398f0 00002020 B primary_core_data
be00a680 000016ee T ipc_cmd
be024ca0 00001203 t dmic_set_config
be037688 00000aa8 b lp_buf_block
be022d98 00000973 t ssp_set_config
be032dd8 00000804 R sine_table
be030a18 00000800 r coeff48000to08000
be031998 00000800 r coeff24000to08000
be038130 00000800 b buf_block
be02e298 00000780 r coeff48000to32000
be02f698 00000780 r coeff48000to16000
be031218 00000780 r coeff24000to16000
be02db98 00000700 r coeff48000to44100
be0326d8 00000700 r coeff44100to48000
be010360 00000651 t volume_cmd
be02ea18 00000640 r coeff48000to24000
be02f058 00000640 r coeff48000to22050
be02fe18 00000600 r coeff48000to12000
be030418 00000600 r coeff48000to11025
be018200 000005e7 t dai_params

build_bdw_gcc/sof
004038b0 00003fa0 b buf_block
00408bf0 00003020 B primary_core_data
00001fd8 000017d0 T ipc_cmd
00401af8 00000804 R sine_table
00007e94 0000065d t volume_cmd
004082b0 00000600 b mod_block16
000084f4 000005fb T src_polyphase_stage_cir
0000fee8 000005fb t dai_params
00008af0 000005eb T src_polyphase_stage_cir_s16
000173dc 00000574 t dw_dma_set_config
0000d768 00000573 t tone_cmd
00016214 00000561 t ssp_set_config
000040e8 000004fa t host_params
00407eb0 00000400 b mod_block32
000064ac 000003f0 T comp_verify_params
00011544 000003c3 t kpb_event_handler
00400588 000003c0 R src_int16_24_25_1814_5000_fir
00011cd0 00000398 t kpb_draining_task
00011938 00000382 t kpb_copy
0001b4c4 00000382 T __divdi3

build_byt_gcc/sof
ff302d30 00003020 B primary_core_data
ff2c1f6c 000017e6 T ipc_cmd
ff30052c 00000804 R sine_table
ff2c7eec 0000065e t volume_cmd
ff2cd7a0 00000613 t ssp_set_config
ff2ca018 00000603 t dai_params
ff301f10 000005d0 b buf_block
[... 253 lines cut ...]
be044614 00000600 r coeff48000to11025
be01e5a8 000005e3 t dai_params

build_tgl_gcc/sof
be05ccfc 00004000 d buf_block
be064200 000021c0 B primary_core_data
be02f050 00001706 T ipc_cmd
be04a460 00001203 t dmic_set_config
be048328 00000973 t ssp_set_config
be058e44 00000804 R sine_table
be056a84 00000800 r coeff48000to08000
be057a04 00000800 r coeff24000to08000
be054304 00000780 r coeff48000to32000
be055704 00000780 r coeff48000to16000
be057284 00000780 r coeff24000to16000
be053c04 00000700 r coeff48000to44100
be058744 00000700 r coeff44100to48000
be034c74 00000651 t volume_cmd
be054a84 00000640 r coeff48000to24000
be0550c4 00000640 r coeff48000to22050
be0619bc 00000600 d sys_rt_x_block64
be055e84 00000600 r coeff48000to12000
be056484 00000600 r coeff48000to11025
be03cb98 000005e3 t dai_params
```

singalsu, 2020-09-29:
Yep. I can do the first two items: SRC and sine lookup tables.

lgirdwood, 2020-10-06:
@slawblauciak any comments on primary_core_data ?

slawblauciak, 2020-10-06:
@lgirdwood not sure, not exactly familiar with the static data FW embeds in the binary.

lgirdwood, 2020-10-06:
@slawblauciak thanks, it may need some Kconfig #idefs around some it's data members as it seems odd that on UP platforms it's 12k and on SMP platforms it's 8k ?

slawblauciak, 2020-10-06:
@lgirdwood I'll ask around. Somebody who worked on SMP code might know.

mmaka1, 2020-10-07:
@lgirdwood @slawblauciak primary_core_data size is not related to UP vs. SMP, it is just coincidence. The size is determined by number of interrupt levels selected in the platform configuration, as each context uses dedicated 4K stack.

lgirdwood, 2020-10-07:
@mmaka1 thanks got you. So these are already optimised.

lgirdwood, 2021-01-12:
@singalsu @johnylin76 @ShriramShastry @aiChaoSONG fyi - any chance we can move more of these under kconfig for v1.7 ?

singalsu, 2021-01-14:
Yes, I'll do the SRC coefficients and trig tables kconfig. For SRC I think the tiny conversions set (as for generic C) with only 16/32/44.1/48 but with the better quality used today should be a good default. So far there has been no need for the near full MPEG-AAC rates conversions matrix.

lgirdwood, 2021-01-14:
@singalsu thanks.

ShriramShastry, 2021-01-15:
I will optimize the  sine lookup table with smaller size and compute the accuracy of lookup table

singalsu, 2021-01-15:
@ShriramShastry Which table do you mean? The table in src/math/trig.c is there to provide about 100 dB SNR for any frequency 48 kHz sampled sine wave produced by the tone generator. I'd not reduce the quality if the sine is used for tone generator component. I was thinking to enable build of sine and the table only when it's part of build. Normally tone generator would not be part of build, unless needed for e.g. production tests.

singalsu, 2021-01-15:
@lgirdwood The listed SRC tables sizes are from CONFIG_COMP_SRC_TINY since it's for gcc build. The next shrink option would be to disable SRC. The tiny set only has a minimum set of convenrsions.

Below are grep results from defconfig files:

```
singalsu@singalsu-mobl:~/work/current/sof/sof/src/arch/xtensa/configs$ grep -r COMP_SRC
baytrail_defconfig:CONFIG_COMP_SRC=n
cherrytrail_gcc_defconfig:CONFIG_COMP_SRC=n
baytrail_gcc_defconfig:CONFIG_COMP_SRC=n
cherrytrail_defconfig:CONFIG_COMP_SRC=n
haswell_defconfig:CONFIG_COMP_SRC_TINY=y
apollolake_defconfig:CONFIG_COMP_SRC_SMALL=y
broadwell_defconfig:CONFIG_COMP_SRC_TINY=y

```

There's currently three different sizes: tiny, small, and std. I could change all platforms to small. Tiny not recommended because of a lot reduced quality. Small is like full but rates over 48 kHz are not supported. Would that be acceptable image size reduction contribution from SRC? I could also switch CONFIG_COMP_SRC=n for all platforms before cAVS. The file src_small_int32.m shows in comments the supported conversion matrix:

```
%             In
%             8 1 2 3 4 4 5
%               6 4 2 4 8 0        Out
fs_matrix = [ 0 1 1 1 0 1 0 ; ... %  8
	      1 0 1 1 0 1 0 ; ... % 16
	      1 1 0 1 0 1 0 ; ... % 24
	      1 1 1 0 0 1 0 ; ... % 32
	      0 0 0 0 0 1 0 ; ... % 44
	      1 1 1 1 1 0 1 ; ... % 48
	      0 0 0 0 0 1 0 ; ... % 50
	    ];

```

singalsu, 2021-01-18:
@dbaluta Do you have preferences for SRC in imx*_defconfig? If no feedback, I think I'll touch only Intel platforms with my SRC component remove/reduce PR to address this memory issue.

paulstelian97, 2021-01-18:
> @dbaluta Do you have preferences for SRC in imx*_defconfig? If no feedback, I think I'll touch only Intel platforms with my SRC component remove/reduce PR to address this memory issue.

For now we won't include it, but just in case plans change we can always update the defconfig as-needed.

lgirdwood, 2021-05-10:
@singalsu @ShriramShastry any update here, are all the maths tables now under Kconfig ?

singalsu, 2021-05-10:
@ShriramShastry We can add kconfig to cordic functions code and tables.

closed 2021-05-19 by lgirdwood