thesofproject/sof #3491: Document (overlapping) codenames for toolchains, platforms, Kconfigurations, signing schemes, etc.

marc-hb, 2020-10-03:
What do "TGL" or "CHT" mean? Are they Intel SoCs, PCH codenames, Kconfigurations, toolchain names, signing schemes? Probably a bit of everything depending on the context, no one knows for sure. No one except @jajanusz , who agreed in https://github.com/thesofproject/sof/pull/3451#issuecomment-700545311 that some documentation is really needed. Quoting myself from there:

> I can think of two places: either sof-docs or the top of src/arch/extensa/CMakeLists.txt but maybe there is a better place.

lgirdwood, 2020-10-03:
@marc-hb there is a platform description table in sof-docs - this info could be added in new columns. They are product names but also used  for fw and toolchain config.

marc-hb, 2020-10-05:
> They are product names but also used for fw and toolchain config.

These "accidental" 1:1:1 mappings seem pretty bad. Because they "accidentally" work until they don't: 
https://github.com/thesofproject/sof/pull/3451#issuecomment-703759584 

Short of renaming everything to tgl-toolchain, tgl-kconfig, etc., some documentation would really help mitigate.

marc-hb, 2020-10-13:
Thanks to https://github.com/thesofproject/sof-test/pull/428 for this useful "bookmark": https://github.com/thesofproject/linux/blob/topic/sof-dev/sound/soc/sof/sof-pci-dev.c

greg-intel, 2022-01-06:
These links helped me a little: 
[ChromeOS Lead Vehicle Programs and Details](https://wiki.ith.intel.com/display/ChromeOS/ChromeOS+Lead+Vehicle+Programs+and+Details) (Updated July 2021)
[ChromeOS Platform Decoder](https://wiki.ith.intel.com/display/ChromeOS/ChromeOS+Platform+Decoder) (Updated December 2021)

lgirdwood, 2022-01-06:
@greg-intel feel free to add links to these in sof-docs if not already stated.

kv2019i, 2025-01-03:
Lowered priority, but actually, let's close this. We have considerable updates in https://thesofproject.github.io/latest/getting_started/intel_debug/introduction.html and https://thesofproject.github.io/latest/platforms/index.html (and more in the works in pull-requests for sof-docs). Let's file more specific bugs if information is missing for a particular configuration.