thesofproject/sof #3492: [BUG][BDW] Stutter on bdw-rt5677 output

caseorum, 2020-10-04:
**Describe the bug**
All sound output stutters until any alsa utility or plugin is used. The stream sounds 50% interlaced with null at ~10hz. The stutter is corrected for any pulseaudio output that starts after any use of alsa (aplay, arec, audacity plugins...) exits.

The stutter is present if and only if `SND_SOC_SOF_DEBUG_ENABLE_FIRMWARE_TRACE=n` in the kernel, which makes debugging interesting. This could also be a driver bug.

**To Reproduce**
```
pulseaudio --start
pacat </dev/urandom #(or anything)
```

**Reproduction Rate**
100%

**Expected behavior**
Clear audio output

**Impact**
 Very annoying with only pulseaudio, irrelevant on alsa.

**Environment**
All FW, tplg, and kernel modules built-in.

1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: torvalds (22fbc037cd32e4e6771d2271b565806cfb8c134c), sof-dev (53b27a20cd5d45544c37b860003e527e01aa8300), sof-dev-rebase (ecda5934206be8d3ceb3eb644e4341a98cb31917)
    * SOF: 1.5.1 (88707ca87e782631fd2da11a89e7a89d6bd95763), master (878f15b34a514f834d18c0b7be2aa06b77e6005a)
2) Name of the topology file
    * Topology: sof-bdw-rt5677.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: bdw-rt5677

**Screenshots or console output**
All logs are identical, with the exception of FW logs, which cannot be produced (see above).

Pragmatic note: bdw-rt5677 is a rare platform, and the workaround of setting SND_SOC_SOF_DEBUG_ENABLE_FIRMWARE_TRACE=y is very easy, so this may not deserve any kind of priority.

lgirdwood, 2020-10-04:
Could be 2 things 
1) DMA position reporting latency where HW pointer being used by PA is too old (if HW pointer is being transfered via IPC rather than read from DSP memory directly). Should default to memory readback.
2) PA is writting to same period that DW DMA is reading. This can cause races with HW vs SW IO. Trace may change the timing here and make it work.

It does sound like PA is writting too close (or beyond) the HW pointer due to either 1 or 2.

plbossart, 2020-10-05:
@caseorum I've heard somewhat random noises at times but I have never been able to point to a specific root cause. I am also confused by the description "All sound output stutters until any alsa utility or plugin is used". PulseAudio will typically idle after 5s and close the streams, so is the problem happening on startup only when PulseAudio is first started?

Also what do you mean by 'alsa utility'. Alsa utils can use the :hw device, or the :pulse device. Do you mean to say that the sound is good only when PulseAudio is NOT used?

caseorum, 2020-10-05:
@plbossart Yes, I mean all pulseaudio output stutters until after anything from alsa closes `-Dhw:...`. `-Dpulse` stutters. I'll edit to clarify. PA's stream idle/wake changes nothing.

caseorum, 2020-10-05:
@lgirdwood That sounds right. This bug is with the low-latency pipeline, and using the media playback pipeline yields the same blip of silence separated by a longer interval of audio. That might be race-condition weirdness, because their buffers have similar size.

plbossart, 2020-10-05:
@caseorum don't use the media pipeline, it was never tested and likely is broken.
I still don't get the sequence that makes things work.

caseorum, 2020-10-05:
@plbossart You're right, the media pipeline fails spectacularly after several seconds, but that looks unrelated. Are you trying to reproduce this? How can I help? PA saves and restores device state, so that database could be affecting what you see on boot. Or maybe systemd, udev, or alsa are storing some kind of persistent state? I boot every time with a fresh install of PA with only default configs and no state from previous boots.

lgirdwood, 2020-10-06:
@caseorum the quickest thing to test is to get PA to render further back from the HW ptr. There may be a PA alsa-sink-module or alsa-source-module config setting for this. 
The topology could also be updated to specify a larger minimum period size and buffer size (on the host side - not the DSP pipeline buffer), alternatively PA config could be used to set buffer/period size.
Finally the DSP pipeline topology could also be updated to have 3 or more periods (instead of 2) on the pipeline "host" component. This is the component that connects to the host and does the data copying to/from host.

caseorum, 2020-10-06:
@lgirdwood I think PA's handling of hw_ptr isn't exposed via module parameters, and I was staying out of modifying PA itself for the time being. I did test ring buffer periods (module-alsa-card fragments). I'll get to topology changes in a bit.
PA defaults to 4 periods of 25ms:
```
D: [pulseaudio] alsa-util.c: Hardware PCM card 0 'sof-bdw rt5677' device 0 subdevice 0
D: [pulseaudio] alsa-util.c: Its setup is:
D: [pulseaudio] alsa-util.c:   stream       : PLAYBACK
D: [pulseaudio] alsa-util.c:   access       : MMAP_INTERLEAVED
D: [pulseaudio] alsa-util.c:   format       : S16_LE
D: [pulseaudio] alsa-util.c:   subformat    : STD
D: [pulseaudio] alsa-util.c:   channels     : 2
D: [pulseaudio] alsa-util.c:   rate         : 48000
D: [pulseaudio] alsa-util.c:   exact rate   : 48000 (48000/1)
D: [pulseaudio] alsa-util.c:   msbits       : 16
D: [pulseaudio] alsa-util.c:   buffer_size  : 4800
D: [pulseaudio] alsa-util.c:   period_size  : 1200
D: [pulseaudio] alsa-util.c:   period_time  : 25000
D: [pulseaudio] alsa-util.c:   tstamp_mode  : ENABLE
D: [pulseaudio] alsa-util.c:   tstamp_type  : MONOTONIC
D: [pulseaudio] alsa-util.c:   period_step  : 1
D: [pulseaudio] alsa-util.c:   avail_min    : 1200
D: [pulseaudio] alsa-util.c:   period_event : 1
D: [pulseaudio] alsa-util.c:   start_threshold  : -1
D: [pulseaudio] alsa-util.c:   stop_threshold   : 5404319552844595200
D: [pulseaudio] alsa-util.c:   silence_threshold: 0
D: [pulseaudio] alsa-util.c:   silence_size : 0
D: [pulseaudio] alsa-util.c:   boundary     : 5404319552844595200
D: [pulseaudio] alsa-util.c:   appl_ptr     : 0
D: [pulseaudio] alsa-util.c:   hw_ptr       : 0
```
(Aside: Those are special values of start_threshold, appl_ptr, and  hw_ptr, right?)

Playback stutters with 4 or 5 periods, and is happy with anything else, even 2 and 3. Here's the odd behavior:
set periods = 4 => stutter
set periods != 4, 5 => clean
set periods = 4 => clean
set periods = 5 => stutter
set periods = 4 => stutter
So it behaves like a "stutter bit" that is active at boot, set by 5 periods, read by 4 periods, and cleared by all other integers.

caseorum, 2020-10-06:
With trace enabled, it's bugged consistently at periods 5, 7(slightly), and 10, no bit-setting behavior. So these numbers are more or less meaningless and this looks like a race condition that is sensitive to timing in the FW and the period count in the ring buffer.

plbossart, 2020-10-06:
@caseorum I don't think we've tested with odd number of periods, I would stay away from this for now.
are you positive that with the 2 period settings things work fine, including with the trace disabled?

plbossart, 2020-10-06:
@caseorum Now that I think of it, this reminds me of https://github.com/thesofproject/linux/issues/2200.
That looks like what you are experiencing. I closed it thinking it was a Debian setup issue but maybe I was on to something.

What distro are you using if I want to reproduce your findings?

lgirdwood, 2020-10-06:
@caseorum can you try host periods at 3 i.e.
```diff
diff --git a/tools/topology/sof/pipe-volume-playback.m4 b/tools/topology/sof/pipe-volume-playback.m4
index cb7d2ad67..0cb84464d 100644
--- a/tools/topology/sof/pipe-volume-playback.m4
+++ b/tools/topology/sof/pipe-volume-playback.m4
@@ -44,7 +44,7 @@ W_DATA(DEF_PGA_CONF, DEF_PGA_TOKENS)
 
 # Host "Passthrough Playback" PCM
 # with 2 sink and 0 source periods
-W_PCM_PLAYBACK(PCM_ID, Passthrough Playback, 2, 0, SCHEDULE_CORE)
+W_PCM_PLAYBACK(PCM_ID, Passthrough Playback, 3, 0, SCHEDULE_CORE)
 
 # "Volume" has 2 source and x sink periods
 W_PGA(0, PIPELINE_FORMAT, DAI_PERIODS, 2, DEF_PGA_CONF, SCHEDULE_CORE,
```
and rebuild topology.

caseorum, 2020-10-06:
I've confirmed that the ring buffer period count is irrelevant and PA isn't involved. It's just buffer size and format.
```
Without trace:
speaker-test -Dhw:0,0 -c2 -S5 -r48000 -FS16_LE -b 1000000 -P4 # stutter for literally any value of -P !
speaker-test -Dhw:0,0 -c2 -S5 -r48000 -FS32_LE -b 1000000 -P4 # horrible popping, any -P
With trace:
speaker-test -Dhw:0,0 -c2 -S5 -r48000 -FS16_LE -b 1250000 -P4 # stutter, any -P
speaker-test -Dhw:0,0 -c2 -S5 -r48000 -FS32_LE -b 1250000 -P4 # stutter, any -P
```
So I think this really is a FW/driver bug. It appears that for any system, there is a buffer size that fails.

@plbossart Yes! I've run into that in testing, but I haven't yet isolated it to make sure it's the same bug. I bet so.
This is an arch linux userspace with alsa-* built from their git HEADs, if you'd like to try it. So only the alsa-* dependencies are really from arch. But my system is userspace-agnostic, so I can try another.

caseorum, 2020-10-06:
> can you try host periods at 3

@lgirdwood Done. No change.

plbossart, 2020-10-06:
@caseorum great, this is good progress.

Can you try with a buffer size that's a multiple of 192 bytes, e.g. -b 960000 ? I suspect limitations in our code. the values you used are not integer multiples for 48kHz sampling rate.

caseorum, 2020-10-06:
@plbossart -b is in microseconds, so `-b100000 -r48000 -FS16_LE` is the PA default 100ms buffer, so 19200 bytes.

plbossart, 2020-10-06:
@caseorum sorry, I read 1s. There's still got to be some magic value that's not well handled by the code, e.g. maybe it's an alignment with 1024 or something.

lgirdwood, 2020-10-06:
> I've confirmed that the ring buffer period count is irrelevant and PA isn't involved. It's just buffer size and format.

Fwiw, Buffer size and format do alter the page tables that the host driver sends to the FW. The page tables are used to program FW DMA to copy from host to DSP memories. We may need a topology PCM ALSA constraint here to make sure buffers are aligned with any HW constraints.

@caseorum do you have a list of good/bad buffers and formats or have you worked out the rules for what works and what doesn't ?

caseorum, 2020-10-06:
@plbossart That's my mistake, I had a typo with one too many 0s. Edited.

@lgirdwood I'll work up an exhaustive list of every error condition I can find tomorrow.

I had estimated the stutter at 10hz by ear. If exact, that would mean the error repeats once per buffer. I'll see if it's slower/faster for larger/smaller (broken) buffers.

caseorum, 2020-10-07:
Awesome! With s16le, the bug is usually "fragile" like that, so setting a
different buffer length makes it go away until reboot.

Things I've noticed:
Which buffers are bugged depends, at least, on sof kernel options,
especially trace. I'll try more. It's probably the length of those code
paths.

If a buffer length is bugged with 16 bits, it's worse with 32, even though
that doubles the buffer size.

All this implicates time much more than size. Maybe timer, scheduler,
interrupts....?

lgirdwood, 2020-10-07:
@plbossart @caseorum it's worth checking if the distortion is synchronous and constant on every test (for the same buffer size and format) i.e. occurs every x millisecs and also the duration in sample (capturing playback to another PC recording at 48kHz is good enough to show this). Audacity renders this very well when the audio is imported (especially if a sine wave is played).
There is also some debug in the FW and kernel to dump the page tables and buffer data. This should align kernel buffer with FW period size of 1ms (based on format).
Fwiw, the CI is passing with some combinations or format and buffer size so we may need to  add these as ALSA PCM constraints.

caseorum, 2020-10-07:
@lgirdwood I'll get a capture off the 3.5mm and try getting those dumps unless @plbossart beats me to it.

Honestly, I have yet to see a failure with a 50ms 2-period buffer on any format, so that's a real option.

caseorum, 2020-10-07:
@lgirdwood scratch that. I found a failure for S32_LE with 50ms buffer. Stutters at 20hz, so it looks like one error per buffer.

lgirdwood, 2020-10-07:
@caseorum the FW and driver code that does the buffer wrap is the same on CHT, BSW and BYT platforms. However the DMA IP on BDW is newer that the IP on CHT, BSW, BYT and hence may have a different configuration wrt DMA burst size, FIFO depth - but these are all quite small and should always align with buffer size. The other DMA difference is that HW reloads the DMA descriptors on BDW (instead of SW on BYT etc). The DMAC HW descriptors are a linked list that should point back to the start when the get to the end of the buffer (we could be miscalculating this).
Can you try
1) Several audio playbacks with large buffer followed by a silence playback (aplay -f dat -Dhw:0,0 /dev/zero). This may let us know if the DMA is reading data after the buffer (as it should not be silence 0s).
2) using a 50ms test buffer, enable trace on host.c of dw-dma.c to make sure the source pointer is incrementing correctly.

lgirdwood, 2020-10-07:
Btw, has anyone tried mmap mode with aplay yet ?

caseorum, 2020-10-07:
@lgirdwood 
- It's somewhat insane to produce an exhaustive list: Failing buffer sizes (currently without trace) are at least
 ...50ms s32le
 ...every (!) buffer between 61-84ms s16le
 ...every buffer 100-125ms both s16le and s32le
For every failing buffer size, exactly double, if possible, fails as well.

- Every error has a unique, reproducible sound for that length/format pair. The error always repeats once per buffer. Most errors are stutters that obey channel volume, some errors are harsh pops that are always in both channels at full volume, even when only one channel should be playing.

- mmap has no effect.

- The suggested several plays and then silence yield perfect silence.

caseorum, 2020-10-07:
Apologies. Slip of the finger on the close button.

plbossart, 2020-10-07:
I captured the output of the SSP0 on Samus, I don't see anything wrong and I can't hear any issues

[samus.csv.gz](https://github.com/thesofproject/sof/files/5343420/samus.csv.gz)

Really don't know where this noise comes from, for me it's not reproducible.

lgirdwood, 2020-10-07:
@caseorum thanks for the info. I've been having a look at src/audio/host.c and can see we have quite a few DMA variables that are used to do optimum copies between host and DSP memories. Are you able to rebuild FW with trace and debug and attach the trace log for working vs non working. I have a suspicion that the buffer last copy calculation is correct (since it works on other platforms) but it may be the copy size is too small or non aligned with the DMA IP on BDW.

plbossart, 2020-10-07:
I was able to completely wreck the device by choosing a size too small for the  buffer.

````
[ 1501.801666] sof-audio-acpi INT3438:00: error: ipc error for 0x60010000 size 20
[ 1501.801670] sof-audio-acpi INT3438:00: error: hw params ipc failed for stream 0
[ 1501.801673] sof-audio-acpi INT3438:00: ASoC: error at snd_soc_pcm_component_hw_params on INT3438:00: -5
[ 1501.801676]  Low Latency: ASoC: hw_params FE failed -5
[ 1508.119568] sof-audio-acpi INT3438:00: error: ipc error for 0x60010000 size 20
[ 1508.119572] sof-audio-acpi INT3438:00: error: hw params ipc failed for stream 0
[ 1508.119574] sof-audio-acpi INT3438:00: ASoC: error at snd_soc_pcm_component_hw_params on INT3438:00: -16
[ 1508.119575]  Low Latency: ASoC: hw_params FE failed -16
[ 1530.768610] sof-audio-acpi INT3438:00: error: ipc error for 0x60010000 size 20
[ 1530.768613] sof-audio-acpi INT3438:00: error: hw params ipc failed for stream 0
[ 1530.768616] sof-audio-acpi INT3438:00: ASoC: error at snd_soc_pcm_component_hw_params on INT3438:00: -16
[ 1530.768618]  Low Latency: ASoC: hw_params FE failed -16
[ 1555.761527] sof-audio-acpi INT3438:00: error: ipc error for 0x60010000 size 20
[ 1555.761530] sof-audio-acpi INT3438:00: error: hw params ipc failed for stream 0
[ 1555.761532] sof-audio-acpi INT3438:00: ASoC: error at snd_soc_pcm_component_hw_params on INT3438:00: -16
[ 1555.761534]  Low Latency: ASoC: hw_params FE failed -16
````

Probably a different issue though.

lgirdwood, 2020-10-07:
@plbossart I think we can probably set some ALSA constraints here, until this is root caised. The interesting thing is that host DMA may be in speculative mode where it always copies to fill the free bytes in the local buffer. This may work well until we hit buffer wrpa and have to do a smaller DMA copy than normal ?

plbossart, 2020-10-07:
@lgirdwood see https://github.com/thesofproject/sof/issues/3503 for the buffer size issue

@caseorum I am really not able to reproduce your findings, so you recompile the firmware yourself? I am using plain vanilla broadwell with GCC, trace included.

plbossart, 2020-10-07:
@caseorum I did retry with a clean install and I found that my aplay tests always works. When I do this sequence on reboot:

speaker-test -Dhw:1,0 -c2 -S5 -r48000 -FS16_LE -b 125000 -P4 # stutter
speaker-test -Dhw:1,0 -c2 -S5 -r48000 -FS16_LE -b 170000 -P2 # no stutter
speaker-test -Dhw:1,0 -c2 -S5 -r48000 -FS16_LE -b 125000 -P4 # no stutter

It's as if there's one good sequence that sets an internal state.

plbossart, 2020-10-07:
Hurray, now I am able to see visible discontinuities with an 8 Hz period.

root@chrx:~# aplay -Dhw:1,0 -c2 -r48000 -B125000 -fS16_LE cnt_1024_left.pcm  -v

[image]

if I remove the -B125000 the gaps are gone for good.

I have no idea what happens but that's at least one reproducible test case for me.

plbossart, 2020-10-07:
I just realized that the gaps happen at the start (or end) of the ring buffer. I knew 0.125 s look suspicious, but it's indeed what I set as the ring buffer size. So this clearly point to a host DMA issue, don't you think @lgirdwood ?

caseorum, 2020-10-07:
@lgirdwood Finally back at my samus. Here are two trace logs. Built with debug and verbose trace.
[50ms-s16le-clean.txt](https://github.com/thesofproject/sof/files/5344027/50ms-s16le-clean.txt)
[125ms-s16le-bug.txt](https://github.com/thesofproject/sof/files/5344028/125ms-s16le-bug.txt)

caseorum, 2020-10-07:
@plbossart
> I was able to completely wreck the device by choosing a size too small for the buffer.
Yeah, I got that one too.
> It's as if there's one good sequence that sets an internal state
Yes, setting a good buffer sets some kind of state. That's that "bit setting" I was talking about. There are also buffer lengths that break it again. Your results look like your kernel has sof tracing enabled, is that so?
> so you recompile the firmware yourself?
Yes. I've also tested releases, various build options, various kernel versions and configs.... I can reproduce this bug absolutely everywhere. The critical buffer lengths change.
> that's 125000, which is exactly the duration of the ring buffer
My ears were right!

caseorum, 2020-10-07:
@lgirdwood Do you still want some tracing added around the source pointer increments?

caseorum, 2020-10-07:
@plbossart Can you try s16le at 250ms? That should be an example of what I called "horrible popping" if I'm right about your setup. Would love to see a clean capture of that.

plbossart, 2020-10-07:
@caseorum here you go, perfect gaps at 4Hz / 250ms...

[image]

plbossart, 2020-10-07:
I think what happens is that the hw_params constraints are not set right.

````
root@chrx:~# aplay -Dhw:1,0 -c2 -r48000 -B125000 -fS16_LE cnt_1024_left.pcm  -v
Playing raw data 'cnt_1024_left.pcm' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 1 'sof-bdw rt5677' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 6000
  period_size  : 1488
  period_time  : 31000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 1488
  period_event : 0
  start_threshold  : 6000
  stop_threshold   : 6000
  silence_threshold: 0
  silence_size : 0
  boundary     : 6755399441055744000
  appl_ptr     : 0
  hw_ptr       : 0
````

I asked for 125 and got weird values.

I think what happens is this. with 125ms, the buffer size *should* be 24000 bytes = 192 * 125.

Now the period size also needs to be a multiple of 192 bytes, I think that's a fundamental limitation on Broadwell. It's not possible here since dividing in 4 periods will break the assumptions, so I guess the ALSA core tries to find a value that's not right.

My brain is a tad fried here but I think we've missing something like this:

````diff
diff --git a/sound/soc/sof/pcm.c b/sound/soc/sof/pcm.c
index cbac6f17c52f..ff5543689552 100644
--- a/sound/soc/sof/pcm.c
+++ b/sound/soc/sof/pcm.c
@@ -485,6 +485,9 @@ static int sof_pcm_open(struct snd_soc_component *component,
        snd_pcm_hw_constraint_step(substream->runtime, 0,
                                   SNDRV_PCM_HW_PARAM_PERIOD_BYTES,
                                   le32_to_cpu(caps->period_size_min));
+       snd_pcm_hw_constraint_integer(substream->runtime,
+                                     SNDRV_PCM_HW_PARAM_PERIODS);
+
 
        /* set runtime config */
        runtime->hw.info = ops->hw_info; /* platform-specific */
````

if I add this change, then the buffer size changes

````
aplay -Dhw:1,0 -c2 -r48000 -B125000 -fS16_LE cnt_1024_left.pcm  -v
Playing raw data 'cnt_1024_left.pcm' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 1 'sof-bdw rt5677' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 5952
  period_size  : 1488
  period_time  : 31000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 1488
  period_event : 0
  start_threshold  : 5952
  stop_threshold   : 5952
  silence_threshold: 0
  silence_size : 0
  boundary     : 6701356245527298048
  appl_ptr     : 0
  hw_ptr       : 0
````

that still makes no sense, the period and the buffer_size are not multiples of the period_size_min, so something's not right in the way the hw_constraints are handled.

plbossart, 2020-10-07:
Actually ALSA isn't so bad. In this case, 125ms could not be reached, so it found 124ms for the buffer with 4 periods of 31ms.

I wonder however if the firmware (and page tables) are able to handle an odd number of 1ms ticks and DMA transfers per period.

caseorum, 2020-10-08:
@plbossart There's an issue with the constraints, like minimum buffer size, but I don't think that's related. The ALSA ring buffers and periods are consistently sane. I also think I've ruled out odd-size errors, since booting a pretty minimal kernel without trace ([config.gz](https://github.com/thesofproject/sof/files/5346691/config.gz)) is bugged for some extremely sane sizes (19200 bytes, 2-periods even). Because there's something stateful going on, even CI-passing configs could hit this error after some amount of use.

That sensitivity to trace is important. Either the IPC message is causing different (better, but still bad) DMA config to be set, or it's the fact that trace is also using DMA. I bet it's the latter, and that @lgirdwood is on the money with the optimized DMA stuff, so that's where I'm turning my attention.

Unfortunately I can't rule out that this isn't just broadwell, since I don't have a later chromebook to torture. It's a sneaky bug.

lgirdwood, 2020-10-08:
@plbossart @caseorum fwiw, the last page (where we hear the distortion) is treated differently in the page tables
```c
/*
 * Parse the host page tables and create the audio DMA SG configuration
 * for host audio DMA buffer. This involves creating a dma_sg_elem for each
 * page table entry and adding each elem to a list in struct dma_sg_config.
 */
static int ipc_parse_page_descriptors(uint8_t *page_table,
				      struct sof_ipc_host_buffer *ring,
				      struct dma_sg_elem_array *elem_array,
				      uint32_t direction)
{
	int i;
	uint32_t idx;
	uint32_t phy_addr;
	struct dma_sg_elem *e;

	/* the ring size may be not multiple of the page size, the last
	 * page may be not full used. The used size should be in range
	 * of (ring->pages - 1, ring->pages] * PAGES.
	 */
	if ((ring->size <= HOST_PAGE_SIZE * (ring->pages - 1)) ||
	    (ring->size > HOST_PAGE_SIZE * ring->pages)) {
		/* error buffer size */
		tr_err(&ipc_tr, "ipc_parse_page_descriptors(): error buffer size");
		return -EINVAL;
	}

	elem_array->elems = rzalloc(SOF_MEM_ZONE_RUNTIME, 0, SOF_MEM_CAPS_RAM,
				    sizeof(struct dma_sg_elem) * ring->pages);
	if (!elem_array->elems) {
		tr_err(&ipc_tr, "ipc_parse_page_descriptors(): There is no heap free with this block size: %d",
		       sizeof(struct dma_sg_elem) * ring->pages);
		return -ENOMEM;
	}

	elem_array->count = ring->pages;

	for (i = 0; i < ring->pages; i++) {
		idx = (((i << 2) + i)) >> 1;
		phy_addr = page_table[idx] | (page_table[idx + 1] << 8)
				| (page_table[idx + 2] << 16);

		if (i & 0x1)
			phy_addr <<= 8;
		else
			phy_addr <<= 12;
		phy_addr &= 0xfffff000;

		e = elem_array->elems + i;

		if (direction == SOF_IPC_STREAM_PLAYBACK)
			e->src = phy_addr;
		else
			e->dest = phy_addr;

		/* the last page may be not full used */
		if (i == (ring->pages - 1))
			e->size = ring->size - HOST_PAGE_SIZE * i;
		else
			e->size = HOST_PAGE_SIZE;
	}

	return 0;
}
```
AFAIU this works on BYT, CHT (as they also use page tables), so I can only assume this is probably correct (but trying a buffer size of 4k multiples would prove it).

My suspicion is the DMA elem config for the buffer wrap (especially if DMA is running in speculative mode). The host DMA is using "one shot" mode
```c
static uint32_t host_dma_get_split(struct host_data *hd, uint32_t bytes)
{
	struct dma_sg_elem *local_elem = hd->config.elem_array.elems;
	uint32_t split_src = 0;
	uint32_t split_dst = 0;

	if (local_elem->src + bytes > hd->source->current_end)
		split_src = bytes -
			(hd->source->current_end - local_elem->src);

	if (local_elem->dest + bytes > hd->sink->current_end)
		split_dst = bytes -
			(hd->sink->current_end - local_elem->dest);

	/* get max split, so the current copy will be minimum */
	return MAX(split_src, split_dst);
}

/**
 * Calculates bytes to be copied in one shot mode.
 * @param dev Host component device.
 * @return Bytes to be copied.
 */
static uint32_t host_get_copy_bytes_one_shot(struct comp_dev *dev)
{
	struct host_data *hd = comp_get_drvdata(dev);
	struct dma_sg_elem *local_elem = hd->config.elem_array.elems;
	uint32_t copy_bytes = 0;
	uint32_t split_value;
	uint32_t flags = 0;

	buffer_lock(hd->local_buffer, &flags);

	/* calculate minimum size to copy */
	if (dev->direction == SOF_IPC_STREAM_PLAYBACK)
		copy_bytes = audio_stream_get_free_bytes(&hd->local_buffer->stream);
	else
		copy_bytes = audio_stream_get_avail_bytes(&hd->local_buffer->stream);

	buffer_unlock(hd->local_buffer, flags);

	/* copy_bytes should be aligned to minimum possible chunk of
	 * data to be copied by dma.
	 */
	copy_bytes = ALIGN_DOWN(copy_bytes, hd->dma_copy_align);

	split_value = host_dma_get_split(hd, copy_bytes);
	if (split_value)
		copy_bytes -= split_value;

	local_elem->size = copy_bytes;

	return copy_bytes;
}

/**
 * Performs copy operation for host component working in one shot mode.
 * It means DMA needs to be reconfigured after every transfer.
 * @param dev Host component device.
 * @return 0 if succeeded, error code otherwise.
 */
static int host_copy_one_shot(struct comp_dev *dev)
{
	struct host_data *hd = comp_get_drvdata(dev);
	uint32_t copy_bytes = 0;
	int ret = 0;

	comp_dbg(dev, "host_copy_one_shot()");

	copy_bytes = host_get_copy_bytes_one_shot(dev);
	if (!copy_bytes) {
		comp_info(dev, "host_copy_one_shot(): no bytes to copy");
		return ret;
	}

	/* reconfigure transfer */
	ret = dma_set_config(hd->chan, &hd->config);
	if (ret < 0) {
		comp_err(dev, "host_copy_one_shot(): dma_set_config() failed, ret = %u", ret);
		return ret;
	}

	ret = dma_copy(hd->chan, copy_bytes, DMA_COPY_ONE_SHOT);
	if (ret < 0) {
		comp_err(dev, "host_copy_one_shot(): dma_copy() failed, ret = %u", ret);
		return ret;
	}

	return ret;
}
```
This means the number of bytes copied is variable depending on free space in the destination buffer and the buffer wrap (handled in the split case above) is possible too small a transfer for the BDW DMA. It's worth adding some debug trace here and print out the copy size when it's good and bad.

plbossart, 2020-10-08:
@caseorum I'll retest on CHT, I have test devices and I am afraid we haven't done a good job at changing the configuration of the ring buffer. Murphy's Law still applies.

plbossart, 2020-10-08:
I can confirm this bug does not occur on BYT/CHT. I tested with all sequences that are reproducibly wrong on Broadwell and no issues at all.

Clearly the Broadwell DMA has some issues.

caseorum, 2020-10-08:
Right. So the thought is that this and  #3503 have the same root cause, namely that the BDW DMA minimum is larger than 4k. Datasheet time.

lgirdwood, 2020-10-08:
@caseorum @plbossart I suspect that the smallest transfer size is limited to the channel burst size. It would be quick to try gradually small buffer (in steps of 4kB) until it fails and then workout the page table size and ultimately transfer size that breaks.

lgirdwood, 2020-10-08:
Oh, furthermore if smallest transfer size is say 16 bytes then I would further assume that all transfers must be aligned on 16 bytes for size.

caseorum, 2020-10-08:
@lgirdwood Got it. This does mean splitting out the logic for broadwell. Or would you restrict all of them, relatively harmlessly?
I don't dip into the audio world very often, but I vaguely remember a discussion about broadwell's odd DMA situation somewhere. Maybe kernel. I'll look.

lgirdwood, 2020-10-08:
@caseorum I think in the case for BDW we can set some attributes for DMA transfer size and alignment. We alreday have similar attributes so adding extra should not be too hard.

caseorum, 2020-12-01:
@lgirdwood @plbossart I've finally been able to look at this again today. Experimentally, the BDW minimum transfer is 4 bytes. This and  #3503 are caused by the ptable DMA transfer size being too small by up to about 2.9 bytes, so the ptable wasn't always copied in full. Are you sure this doesn't affect other ptable platforms? Maybe  #3503 doesn't affect them because they have smaller minimum copies, but this error should get them all.

I've submitted a PR that works on broadwell (#3659).

plbossart, 2020-12-01:
Impressive work @caseorum. 

I must admit I never understood the concept of page table, and the arithmetic used both on driver https://github.com/thesofproject/linux/blob/332d9bf12a58fe890e5ce47ce22e5af89a7699d8/sound/soc/sof/utils.c#L133 and firmware sides.

@lgirdwood and @keyonjie are probably the best people to comment on this.

lgirdwood, 2020-12-02:
@caseorum good work - fwiw, the Designware DMA IP has a lot of config options and it's likely on BDW the minimum transfer is 4 bytes (which may be different on other IP that uses this DMAC).

caseorum, 2020-12-02:
Thank you @lgirdwood and @plbossart. I'm glad I went bug hunting.

PR #3659 (2020-12-03): ipc: Fix page table DMA transfer size
This corrects the minimum transfer size calculation for the ptable and uses the relevant DMA attribute.

Fixes #3492, fixes  #3503
closed 2020-12-03 by lgirdwood