thesofproject/sof #3495: [BUG] Dell Precision 5750 doesn't work with stable-v1.6 sof rimage and tplg

KaiChuan-Hsieh, 2020-10-06:
Hello,

I try to update the sof-cnl.ri and sof-cml-rt711-rt1308-rt715.tplg from sof-bin branch stable-v1.6, but the audio doesn't work.
The sof-soundwire sound card is created, but I can't play to the designated playback device.
I also try to update to the kernel built from topic/sof-dev, the result is the same.

The OS I am using is Ubuntu 20.04

The log of alsa-info and journalctl are attached.
[precision-5750.tar.gz](https://github.com/thesofproject/sof/files/5331665/precision-5750.tar.gz)

By the way, the previous v1.5.1 sof-cnl.ri and sof-cml-rt711-rt1308-rt715.tplg seems okay.

lgirdwood, 2020-10-06:
@mengdonglin @keyonjie fyi

lgirdwood, 2020-10-06:
Driver logs
```
grep -rn "sof-" journalctl-b.log 
1280: 十  06 13:41:01 u-Precision kernel: sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
1281: 十  06 13:41:01 u-Precision kernel: sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
1286: 十  06 13:41:01 u-Precision kernel: sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
1288: 十  06 13:41:01 u-Precision kernel: sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
1289: 十  06 13:41:01 u-Precision kernel: sof-audio-pci 0000:00:1f.3: use msi interrupt mode
1305: 十  06 13:41:01 u-Precision kernel: sof-audio-pci 0000:00:1f.3: hda codecs found, mask 4
1330: 十  06 13:41:01 u-Precision kernel: sof-audio-pci 0000:00:1f.3: Firmware info: version 1:5:1-88707
1331: 十  06 13:41:01 u-Precision kernel: sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:16:0 Kernel ABI 3:17:0
1332: 十  06 13:41:01 u-Precision kernel: sof-audio-pci 0000:00:1f.3: Topology: ABI 3:17:0 Kernel ABI 3:17:0
1333: 十  06 13:41:01 u-Precision kernel: sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
2225: 十  06 13:41:01 u-Precision kernel: input: sof-soundwire Headset Jack as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input21
2226: 十  06 13:41:01 u-Precision kernel: input: sof-soundwire HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input25
2227: 十  06 13:41:01 u-Precision kernel: input: sof-soundwire HDMI/DP,pcm=6 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input26
2228: 十  06 13:41:01 u-Precision kernel: input: sof-soundwire HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input27
2255: 十  06 13:41:01 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: ipc error for 0x50010000 size 108
2256: 十  06 13:41:01 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: set/get ctrl ipc comp 1
2258: 十  06 13:41:01 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: ipc error for 0x50010000 size 108
2259: 十  06 13:41:01 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: set/get ctrl ipc comp 32
2260: 十  06 13:41:01 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: ipc error for 0x50010000 size 108
2261: 十  06 13:41:01 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: set/get ctrl ipc comp 38
2262: 十  06 13:41:01 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: ipc error for 0x50010000 size 108
2263: 十  06 13:41:01 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: set/get ctrl ipc comp 44
2493: 十  06 13:41:02 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: ipc error for 0x50010000 size 108
2494: 十  06 13:41:02 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: set/get ctrl ipc comp 1
2495: 十  06 13:41:02 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: ipc error for 0x50010000 size 108
2496: 十  06 13:41:02 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: set/get ctrl ipc comp 32
2497: 十  06 13:41:02 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: ipc error for 0x50010000 size 108
2498: 十  06 13:41:02 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: set/get ctrl ipc comp 38
2499: 十  06 13:41:02 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: ipc error for 0x50010000 size 108
2500: 十  06 13:41:02 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: set/get ctrl ipc comp 44
3258: 十  06 13:41:07 u-Precision /usr/lib/gdm3/gdm-x-session[2245]: (II) config/udev: Adding input device sof-soundwire Headset Jack (/dev/input/event10)
3259: 十  06 13:41:07 u-Precision /usr/lib/gdm3/gdm-x-session[2245]: (**) sof-soundwire Headset Jack: Applying InputClass "libinput keyboard catchall"
3260: 十  06 13:41:07 u-Precision /usr/lib/gdm3/gdm-x-session[2245]: (II) Using input driver 'libinput' for 'sof-soundwire Headset Jack'
3262: 十  06 13:41:07 u-Precision /usr/lib/gdm3/gdm-x-session[2245]: (**) sof-soundwire Headset Jack: always reports core events
3265: 十  06 13:41:07 u-Precision /usr/lib/gdm3/gdm-x-session[2245]: (II) event10 - sof-soundwire Headset Jack: is tagged by udev as: Keyboard Switch
3266: 十  06 13:41:07 u-Precision /usr/lib/gdm3/gdm-x-session[2245]: (II) event10 - sof-soundwire Headset Jack: device is a keyboard
3267: 十  06 13:41:07 u-Precision /usr/lib/gdm3/gdm-x-session[2245]: (II) event10 - sof-soundwire Headset Jack: device removed
3269: 十  06 13:41:07 u-Precision /usr/lib/gdm3/gdm-x-session[2245]: (II) XINPUT: Adding extended input device "sof-soundwire Headset Jack" (type: KEYBOARD, id 14)
3272: 十  06 13:41:07 u-Precision /usr/lib/gdm3/gdm-x-session[2245]: (II) event10 - sof-soundwire Headset Jack: is tagged by udev as: Keyboard Switch
3273: 十  06 13:41:07 u-Precision /usr/lib/gdm3/gdm-x-session[2245]: (II) event10 - sof-soundwire Headset Jack: device is a keyboard
3274: 十  06 13:41:07 u-Precision /usr/lib/gdm3/gdm-x-session[2245]: (II) config/udev: Adding input device sof-soundwire HDMI/DP,pcm=5 (/dev/input/event14)
3277: 十  06 13:41:07 u-Precision /usr/lib/gdm3/gdm-x-session[2245]: (II) config/udev: Adding input device sof-soundwire HDMI/DP,pcm=6 (/dev/input/event15)
3280: 十  06 13:41:07 u-Precision /usr/lib/gdm3/gdm-x-session[2245]: (II) config/udev: Adding input device sof-soundwire HDMI/DP,pcm=7 (/dev/input/event16)
3743: 十  06 13:41:50 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: ipc error for 0x50010000 size 108
3744: 十  06 13:41:50 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: set/get ctrl ipc comp 44
3745: 十  06 13:41:50 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: failed kcontrol value set for widget: 44
3746: 十  06 13:41:50 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: restoring kcontrols after resume
3747: 十  06 13:41:50 u-Precision kernel: sof-audio-pci 0000:00:1f.3: error: failed to restore pipeline after resume -22
```

lgirdwood, 2020-10-06:
@slawblauciak fyi - there is also and IPC error here, maybe setting invalid or old kcontrol data
@KaiChuan-Hsieh - it could be possible that old alsactl or pusleaudio or UCM configs are being used to set incorrect values for some of the kcontrols (as seen in kernel logs). Can you check any config files.

juimonen, 2020-10-06:
@lgirdwood the log says it is still 1.5.1 firmware...?

lgirdwood, 2020-10-06:
@KaiChuan-Hsieh can you confirm this is v1.6, @juimonen can you confirm the version number is not a regression (like the compiler name issue reported by @lyakh ).

KaiChuan-Hsieh, 2020-10-06:
Ah sorry, it seems I am using the wrong rimage(v1.5.1) with v1.6 tplg file.
After double confirm it, the combination is okay with the latest sof-dev kernel.

[3494.tar.gz](https://github.com/thesofproject/sof/files/5333190/3494.tar.gz)

Sorry for not checking carefully.

closed 2020-10-06 by KaiChuan-Hsieh

lgirdwood, 2020-10-06:
@KaiChuan-Hsieh no worries, this is useful info to say that v1.6 works fine on this platform !

plbossart, 2020-10-06:
@KaiChuan-Hsieh can you confirm that audio works fine, including for short notifications? we have another user report https://github.com/thesofproject/linux/issues/2479 where audio distortion is reported on Dell Precision 5750 and Dell 9700 (both with the same hardware with two amps)

@ruffson @maaarghk FYI

maaarghk, 2020-10-06:
@KaiChuan-Hsieh Since updating to the latest kernel - topic/sof-dev - I can confirm that I still have the issue where notifications are occasionally played with distortion in the left channel, which I never had on 1.5.1 firmware. What kernel version are you on? It could be a combination of latest kernel + latest firmware, for example that causes the issue.

plbossart, 2020-10-06:
@maaarghk now I am confused, in https://github.com/thesofproject/linux/issues/2479#issue-711401972 you mentioned that you had DSP panics with 1.5.1 so it's not clear to me if this can be considered as a solid base to triangulate the audio quality issue.

maaarghk, 2020-10-06:
hmm, I had DSP panics but it was after 3 or 4 days of uptime. This intermittent issue is multiple times per day.

lgirdwood, 2020-10-06:
@maaarghk could you open a new issue for intermittent distortion on v1.6. Easier for us to track to make sure it's fixed for v.16 final.

maaarghk, 2020-10-06:
Sure. Do you think https://github.com/thesofproject/linux/issues/2479 should be closed as it is an issue with 1.5.1 and seems fixed in 1.6? Or would you intend to backport to 1.5.2?

KaiChuan-Hsieh, 2020-10-07:
@maaarghk 
I remembered that I encounter the audio distortion on left channel of internal speaker once, but I didn't pay attention to it. Currently, I am working on our linux-oem kernel, https://code.launchpad.net/~vicamo/+git/ubuntu-kernel/+ref/bug-1894184/tgl-soundwire-support/oem-5.6 by adding 
[0001-ASoC-Intel-Correct-codec-name-prefix-to-ACPI-machine.zip](https://github.com/thesofproject/sof/files/5337909/0001-ASoC-Intel-Correct-codec-name-prefix-to-ACPI-machine.zip)
with v1.6 sof-cnl.ri + sof-cml-rt711-rt1308-rt715.tplg. I'll try to see if I can reproduce it with this kernel, and confirm that the latest sof-dev kernel can see the distortion issue.

maaarghk, 2020-10-07:
Hi, you could take a look at #3499