thesofproject/sof #3498: [FEATURE] Document how to build SOF with Zephyr

yongzhi1, 2020-10-06:
Since SOF Zephyr upstream is close to complete, the time is ready to publish the build instructions so that people can build SOF FW based on stable_v1.6 branch with Zephyr RTOS on master branch.

lgirdwood, 2020-10-06:
@lyakh fyi

lyakh, 2020-10-08:
> @lyakh fyi

@lgirdwood do I understand it correctly that we first need Zephyr upstream to add SOF as a module, which would make it possible to build firmware and, respectively, to document the process?

lgirdwood, 2020-10-08:
@lyakh we should state that in the intro and that's planned for Zephyr 2.5. The intro should state that these instructions are temporary and will change as the upstreaming completes. These instruction should show how to build it **today**
We can always update to module when @andyross and @nashif have everything ready on the Zephyr side.

lyakh, 2020-10-08:
> @lyakh we should state that in the intro and that's planned for Zephyr 2.5. The intro should state that these instructions are temporary and will change as the upstreaming completes. These [...]

@lgirdwood today we can build Zephyr with SOF in one of (at least) 3 ways: using as a basis SOF's clone, or the clone by @andyross or mine. The first option has the advantage of being at least a project repository, not a private one, but it's outdated. The latter two options should produce the same result (once @andyross merges my patches) but they use private trees. We could maybe update SOF's zephyr clone?

lgirdwood, 2020-10-08:
Yes, please update SOF Zephyr clone to match @andyross branch with your fixes too.

lgirdwood, 2020-10-08:
> Yes, please update SOF Zephyr clone to match @andyross branch with your fixes too.

and use this for the build instructions as it's probably closest to what will merge in Zephyr master.

marc-hb, 2020-10-13:
> the time is ready to publish the build instructions

Where?

How about https://github.com/thesofproject/sof/wiki ?

> The intro should state that these instructions are temporary and will change as the upstreaming completes

lgirdwood, 2020-10-13:
> > the time is ready to publish the build instructions
> 
> Where?
> 
> How about https://github.com/thesofproject/sof/wiki ?

Please add to sof-docs

marc-hb, 2020-10-14:
sof-docs is good for slow-changing documentation (and even for this it tends to be out of date)

lgirdwood, 2020-10-14:
sof-docs is for all docs today as it's checked by technical writers. If any parts are out of date then they can be updated or marked as bug and assigned for update.

yongzhi1, 2021-02-08:
Close for lowered priority.

lgirdwood, 2021-02-08:
@lyakh @yongzhi1 reopen as we need to add more context to existing instructions.

iuliana-prodan, 2021-02-09:
@lgirdwood @lyakh @yongzhi1 where/what are the existing instructions?

I want to build SOF with Zephyr and I saw this README: https://github.com/thesofproject/sof/blob/master/zephyr/README
Are the repos and branches mentioned there still the ones to use?
Otherwise, which are the latest repos/branches for SOF and Zephyr?

Thanks!

lgirdwood, 2021-02-09:
> @lgirdwood @lyakh @yongzhi1 where/what are the existing instructions?
>
> I want to build SOF with Zephyr and I saw this README: https://github.com/thesofproject/sof/blob/master/zephyr/README
> Are the [...]

Latest PR for sof docs is here 

https://github.com/thesofproject/sof-docs/pull/330#pullrequestreview-586525659

This has not been tested for i.MX yet, so you will need help for iMX SOC support from Zephyr team. @dbaluta is in the loop.

> Thanks!

aiChaoSONG, 2022-08-30:
@lgirdwood I think we have nice document here: https://thesofproject.github.io/latest/getting_started/build-guide/build-with-zephyr.html, can we close?

marc-hb, 2022-08-30:
https://thesofproject.github.io/latest/getting_started/build-guide/build-with-zephyr.html has been available for a long time and was just updated again. File more specific issues if/when needed, closing this super generic ask which was done a long time ago. Thanks @aiChaoSONG for the scrubbing help.

closed 2022-08-30 by marc-hb