thesofproject/sof #3499: [BUG] Intermittent audio distortion on Dell 9700 / sof-cml-rt711-rt1308-rt715.tplg / firmware 1.6

maaarghk, 2020-10-06:
Intermittent audio distortion is experienced on firmware 1.6. 

Videos of the issue:
* https://photos.app.goo.gl/jTZDGzPGSP8QckDf7
* https://photos.app.goo.gl/TyoTGeJuxsYAumtf9

the relevant outputs in alsamixer (distortion goes away when muted) are rt1308-1 DAC L and rt1308-1 DAC R - if you're unfamiliar with the hardware, as a reminder, these are both for the left channel despite the name but one seems to be doing a woofer and another a tweeter.

I used 1.5.1 for a long time and never experienced this issue.

**To Reproduce**
The issue occurs when the pcm stream is first opened and continues until it is closed. Stopping all sound until the stream closes and then starting up again is enough to stop the issue.

**Reproduction Rate**
I can't see anything in the logs which indicates the issue occurred but I would estimate to be 1/30 or thereabouts. It seems to be more common with notification sounds than, for example, music or a video playing, but I couldn't say for sure, because of course I just have more notifications going off than starting videos like that.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/96b9d7cf347158efac3d830d9937e8efc39579c8
    * SOF: built files from https://github.com/thesofproject/sof-bin/tree/stable-v1.6
2) Name of the topology file
    * Topology: `intel/sof-tplg/sof-cml-rt711-rt1308-rt715.tplg`
3) Name of the platform(s) on which the bug is observed.
    * Platform: Dell 9700  - 0x098F 

**Screenshots or console output**
dmesg and soflog whilst I played a notification sound on repeat roughly 100 times.

[soflog-20201006-2126-dmesg.log](https://github.com/thesofproject/sof/files/5336726/soflog-20201006-2126-dmesg.log)
[soflog-20201006-2126-soflog.log](https://github.com/thesofproject/sof/files/5336727/soflog-20201006-2126-soflog.log)

Audio distortion was heard around the following timestamps
* dmesg 98395.48 / soflog 70109...
* dmesg 98506.40 / soflog 70292...
* dmesg 98606.48 / soflog 70482...
* dmesg 98727.29 / soflog 70711...
* dmesg 98780.62 / soflog 70806..

The rate of occurrence is therefore slightly above 5%.

plbossart, 2020-10-06:
@maaarghk you only made references to rt1308-1 and the video seems to hint at a problem on the left speaker. Can you confirm if you hear this distorsion on the right speakers as well?

This is important for us, what happens is that the firmware will push *exactly* the same content to both amplifiers which are located on separate SoundWire links. rt1308-1 then picks the left channels and rt1308-2 pick the right channel only.

maaarghk, 2020-10-06:
I can confirm for sure it is on left speaker only.

plbossart, 2020-10-06:
@maaarghk thanks, this is super-useful information. I will try to cook a patch that ignores the right amplifier completely and uses the 'other' topology for Dell devices, where there is only a single amp. If you still hear the distorsion, it might be your hardware. if you don't then it's likely a problem in the firmware demux functionality.

plbossart, 2020-10-06:
@maaarghk can you try this patch on top of your kernel: 
[soundwire-left-amp.diff.txt](https://github.com/thesofproject/sof/files/5337173/soundwire-left-amp.diff.txt)

This should disable the right amp entirely as well as the demux in the firmware. You will need a different topology (-mono suffix) and a different UCM file will be loaded, but that should be fine if you have the latest stuff installed.

if you can confirm before and after applying the patch the results of amixer -Dhw:0 info that will help check that you've indeed move from 2 amps to one.

Thanks for your patience and help in tracking down this issue.

maaarghk, 2020-10-06:
Hi Pierre, I retried the same test of 100 notification sounds with your patch applied. As expected, audio from left speaker only. There was no distortion present. 

[soflog-20201006-2340-soflog.log](https://github.com/thesofproject/sof/files/5337314/soflog-20201006-2340-soflog.log)
[soflog-20201006-2340-dmesg.log](https://github.com/thesofproject/sof/files/5337316/soflog-20201006-2340-dmesg.log)
[amixer-after-patch.txt](https://github.com/thesofproject/sof/files/5337317/amixer-after-patch.txt)
[amixer-before-patch.txt](https://github.com/thesofproject/sof/files/5337318/amixer-before-patch.txt)

lgirdwood, 2020-10-07:
@KaiChuan-Hsieh fyi.

plbossart, 2020-10-07:
@maaarghk thank you for your tests. This rules out a broken speaker or hardware and points to the firmware demux in my opinion. That's really the only difference with the single amp mode, the driver uses the same sequence in single and multi-link cases.

@bardliao @slawblauciak @RanderWang do you concur with my analysis?

lgirdwood, 2020-10-07:
@juimonen btw are we able to validate demux routing logic in test bench yet ? Would be good to have such tests in CI

maaarghk, 2020-10-07:
Hi @juimonen  - is the "soflog.log" file I attached not what you need? I am happy to do whatever other traces you need, but might need talked through it in a bit of detail because I have no familiarity with alsa nevermind sof or soundwire.

juimonen, 2020-10-07:
@maaarghk yeah, sorry I should read more carefully. I'm not sure can we set the firmware debug level to be more verbose, we should see inside the mux_params and others what it is actually setting. I remember these kind of issues happening when for some reason the demux parameter forwarding logic goes wrong, so some audio branch thinks it using 24 bits etc. even though it is not.

maaarghk, 2020-10-07:
If you want to email me some firmware build with extra debug info feel free.

juimonen, 2020-10-08:
@maaarghk really appreciate your help, but I'm not sure you can install unsigned firmware (not signed with intel key) to your device, we might need to debug this with unlocked device internally. The official signing process is quite tedious, so we use it for official releases, then develop with devices with debug keys etc.

juimonen, 2020-10-08:
@mengdonglin do we have this kind of device that someone could take a look? we should enable more debugs in firmware and try to replicate the distortion, then check that there are correct params in demux for both channels, or that the pga before is not distorting the other channel.

bardliao, 2020-10-09:
> @mengdonglin do we have this kind of device that someone could take a look? we should enable more debugs in firmware and try to replicate the distortion, then check that there are correct params in demux for both channels, or that the pga before is not distorting the other channel.

I can reproduce this issue on my CML-H laptop.

plbossart, 2020-10-09:
welcome back @bardliao. Can you confirm if the issue happens only with 1.6 and does not happen with 1.5.1?

bardliao, 2020-10-10:
> welcome back @bardliao. Can you confirm if the issue happens only with 1.6 and does not happen with 1.5.1?

Yes. I tried a 50 iterations test script and didn't meet the issue with 1.5.1 and I can see the issue with the same script with 1.6.
Here is my test script.
<details>
<summary>test.sh</summary>

```
#!/bin/bash

for i in {1..50}
do
        aplay -Dhw:0,2 wav/LR_48k.wav -d 2 -vv;
        sleep 10;
done
```
</details>

slawblauciak, 2020-10-28:
We were unable to reproduce this issue with a slim driver, no-codec setup. @jdrzaszcz replicated the topology in the tests.
Furthermore, I extracted the audio signal from the pipeline and I was able to confirm that this is not a demux issue, the audio is clear.
I noticed that changing the downmix channel configuration in alsamixer for each codec alters the way distortion occurs, it makes it a bit random as to which channel is distorted.
Having all those three points in mind I think there might be something funky happening with the codecs.
On the other hand, the fact that it's not reproduced on 1.5.1 is concerning and would imply otherwise.

plbossart, 2020-10-28:
@slawblauciak thanks for all these experiments. It's disappointing that the root-cause wasn't found but nobody assumed it'd be easy.

Can you clarify what you mean by a 'nocodec' setup?

I am starting to wonder if we should focus on the distortion itself to figure out if the data is corrupted or just offset.

Maybe we can use a 'snooping' capture channel where we just record with one sink port what is transmitted by a source port. We only need to use the same parameters for the ports.

slawblauciak, 2020-10-30:
After further investigation and support from @bardliao we were able to determine that it might actually be a firmware/demux issue, contrary to my previous statement.
After revert of https://github.com/thesofproject/sof/commit/5d763d63499d67ec482b35bb66f56410df26f58e the issue goes away. 
It's very strange how it doesn't show up with our slimdriver tests and probes.

@plbossart by no-codec I mean doing loopback directly on ALH.

mwasko, 2020-10-30:
> After revert of [5d763d6](https://github.com/thesofproject/sof/commit/5d763d63499d67ec482b35bb66f56410df26f58e) the issue goes away.

@slawblauciak please also take a closer look on routines that calculate available data. As discussed the reverted patch was an optimization that check available data only once and then consume until reach the end, previously we did the available data check regularly. 

However that does not explain why we do not see reproduction with probes.

plbossart, 2020-10-30:
> After further investigation and support from @bardliao we were able to determine that it might actually be a firmware/demux issue, contrary to my previous statement.
> After revert of [5d763d6](https://github.com/thesofproject/sof/commit/5d763d63499d67ec482b35bb66f56410df26f58e) the issue goes away.

Nice progress, thanks @bardliao and @slawblauciak.
 
> @plbossart by no-codec I mean doing loopback directly on ALH.

ok, it's a valid test, but I was talking about snooping data on the bus itself. I think we could add this capability in the Linux driver, there's a know limitation that we cannot start two streams at the same time, so we'd need to have this snooping capability in the same stream, but a different ALH port. Need a lot more coffee.

libinyang, 2020-11-03:
As Liam required yesterday, I tested the 3-period buffer in tplg. There is not distortion now.

The tough patch is like below:

```
diff --git a/rimage b/rimage
index fccf52133..9f3f2d7d7 160000
--- a/rimage
+++ b/rimage
@@ -1 +1 @@
-Subproject commit fccf52133104987d9a532e6a9c332ba5c7a1bd91
+Subproject commit 9f3f2d7d714908c1f8f3f77baebe9175682f02bc
diff --git a/tools/topology/m4/pipeline.m4 b/tools/topology/m4/pipeline.m4
index 37c5ca8d8..671b491ff 100644
--- a/tools/topology/m4/pipeline.m4
+++ b/tools/topology/m4/pipeline.m4
@@ -5,7 +5,7 @@ define(`SCHEDULE_TIME_DOMAIN_DMA', 0)
 define(`SCHEDULE_TIME_DOMAIN_TIMER', 1)

 # default number of DAI periods
-define(`DAI_DEFAULT_PERIODS', 2)
+define(`DAI_DEFAULT_PERIODS', 3)

 dnl Define macro for pipeline widget
```

slawblauciak, 2020-11-03:
We've finally managed to reproduce it on RVP/no-codec. Turns out, it requires a specific demux configuration of 8 channels. We were only testing a 2ch demux configuration.

plbossart, 2020-11-03:
> As Liam required yesterday, I tested the 3-period buffer in tplg. There is not distortion now.
>
> The tough patch is like below:
>
> ```
> diff --git a/rimage b/rimage
> index fccf52133..9f3f2d7d7 160000
> --- [...]

why would we do this? This looks like masking an error. We have way too much buffering in SOF IMHO.

juimonen, 2020-11-03:
@plbossart I think we found the cause for this... there's something fishy in the configuration setting in demux. So full 8x8 matrix with diagonal 1's makes the distortion and if only first 2x2 square is "diagonalized" (rest is 0) -> audio is clean. So somehow the current implementation is doing some extra copies with full diagonal matrix, it thinks it has 8 inputs and outputs. So we should discover the channel count at runtime, not from matrix. I think we also removed the channel count from demux config struct while ago...

juimonen, 2020-11-03:
@plbossart so you could "fix" this by zeroing all other entries outside 2x2 in topology matrices...

bkokoszx, 2020-11-03:
@libinyang
Apart from topology change mentioned above by @juimonen, could you try to reproduce this issue with following patch?
https://github.com/bkokoszx/sof/commit/da1a5cd16be9e0eedc66b91571512a98f20b463d

@slawblauciak @juimonen 
FYI

lgirdwood, 2020-11-03:
@bkokoszx could you let us know any MCPS changes with your new patch. We dont want to go back to 70MCPS as before with demux.

libinyang, 2020-11-04:
I have submitted a patch in tplg https://github.com/thesofproject/sof/pull/3580 to use the 3 period in sof-icl-rt711-rt1308-rt715-hdmi.m4

libinyang, 2020-11-04:
> why would we do this? This looks like masking an error. We have way too much buffering in SOF IMHO.

@plbossart This is a draft patch. The final patch https://github.com/thesofproject/sof/pull/3580 applies to the special platforms.

libinyang, 2020-11-04:
> @libinyang
> Apart from topology change mentioned above by @juimonen, could you try to reproduce this issue with following patch?
> [bkokoszx@da1a5cd](https://github.com/bkokoszx/sof/commit/da1a5cd16be9e0eedc66b91571512a98f20b463d)
> 
> @slawblauciak @juimonen
> FYI

@bkokoszx It seems your patch can fix the distortion. Bard will test this patch on his platform too.

juimonen, 2020-11-04:
@libinyang thanks! @bardliao if Bard's test is good, @bkokoszx please make a proper PR, this is the right way to fix this.

bardliao, 2020-11-04:
> @libinyang
> Apart from topology change mentioned above by @juimonen, could you try to reproduce this issue with following patch?
> [bkokoszx@da1a5cd](https://github.com/bkokoszx/sof/commit/da1a5cd16be9e0eedc66b91571512a98f20b463d)
> 

I tested it and I can't find the distortion with [bkokoszx@da1a5cd](https://github.com/bkokoszx/sof/commit/da1a5cd16be9e0eedc66b91571512a98f20b463d) on my side

bkokoszx, 2020-11-04:
@lgirdwood 
I've performed tests with slim driver with format: ```48 kHz, 32/32, 8```, on the following topology:
[image]

And I've got following MCPS value for DEMUX:
```
[  1051521.562500] (       32.005207) c0 demux        1.1       src/audio/pipeline.c:206  perf comp_copy peak plat 1031 cpu 10787
```

plbossart, 2020-11-04:
> And I've got following MCPS value for DEMUX:
> 
> ```
> [  1051521.562500] (       32.005207) c0 demux        1.1       src/audio/pipeline.c:206  perf comp_copy peak plat 1031 cpu 10787
> ```

What are the units for 1031 and 10787?

bkokoszx, 2020-11-04:
The units are ```plat_ticks/pipeline_period``` or ```cpu_ticks/pipeline_period```. In above case (```pipeline_period = 1ms):
```
1031 plat_ticks/pipeline_period = 1031 plat_ticks/1ms = 1031000 plat_ticks/1s
10787 cpu_ticks/pipeline_period = 10787 cpu_ticks/1ms = 10787000 cpu_ticks/1s = 10,787 MCPS
```

plbossart, 2020-11-04:
@bkokoszx ~10 MCPS doesn't seem super-optimized for a plain data mover. We handle 48kHz data, that means an average of ~200 cycles per sample frame. DSPs should be able to do better.

It's not the end of the world for now but we will have to use an 8-ch MUX to deal with speaker-protection with I/V feedback and the number will start to be significant. see https://github.com/thesofproject/linux/pull/2514 for more details.

juimonen, 2020-11-04:
@bkokoszx just for comparison can you provide the figure without your patch with same settings?

bkokoszx, 2020-11-05:
@juimonen 
Do you mean without PR #3583 or without whole simplification in PR #3000 ?

juimonen, 2020-11-05:
@bkokoszx I meant without #3583. As I understood that @lgirdwood concern was that we are introducing more mips compared to the previous simplification. Or maybe you already have the figures from last simplification?

bkokoszx, 2020-11-05:
@juimonen 
Without #3000 (I've reverted https://github.com/thesofproject/sof/pull/3000/commits/5bf8188baad62eda98565d4bacc9a04712cba1e8) and without #3586 :
```
[  1840382.864583] (    42163.125000) c0 demux        1.1       src/audio/pipeline.c:206  perf comp_copy peak plat 6317 cpu 65785
```

Without #3583 (actual master):
```
[  1005080.442708] (       29.635416) c0 demux        1.1       src/audio/pipeline.c:206  perf comp_copy peak plat 947 cpu 9860
```

juimonen, 2020-11-05:
@bkokoszx ok so 1mpcs addition to the previous 9 (9->10)? I think that's ok then...

PR #3583 (2020-11-09): mux: add channels checking in *_prepare_active_look_up() functions
This commit adds channel parameter checking in demux/mux_prepare_active_look_up() functions.
Look up table should not include elements referring to channels higher than those actually set in
source/sink buffers. 

fixes #3499
closed 2020-11-09 by lgirdwood