thesofproject/sof #3503: [BUG] [BDW] IPC error on hw_params when buffer is too small

plbossart, 2020-10-07:
**Describe the bug**

On Broadwell Samus, the following command works fine

````
root@chrx:~# speaker-test -Dhw:0,0 -c2 -r48000 -FS32_LE -b 100000 -P2 -S5

speaker-test 1.2.2

Playback device is hw:0,0
Stream parameters are 48000Hz, S32_LE, 2 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 48 to 8184
Period size range from 24 to 2040
Requested buffer time 100000 us
Periods = 2
was set period_size = 1608
was set buffer_size = 4800
 0 - Front Left
^C 1 - Front Right
Time per period = 1.273407
````

Reducing the buffer size to a much smaller value creates unrecoverable IPC errors

````
root@chrx:~# speaker-test -Dhw:0,0 -c2 -r48000 -FS32_LE -b 10000 -P2 -S5

speaker-test 1.2.2

Playback device is hw:0,0
Stream parameters are 48000Hz, S32_LE, 2 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 48 to 8184
Period size range from 24 to 2040
Requested buffer time 10000 us
Periods = 2
Unable to set hw params for playback: Input/output error
Setting of hwparams failed: Input/output error
root@chrx:~# dmesg
[   61.194616] sof-audio-acpi INT3438:00: error: ipc error for 0x60010000 size 20
[   61.194621] sof-audio-acpi INT3438:00: error: hw params ipc failed for stream 0
[   61.194624] sof-audio-acpi INT3438:00: ASoC: error at snd_soc_pcm_component_hw_params on INT3438:00: -5
[   61.194627]  Low Latency: ASoC: hw_params FE failed -5
root@chrx:~# speaker-test -Dhw:0,0 -c2 -r48000 -FS32_LE -b 10000 -P2 -S5

speaker-test 1.2.2

Playback device is hw:0,0
Stream parameters are 48000Hz, S32_LE, 2 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 48 to 8184
Period size range from 24 to 2040
Requested buffer time 10000 us
Periods = 2
Unable to set hw params for playback: Device or resource busy
Setting of hwparams failed: Device or resource busy
root@chrx:~# ls
````
sof-logger trace

````
root@chrx:~# ./sof-logger -l sof-bdw.ldc 
         TIMESTAMP              DELTA C# COMPONENT          LOCATION                      CONTENT
[423296081.770833] (423296096.000000) c0 dma-trace             src/trace/dma-trace.c:327  ERROR FW ABI 0x3011000 DBG ABI 0x5002001 tag v1.5-rc1-756-g242be345ca9f src hash 0x0b028c9a (ldc hash 0x0b028c9a)
[1019902789.739583] (596606720.000000) c0 wait                         src/lib/wait.c:46   ERROR ewt
[1019902837.864583] (       48.125000) c0 ipc                ../ipc/ipc-host-ptable.c:126  ERROR ipc_get_page_descriptors(): dma_start() failed
[1019903013.489583] (      175.625000) c0 ipc                ../ipc/ipc-host-ptable.c:153  ERROR ipc: get descriptors failed -5
[1189245422.604167] (169342416.000000) c0 dw-dma                 src/drivers/dw/dma.c:283  ERROR dw_dma_start(): dma 1 channel 3 not ready ena 0x8 status 0x3
[1189245486.145833] (       63.541668) c0 ipc                ../ipc/ipc-host-ptable.c:126  ERROR ipc_get_page_descriptors(): dma_start() failed
[1189245661.770833] (      175.625000) c0 ipc                ../ipc/ipc-host-ptable.c:153  ERROR ipc: get descriptors failed -16
````

Full trace included
[sof-logger.txt](https://github.com/thesofproject/sof/files/5343469/sof-logger.txt)

**To Reproduce**

Install latest firmware and kernel and run command above

**Reproduction Rate**

100%

**Expected behavior**

Error flagged by kernel and/or recoverable IPC error

**Impact**

Showstopper

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 96b9d7cf347158efac3d830d9937e8efc39579c8
    * SOF: 242be345ca9f87db29b9cb4b12b91438d932814d
2) Name of the topology file
    * Topology: sof-bdw-rt5677.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: Broadwell

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

lgirdwood, 2020-10-08:
@caseorum @plbossart this looks like the transfer size is too small for the DMAC.
```c
	/* source buffer size is always PAGE_SIZE bytes */
	/* 20 bits for each page, round up to 32 */
	elem.size = (ring->pages * 5 * 16 + 31) / 32;
```
i.e. a small buffer means a small page table and a small DMA page table copy.

plbossart, 2020-10-08:
> @caseorum @plbossart this looks like the transfer size is too small for the DMAC.
>
> ```c
> 	/* source buffer size is always PAGE_SIZE bytes */
> 	/* 20 bits for each page, round up to 32 */
> 	elem.size = [...]

Not following, sorry @lgirdwood . I really never understood the whole concept of compressed page tables, and in this specific case I don't know how we could put a constraint on the buffer size after determining that the buffer size is too small. It's non-causal.

caseorum, 2020-10-08:
@plbossart I believe he means either 
1. constrain ALSA to k*(DMAC minimum, often the 4k page size IIRC) buffer sizes, which also puts a lower bound on latency, or
2. just always copy at least the minimum, which is the usual way in many applications.

(2) requires you to be sure that you won't segfault, that the dest buffer can fit the junk on the end, that that junk will be ignored, and that you aren't exposing sensitive data that's either left uninitialized or bitpacked into the end of that last page.
Unless I'm way off. @lgirdwood?

plbossart, 2020-10-08:
I must be missing something. I thought the compressed page tables were passed to the firmware with an IPC? I don't see how the DMA configuration could be transmitted with a DMA, that's seems like a technical oxymoron.

plbossart, 2020-10-08:
I just tested that this issue does NOT appear on Cherrytrail, I was able to test with really small ring buffers without issues

````
root@Zotac:~/sof-test/test-case# speaker-test -Dhw:0,0 -c2 -r48000 -FS32_LE -b 10000 -P2 -S5

speaker-test 1.2.3

Playback device is hw:0,0
Stream parameters are 48000Hz, S32_LE, 2 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 1536 to 7680
Period size range from 768 to 1536
Requested buffer time 10000 us
Periods = 2
was set period_size = 768
was set buffer_size = 1536
 0 - Front Left
 1 - Front Right
Time per period = 5.948875
 0 - Front Left
 1 - Front Right
^CTime per period = 4.239957
root@Zotac:~/sof-test/test-case# speaker-test -Dhw:0,0 -c2 -r48000 -FS32_LE -b 1000 -P2 -S5

speaker-test 1.2.3

Playback device is hw:0,0
Stream parameters are 48000Hz, S32_LE, 2 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 1536 to 7680
Period size range from 768 to 1536
Requested buffer time 1000 us
Periods = 2
was set period_size = 768
was set buffer_size = 1536
 0 - Front Left
 1 - Front Right
Time per period = 5.948957
````

caseorum, 2020-10-08:
@plbossart I'm new to this codebase, but I'm thinking the DMA is being configured (via IPC) for a copy that is too small for the DMAC on broadwell. So now we're getting into datasheet territory, but the implication here is that the newer ones have a 4k minimum, but broadwell's minimum is larger.

lgirdwood, 2020-10-08:
@plbossart @caseorum  the compressed page table

1) takes a host 32 bit DMA SG buffer.
2) converts to PHY pages (i.e. drops LSB 12 bits from each 32bit PHY addr )
3) Squashes those remaining 20bits PHY addresses for each page into a compressed array.

The size of the compressed array of 20bit PHY page addresses is then sent to DSP. The size of the array is determined by the size of the host audio buffer. i.e. one array elem for each 4kB audio buffer page, so small buffers will have less array elems in the page table. 16kB audio buffer will have array of 4 * 20bit elems (rounded up to nearest word)

plbossart, 2020-10-08:
I can confirm this bug does not occur on BYT/CHT. I tested with all sequences that are reproducibly wrong on Broadwell and no issues at all.

Clearly the Broadwell DMA as some issues.

lgirdwood, 2020-10-08:
> Clearly the Broadwell DMA has some issues.

Ack, we need to find the min transfer size and make sure we never go under that for page tables or audio buffer wrap.

PR #3659 (2020-12-03): ipc: Fix page table DMA transfer size
This corrects the minimum transfer size calculation for the ptable and uses the relevant DMA attribute.

Fixes #3492, fixes  #3503
closed 2020-12-03 by lgirdwood