thesofproject/sof #3507: [BUG] ext manifest has extra element

fredoh9, 2020-10-08:
**Describe the bug**
Working on https://github.com/thesofproject/linux/pull/2297.
Parsing CAVS EXT MANIFEST, ext_man_cavs_config_data. total 5 of elems are parsed. There are unnecessary extra elems.

Should be up to date with last element? like below?
```
/* EXT_MAN_ELEM_PLATFORM_CONFIG_DATA elements identificators */
enum cavs_config_elem_type {
	EXT_MAN_CAVS_CONFIG_LPRO	= 1,
	EXT_MAN_CAVS_CONFIG_OUTBOX_SIZE,
	EXT_MAN_CAVS_CONFIG_INBOX_SIZE,
	EXT_MAN_CAVS_CONFIG_LAST_ELEM = EXT_MAN_CAVS_CONFIG_INBOX_SIZE,	/**< keep up to date with last element */
};

/* EXT_MAN_ELEM_PLATFORM_CONFIG_DATA elements */
struct ext_man_cavs_config_data {
	struct ext_man_elem_header hdr;

	struct config_elem elems[EXT_MAN_CAVS_CONFIG_LAST_ELEM];
} __packed;
```

And even with this fix, one more question is... I'm not sure where is 5th element from. (short log is below)

**To Reproduce**
Need to test with https://github.com/thesofproject/linux/pull/2297.

**Reproduction Rate**
100%

**Expected behavior**
Only 3 elements are parsed

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
SOF master and linux pr 2297

**Screenshots or console output**
```
[    1.582426] sof-audio-pci 0000:00:1f.3: hdr->type: 6, hdr->size = 48, elem_num=5
[    1.582428] sof-audio-pci 0000:00:1f.3: FW clock config: LPRO
[    1.582430] sof-audio-pci 0000:00:1f.3: unsupported token type: 2 value=4096
[    1.582431] sof-audio-pci 0000:00:1f.3: unsupported token type: 3 value=8192
[    1.582433] sof-audio-pci 0000:00:1f.3: unsupported token type: 0 value=0
[    1.582434] sof-audio-pci 0000:00:1f.3: unsupported token type: 0 value=0
```

fredoh9, 2020-10-08:
In second thought, I don't remember why we have to start this from 1. I kind of remember it caused some issues but I'm not sure that concern is still valid. Can we start this enum from 0 also?

`EXT_MAN_CAVS_CONFIG_LPRO	= 0,`

ranj063, 2020-10-08:
> In second thought, I don't remember why we have to start this from 1. I kind of remember it caused some issues but I'm not sure that concern is still valid. Can we start this enum from 0 also?
> 
> `EXT_MAN_CAVS_CONFIG_LPRO = 0,`

It needs to start from one for the exac reason that we have static sizes array and the elems in the static array will be initialized to 0. Then you wouldnt know if it is a real element or the default element in the array?

ranj063, 2020-10-08:
besides if you only parse the exact number of elems based on size, you wont have this problem isnt it?

fredoh9, 2020-10-08:
Ah, I know remember the discussion why it starts from 1.

lyakh, 2020-10-09:
> It needs to start from one for the exac reason that we have static sizes array and the elems in the static array will be initialized to 0. Then you wouldnt know if it is a real element or the [...]

Aren't there also maybe some other field that we can check to verify, that the element is valid? Or you could make the array size `array[LAST - 1];`? But as discussed in https://github.com/thesofproject/linux/pull/2297 I don't find the idea of the size of an array being exactly equal to the current number of possible elements, good.

ktrzcinx, 2020-10-09:
> Aren't there also maybe some other fields that we can check to verify, that the element is valid?

What profit it has above just skipping zero filled elements ? Maybe we should just explicit mark EXT_MAN_CONFIG_EMPTY.

>I don't find the idea of the size of an array being exactly equal to the current number of possible elements

it's rather array capacity, defined to workaround xcc problems with static flex arrays

lgirdwood, 2020-10-09:
> Parsing CAVS EXT MANIFEST, it is supposed to parse total 3 elems from ext_man_cavs_config_data but total 5 of elems are parsed. There are unnecessary extra elems. It causes warnings.

This is intentional as it means we can append new manifest elems without breaking the kernel (kernel just ignores the new feature and in this case prints a warning). The last elem ID does NOT have to be ABI aligned between FW and kernel, as long as the array elem count is correct.

xiulipan, 2020-10-20:
@fredoh9 I also have a RFC https://github.com/thesofproject/sof/pull/3527 to add a FIRST enum. This will keep minor when we add new enum.

lgirdwood, 2020-10-20:
@fredoh9 just fyi @ktrzcinx and @xiulipan are syncing on fix for this given discussion today on call.

closed 2025-07-04 by github-actions[bot]