thesofproject/sof #3518: [BUG][JSL][KD]v4.0-rc2: Not able to wake DUT from S0ix with "OK Google"

yongzhi1, 2020-10-13:
**Describe the bug**
While integrating the Google Hotword module, the Key Phrase is detected in S0, but once upon entering the S0ix, wake up DUT with Key Phase is not successful. Tested on both Waddledoo and Drawlat.

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

Checkout the SOF fw branch:
1.https://github.com/yongzhi1/sof/tree/hotword-jsl-v4.0-rc2
2. Build SOF for JSL + Google module with xcc
3. Find the mixer control and set the language model on DUT:
./sof-ctl -n 41 -t 1 -b -r -s language_model
4.Open the WoV PCM device for listening:
arecord -Dhw:0,6 -M -N -r 16000 -c2 -f s16_le --buffer-size=67200 -vvv /tmp/wov.wav
5.Open another DUT console and put the system to sleep
echo freeze > /sys/power/state

Wait a little while, say "OK Google", watch the DUT screen to turn on (same effect as touching keypad)

**Reproduction Rate**

> 8/10

**Expected behavior**
Keyphase should wake up the DUT from S0ix reliably. 

**Impact**
Blocking for the integration

**Environment**
Kernel:
https://chromium.googlesource.com/chromiumos/third_party/kernel/+log/refs/heads/chromeos-5.4
   
SOF:
v4.0-rc2 and cherry-pick two KWD wrapper patches.
(https://github.com/bzhg/sof/commits/hotword-tgl-007 )

lgirdwood, 2020-10-14:
@slawblauciak fyi

slawblauciak, 2020-10-15:
@yongzhi1 does the issue occur on the releases/jsl/v4.0 tag? If so, please provide dmesg and FW traces.

udaymb, 2020-10-19:
In both LPS and no LPS mode, KWD works 
No LPS mode - 
- KWD works sof-log-nolps-kwd.zip
- WoV does not work - sof-log-nolps-kwd.zip 
- /var/log/messages - contains KWD and WoV logs

[sof-log-nolps-kwd.zip](https://github.com/thesofproject/sof/files/5403722/sof-log-nolps-kwd.zip)
[messages-nolps.zip](https://github.com/thesofproject/sof/files/5403524/messages-nolps.zip)
[sof-log-nolps-wov.zip](https://github.com/thesofproject/sof/files/5403522/sof-log-nolps-wov.zip)

LPS mode - WoV works 
- KWD works sof-log-nolps-kwd.zip
- WoV works - sof-log-nolps-kwd.zip 
- /var/log/messages - contains KWD and WoV logs
[sof-log-lps-kwd.zip](https://github.com/thesofproject/sof/files/5403673/sof-log-lps-kwd.zip)
[messages-lps.zip](https://github.com/thesofproject/sof/files/5403674/messages-lps.zip)
[sof-logger-lps-wov.zip](https://github.com/thesofproject/sof/files/5403677/sof-logger-lps-wov.zip)

yongzhi1, 2020-10-20:
Thanks @udaymb, guess KWD means detect_test.c method while WoV refers to google_hotword_detect.c in your description.

udaymb, 2020-10-21:
Yes @yongzhi1 - KWD means Ok-Google - Key Word

slawblauciak, 2020-11-13:
@yongzhi1 @udaymb can you please use the v4.0 (non-rc) tag instead? Also, please cherry-pick this commit, it will make traces a bit more clear https://github.com/slawblauciak/sof/commit/28df367a5bf51e7b62f5b025c28700945281607b

yongzhi1, 2020-12-04:
Logs sent in email, thanks!!

yongzhi1, 2021-01-20:
Close it for now due to lowered priority. Thank you.

closed 2021-01-20 by yongzhi1