thesofproject/sof #3519: [BUG] Microphone on HP Spectre x360 13-aw0xxx doesn't work

mattst88, 2020-10-13:
As mentioned in the Arch Wiki (https://wiki.archlinux.org/index.php/HP_Spectre_x360_-_13-aw0xxx#Audio) and in many other places on the internet, the microphone array in the HP Spectre x360 13-aw0xxx doesn't work under SOF. It shows up in `alsamixer` but e.g. gnome-settings doesn't register any sound from the microphones. An external microphone plugged into the machine using the 3.5mm jack works fine.

I've used kernels v5.7 to v5.9 all with the same results. Using SOF firmware v1.5.1.

Is this hardware supposed to be supported? I'm honestly horrified at how complex the SOF kernel configuration is. I've spent a significant amount of time trying to make this work under the assumption that it's just a missing kernel config option (and perhaps it is!) but it shouldn't be this difficult to figure out.

`lsmod`:

````
Module                  Size  Used by
snd_soc_skl_hda_dsp    20480  3
snd_soc_hdac_hdmi      28672  1 snd_soc_skl_hda_dsp
snd_hda_codec_hdmi     57344  1
snd_hda_codec_realtek   118784  1
snd_soc_dmic           16384  1
snd_hda_codec_generic    77824  1 snd_hda_codec_realtek
iwlmvm                307200  0
snd_sof_pci            16384  0
mac80211              749568  1 iwlmvm
snd_sof_intel_hda_common    65536  1 snd_sof_pci
snd_soc_acpi_intel_match    36864  2 snd_sof_pci,snd_sof_intel_hda_common
snd_soc_acpi           16384  2 snd_soc_acpi_intel_match,snd_sof_intel_hda_common
snd_soc_hdac_hda       20480  1 snd_sof_intel_hda_common
snd_sof_xtensa_dsp     16384  1 snd_sof_intel_hda_common
snd_sof                94208  2 snd_sof_pci,snd_sof_intel_hda_common
snd_sof_intel_hda      16384  1 snd_sof_intel_hda_common
snd_hda_ext_core       20480  4 snd_sof_intel_hda_common,snd_soc_hdac_hdmi,snd_soc_hdac_hda,snd_sof_intel_hda
iwlwifi               221184  1 iwlmvm
snd_soc_core          221184  6 snd_sof,snd_sof_intel_hda_common,snd_soc_hdac_hdmi,snd_soc_hdac_hda,snd_soc_dmic,snd_soc_skl_hda_dsp
snd_hda_intel          40960  0
snd_intel_dspcfg       16384  3 snd_hda_intel,snd_sof_pci,snd_sof_intel_hda_common
snd_hda_codec         126976  6 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek,snd_soc_hdac_hda,snd_soc_skl_hda_dsp
snd_hwdep              16384  1 snd_hda_codec
snd_hda_core           81920  11 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_ext_core,snd_hda_codec,snd_hda_codec_realtek,snd_sof_intel_hda_common,snd_soc_hdac_hdmi,snd_soc_hdac_hda,snd_sof_intel_hda,snd_soc_skl_hda_dsp
snd_pcm                94208  8 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_sof,snd_sof_intel_hda_common,snd_soc_hdac_hdmi,snd_soc_core,snd_hda_core
snd_timer              32768  1 snd_pcm
snd                    73728  16 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_soc_core,snd_pcm,snd_soc_skl_hda_dsp
soundcore              16384  1 snd
hid_multitouch         28672  0
cfg80211              700416  3 iwlmvm,iwlwifi,mac80211
ucsi_acpi              16384  0
typec_ucsi             32768  1 ucsi_acpi
typec                  36864  1 typec_ucsi
tpm_tis                16384  0
tpm_tis_core           20480  1 tpm_tis
````

Relevant `sudo lspci -vvnn` output:

````
00:1f.3 Multimedia audio controller [0401]: Intel Corporation Smart Sound Technology Audio Controller [8086:34c8] (rev 30)
        Subsystem: Hewlett-Packard Company Smart Sound Technology Audio Controller [103c:86fa]
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 32, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 166
        Region 0: Memory at 603d170000 (64-bit, non-prefetchable) [size=16K]
        Region 4: Memory at 603d000000 (64-bit, non-prefetchable) [size=1M]
        Capabilities: [50] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=55mA PME(D0-,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [80] Vendor Specific Information: Len=14 <?>
        Capabilities: [60] MSI: Enable+ Count=1/1 Maskable- 64bit+
                Address: 00000000fee007f8  Data: 0000
        Kernel driver in use: sof-audio-pci
        Kernel modules: snd_hda_intel, snd_sof_pci
````

`dmesg | egrep -i '(sof|dsp|snd)'`:

[... 5629 lines cut ...]
# CONFIG_CPA_DEBUG is not set
# CONFIG_DEBUG_ENTRY is not set
# CONFIG_DEBUG_NMI_SELFTEST is not set
# CONFIG_X86_DEBUG_FPU is not set
# CONFIG_PUNIT_ATOM_DEBUG is not set
CONFIG_UNWINDER_ORC=y
# CONFIG_UNWINDER_FRAME_POINTER is not set
# CONFIG_UNWINDER_GUESS is not set
# end of x86 Debugging

#
# Kernel Testing and Coverage
#
# CONFIG_KUNIT is not set
# CONFIG_NOTIFIER_ERROR_INJECTION is not set
CONFIG_FUNCTION_ERROR_INJECTION=y
# CONFIG_FAULT_INJECTION is not set
CONFIG_ARCH_HAS_KCOV=y
CONFIG_CC_HAS_SANCOV_TRACE_PC=y
# CONFIG_KCOV is not set
# CONFIG_RUNTIME_TESTING_MENU is not set
# CONFIG_MEMTEST is not set
# end of Kernel Testing and Coverage
# end of Kernel hacking
````

mattst88, 2020-10-13:
````
# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 3: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 4: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 5: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
````

plbossart, 2020-10-13:
@mattst88 we routinely test the DMIC input on a variety of platforms. The hardware hasn't changed in several years and should just work (tm). What worries me the most is the guidance provided in the Arch wiki, which makes no sense really.

Can you confirm that
a) you are using the latest alsa-lib and alsa-ucm-conf
b) you've tested the mic capture at the ALSA hw: device level (should be something like arecord -Dhw:0,4 -c2 -r48000 -fS32_LE -d10 test.wav). please check the device number with arecord -l
c) that you've removed all pulseaudio configuration for sources and sinks, the Arch wiki info is obsolete.

we really need to figure out if you have a platform issue where no capture can be seen at the hw: level device, or something due to your distro higher in userspace.

Also what would help is that if you add a link to the alsa-info results.

In general it's also better to attach a file than paste it...

Thanks!

mattst88, 2020-10-14:
Thanks, I installed `media-libs/alsa-ucm-conf` and `media-libs/alsa-topology-conf` in Gentoo and the microphone now works. Without them, `arecord` showed that the microphone was working at the ALSA layer.

Nothing in all of Gentoo depends on either of these packages. Should, perhaps, `media-sound/pulseaudio` depend on them?

When did getting audio working become more complex than getting 3D working? :|

closed 2020-10-14 by mattst88