thesofproject/sof #3520: [BUG] BYT: Random stereo channels swaps happen with sof-byt-nocodec.tplg

singalsu, 2020-10-14:
**Describe the bug**
In a test that plays different frequencies sine waves to PCM0P and captures from PCM0C the sine wave frequencies have swapped that indicates that the playback, loopback, or capture channels got swapped.

**To Reproduce**
With sof-test PR #410 (to add the test) and sof PR #3509 (add the volume switch to nocodec topologies) run:
`sof-test/test-case/check-volume-levels.sh
`
**Reproduction Rate**
For one sub-test (play-capture) about 50% occurrence.

**Expected behavior**
No channels swapped, the test passed as it does in cAVS platforms. 

**Impact**
Annoyance for stereo music, showstopper for applications where channels order is critical.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**

```
2020-10-14 11:20:57 UTC [INFO] Measuring volume gains
File /tmp/tmp.iat6kH01et.wav:
pass (1/3)
File /tmp/tmp.EIsHmXIzcE.wav:
Channel 1 failed lower gain limit at  0.5 -  0.9s, gain -47.2 dB, min -10.5 dB
Channel 2 failed lower gain limit at  0.5 -  0.9s, gain -27.1 dB, min   9.5 dB
Channel 2 failed lower gain limit at  1.5 -  1.9s, gain -37.1 dB, min  -0.5 dB
fail (2/3)
Note: The test would pass with swapped channels.
File /tmp/tmp.3pEPEAF8DL.wav:
Channel 1 failed lower gain limit at  2.5 -  2.9s, gain -37.2 dB, min  -0.5 dB
Channel 1 failed lower gain limit at  3.5 -  3.9s, gain -37.2 dB, min  -0.5 dB
Channel 2 failed lower gain limit at  3.5 -  3.9s, gain -37.1 dB, min  -0.5 dB
fail (3/3)
Note: The test would pass with swapped channels.
FAIL
2020-10-14 11:21:11 UTC [ERROR] Error: Failed one or more tests in volume levels check.
2020-10-14 11:21:11 UTC [ERROR] Please inspect files: /tmp/tmp.iat6kH01et.wav, /tmp/tmp.EIsHmXIzcE.wav, and /tmp/tmp.3pEPEAF8DL.wav.
2020-10-14 11:21:11 UTC [ERROR] Starting func_exit_handler(), exit status=1, FUNCNAME stack:
2020-10-14 11:21:11 UTC [ERROR]  die()  @  /home/singalsu/git/sof-test/case-lib/lib.sh
2020-10-14 11:21:11 UTC [ERROR]  measure_levels()  @  sof-test/test-case/check-volume-levels.sh:328
2020-10-14 11:21:11 UTC [ERROR]  main()  @  sof-test/test-case/check-volume-levels.sh:84
2020-10-14 11:21:11 UTC [ERROR]  main()  @  sof-test/test-case/check-volume-levels.sh:336
2020-10-14 11:21:11 UTC [INFO] Test Result: FAIL!

```

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

keyonjie, 2020-10-20:
@singalsu I figured out a fix for this issue here https://github.com/thesofproject/sof/pull/3535, could you please help try if it works for you also?

singalsu, 2020-11-18:
@keyonjie The channels swap issue happens still with BYT nocodec topology (LBM quirk is set). The kernel, firmware etc. are from git today.

```
Kernel Branch: topic/sof-dev
Kernel Commit: cfa9a863e20a
SOF Branch: master
SOF Commit: e7a339bd4

```
When I last tested near month ago #3535 it didn't help with LBM, didn't help with Tx-Rx wire and no LBM quirk.

keyonjie, 2020-12-01:
Hi @singalsu can you help to try if the PR https://github.com/thesofproject/sof/pull/3656 can help on BYT for external wired loopback?

For the internal LBM, the channel random swapped in the capture is the expected behavior.

keyonjie, 2020-12-02:
Hi @singalsu unfortunately, after several hours debugging, it looks to me that the external wired mode looks quite similar to internal LBM for BYT SSP, the capture can't be stopped separately so the RX FIFO watermark(RFL) keep increasing even we have cleared the RSRE (while SSE is set), so, to me, this issue looks is not easy to be fixed.

lgirdwood, 2021-01-19:
@lyakh fyi

lyakh, 2021-02-01:
presumably fixed by #3777

singalsu, 2021-02-02:
@lyakh I got still error and detected channels swap from check-volume-levels.sh. The same happens with LBM quirk an without with Tx-Rx jumper wire.

keyonjie, 2021-06-11:
@singalsu @lyakh looks we have to close this as wont_fix, no advance effort is planned to work on it.

lyakh, 2021-06-11:
do we have a list of "known won't-fix issues" to be tracked in case they do get fixed by accident or they have to be re-opened and we can benefit from earlier discussions?

keyonjie, 2021-06-11:
yes, we have a label for that, just add it to this issue

keyonjie, 2021-06-15:
closing this as wont_fix.

closed 2021-06-15 by keyonjie