thesofproject/sof #3522: [BUG][CHT] IPC error during multiple pause/resume test with Deep Buffer PCM

plbossart, 2020-10-14:
Filing as a Linux bug for now since the result is an IPC error. I've seen this error multiple times with sof-test, this is not a random error but a mixer issue from the logs.

@lgirdwood @lyakh @singalsu FYI

Log files: [multiple-pause-resume.tar.gz](https://github.com/thesofproject/linux/files/5374776/multiple-pause-resume.tar.gz)

excerpts:
dmesg: 
````
kernel: [  531.157632] sof-audio-acpi 808622A8:00: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
kernel: [  531.157787] sof-audio-acpi 808622A8:00: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
kernel: [  531.317334] sof-audio-acpi 808622A8:00: pcm: trigger stream 0 dir 0 cmd 3
kernel: [  531.317358] sof-audio-acpi 808622A8:00: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
kernel: [  531.317486] sof-audio-acpi 808622A8:00: ipc tx succeeded: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
kernel: [  531.417219] sof-audio-acpi 808622A8:00: pcm: trigger stream 1 dir 0 cmd 1
kernel: [  531.417276] sof-audio-acpi 808622A8:00: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
kernel: [  531.417408] sof-audio-acpi 808622A8:00: error: ipc error for 0x60040000 size 12
kernel: [  531.417435] sof-audio-acpi 808622A8:00: ASoC: error at snd_soc_pcm_component_trigger on 808622A8:00: -22
kernel: [  531.417450]  PCM Deep Buffer: ASoC: trigger FE cmd: 1 failed: -22
kernel: [  531.417472] sof-audio-acpi 808622A8:00: pcm: trigger stream 1 dir 0 cmd 0
````

etrace:
````
        TIMESTAMP              DELTA C# COMPONENT          LOCATION                      CONTENT
[  8250820.781250] (  8250821.000000) c0 dma-trace             src/trace/dma-trace.c:327  ERROR FW ABI 0x3012000 DBG ABI 0x5002001 tag v1.5-rc1-778-ge5d17f3c10f3 src hash 0xa3b71c81 (ldc hash 0xa3b71c81\
)
[531415530.364583] (523164704.000000) c0 mixer        1.1      src/audio/component.c:190  ERROR comp_set_state(): wrong state = 4, COMP_TRIGGER_START
[531415540.364583] (       10.000000) c0 pipe         4.25      src/audio/pipeline.c:884  ERROR pipeline_trigger(): ret = -22, host->comp.id = 21, cmd = 1
[531415563.750000] (       23.385416) c0 ipc                       src/ipc/handler.c:443  ERROR ipc: comp 21 trigger 0x40000 failed -22
```

plbossart, 2020-10-14:
@ranj063 can you help identify which platforms expose PCM devices that get mixed? I only see byt/cht for now. Thanks!

ranj063, 2020-10-14:
@plbossart apart from the byt/cht codec topologies, the byt/cght nocodec and apl nocodec topologies have mixer based PCMs

plbossart, 2020-10-14:
@ranj063 who is the platform owner for Up2 nocodec? that bug should also impact that device.

fredoh9, 2020-10-14:
@marc-hb is owner of APL_UP2_NOCODEC.

plbossart, 2020-10-14:
I actually don't see any mixers in the apl-nocodec topology

````
# PCM0 <---> volume <----> SSP0
# PCM1 <---> volume <----> SSP1
# PCM2 <---> volume <----> SSP2
# PCM3 <---> volume <----> SSP3
# PCM4 <---> volume <----> SSP4
# PCM5 <---> volume <----> SSP5
# PCM6 <---- volume <----- DMIC6 (DMIC01)
````

marc-hb, 2020-10-14:
I'm the UP2 platform owner but I stuck to pcm512x so far. Finding a magic system configuration (BIOS version, ACPI override, topology, kernel config,...) that actually produced some sound was by far the most time-consuming part of the "SOF rite of passage" and a Zephyr detour + the lack of firmware loader logs made it even more painful so I avoided any configuration change so far to focus on more productive things. Let me know if/when I should bite the bullet.

fredoh9, 2020-10-14:
I saw a mixer.
[image]

ranj063, 2020-10-14:
> I actually don't see any mixers in the apl-nocodec topology
>
> ```
> # PCM0 <---> volume <----> SSP0
> # PCM1 <---> volume <----> SSP1
> # PCM2 <---> volume <----> SSP2
> # PCM3 <---> volume <----> SSP3
> # [...]

clearly someone missed updating this when they added the mixer. I wonder if that was me

plbossart, 2020-10-14:
> I'm the UP2 platform owner but I stuck to pcm512x so far. Finding a magic system configuration (BIOS version, ACPI override, topology, kernel config,...) that actually produced some sound was by far [...]

The Kconfig FORCE_NOCODEC is your friend. It will ignore whatever you've setup with ACPI overrides, etc, you don't need to change anything on the device itself - except make sure the sof-apl-nocodec.tplg file is present on the disk.

plbossart, 2020-10-15:
@mmaka1 the state diagram in https://thesofproject.github.io/latest/developer_guides/firmware/components/component-overview.html#handle-the-component-device-state cannot work for a mixer component. When the mixer component is paused, it should go back to active when a new source starts. The trigger_release isn't the only stimulus to go from paused to active.

plbossart, 2020-10-15:
The alternative would be that the mixer never pauses but keeps generating zeroes.

plbossart, 2020-10-15:
> I'm the UP2 platform owner but I stuck to pcm512x so far. Finding a magic system configuration (BIOS version, ACPI override, topology, kernel config,...) that actually produced some sound was by far [...]

I just did it for you while I was debugging sof-test scripts, same error on MinnowBoard nocodec. No surprise, it's a conceptual bug of the mixer component.

marc-hb, 2020-10-15:
> The Kconfig FORCE_NOCODEC is your friend.

Thanks for the tip, BTW this pushed me to finally fix this kernel section in the getting started guide: https://github.com/thesofproject/sof-docs/pull/295/

lgirdwood, 2020-10-19:
> I just did it for you while I was debugging sof-test scripts, same error on MinnowBoard nocodec. No surprise, it's a conceptual bug of the mixer component.

ok, so it looks like this will hit all platforms and topologies that use mixer. @slawblauciak fyi for scrub.

marc-hb, 2020-10-19:
> This bug is 100% reproducible. Connect with the target with two terminals. ...

At the risk of stating the obvious: don't close this bug until there is a test in sof-test that performs something as (deceptively?) simple as this or triggers the same bug in some other, equally simple way.

plbossart, 2020-10-19:
> > This bug is 100% reproducible. Connect with the target with two terminals. ...
> 
> At the risk of stating the obvious: don't close this bug until there is a test in sof-test that performs something as (deceptively?) simple as this or triggers the same bug in some other, equally simple way.

yep, adding an issue against sof-test here: https://github.com/thesofproject/sof-test/issues/449

slawblauciak, 2020-10-20:
@wwittbrx please try to reproduce with slim driver tests.

mmaka1, 2020-10-22:
@plbossart The diagram you mentioned is a common generic state machine. Do you want to add another transition for all, or just for the mixer component as an exception? The thread is about the mixer, but there are/could be other components of different types in the same pipeline and they receive START being in the paused state as well.

plbossart, 2020-10-22:
@mmaka1 I think the pause case is not well handled in a component can have more than one source, and each of the sources can be paused *independently*. We can either keep the existing diagram for simplicity and add a footnote, or change the diagram with e.g. a doted line arrow.

This could happen e.g. for a mux as well.

slawblauciak, 2020-10-22:
If I understand correctly, the issue is that there are only two pipelines. If the mixer component is inside one of the source pipelines, it effectively removes your ability to have a clean way of independently stoppings/starting those sources.

Say, the pipeline 1 has source 1 and mixer.
Pipeline 2 has source 2 and connects to the mixer from the pipeline 1.

Both pipelines are stopped. You start pipeline 2, keeping 1 stopped. What's supposed to happen? In order to activate mixer, pipeline 1 needs to be started. But the driver doesn't provide any data for pipeline 1, host component will be starved and fun things will happen.

The way I see it is the mixer should be separated into a separate pipeline. Only then can you properly decide which source pipeline should be on or off. However, that possibility is currently being tested by @wwittbrx

plbossart, 2020-10-22:
look at the BYT topologies, the DAI pipeline START with a mixer - same as cAVS firmware:

# PCM0 -----> volume -------v
#                            low latency mixer ----> volume ---->  SSP2
# PCM1 -----> volume -------^
# PCM0 <---- Volume <---- SSP2

Having a mixer inside of a pipeline is silly and I've been fighting for 2+ years to stop this abuse.

lgirdwood, 2020-10-23:
> Having a mixer inside of a pipeline is silly and I've been fighting for 2+ years to stop this abuse.

It looks like the issue here is more to do with pipeline graph walking rules and logic since mixer usage is a normal and proper pipeline usage unless I'm missing something deeper here ?

mwasko, 2020-10-28:
> > Having a mixer inside of a pipeline is silly and I've been fighting for 2+ years to stop this abuse.
> 
> It looks like the issue here is more to do with pipeline graph walking rules and logic since mixer usage is a normal and proper pipeline usage unless I'm missing something deeper here ?

@lgirdwood  pipeline graph walking rules will not resolve the problem with having mixer inside of a pipeline. Please take as a reference @slawblauciak  comment:

> Both pipelines are stopped. You start pipeline 2, keeping 1 stopped. What's supposed to happen? In order to activate mixer, pipeline 1 needs to be started. But the driver doesn't provide any data for pipeline 1, host component will be starved and fun things will happen.

This is a design limitation and the valid use case is to keep mixer as a start of DAI pipeline:

> # PCM0 -----> volume -------v
> #                            low latency mixer ----> volume ---->  SSP2
> # PCM1 -----> volume -------^
> # PCM0 <---- Volume <---- SSP2

@wwittbrx has already confirmed that in above configuration problem no longer reproduce for playback path.

I would like to suggest to close this issue as a 'not a bug'.

plbossart, 2020-10-28:
> This is a design limitation and the valid use case is to keep mixer as a part of separate DAI pipeline that start with a mixer.
> 
> @wwittbrx has already confirmed that in above configuration problem no longer reproduce for playback path.
> 
> I would like to suggest to close this issue as a 'not a bug'.

@mwasko the bug DOES happen with the mixer starting at the beginning of a pipeline, that's how I found it with two aplay commands. I don't understand your statement at all.

wwittbrx, 2020-10-29:
@mwasko @plbossart Sorry for the confusion, I had incorrect flow compared to Pierre in my test case. After correcting I can see the same result.

To summarize the issue and get on the same page, please confirm if recap below is accurate:

[image]

We have 3 pipelines P1, P2, P3 in state COMP_STATE_PREPARE.

We trigger start on P1 and it propagates to P3. They are now in COMP_STATE_ACTIVE.
We trigger pause on P1 and it propagates to P3. They are now in COMP_STATE_PAUSED.
We trigger start on P2 and it propagates to P3. We get an error because P3 is in STATE_PAUSED instead of STATE_PREPARED and expects trigger release instead of trigger start.

plbossart, 2020-10-29:
Thanks for the clarification @wwittbrx, glad we are on the same wavelength now.

> We have 3 pipelines P1, P2, P3 in state COMP_STATE_PREPARE.
>
> We trigger start on P1 and it propagates to P3. They are now in COMP_STATE_ACTIVE.
> We trigger pause on P1 and it propagates to P3. They [...]

That seems correct. 

But as noted earlier, the restriction is aligned with the state diagram here:  https://thesofproject.github.io/latest/developer_guides/firmware/components/component-overview.html#handle-the-component-device-state

So before lifting the restrictions, we'd probably need to tag components to allow for this paused->started transition. 

As an alternative to tagging, you could determine at run-time the number of sources in Active and paused states and only allow for this transition when it makes sense. I don't want to prescribe a solution since I am not familiar enough with the state management.

mwasko, 2020-10-29:
> So before lifting the restrictions, we'd probably need to tag components to allow for this paused->started transition.
>
> As an alternative to tagging, you could determine at run-time the number of [...]

Keep in mind that what your proposing would require to allow paused->active transition not only for mixer but all the other components that are behind it which basically can by any component.

I am wondering if instead of that we could on P1 pipeline pause keep the P3 mixer pipeline in active state and generate zero data at mixer level in such situation.

plbossart, 2020-10-29:
> Keep in mind that what your proposing would require to allow paused->active transition not only for mixer but all the other components that are behind it which basically can by any component.

you've probably right, it's not clear if we can detect 'when it makes sense'.

> I am wondering if instead of that we could on P1 pipeline pause keep the P3 mixer pipeline in active state and generate zero data at mixer level in such situation.

That's certainly a possibility - and a force-in-function design decision to start 'dai pipelines' by a mixer and prevent mixers from being used anywhere else.

the P3 mixer pipeline is the moral equivalent of what an audio server like PulseAudio would do, and these sort of servers never pause their output. Same for a capture pipeline, pausing makes no sense. I wonder how invasive such a change might be though, e.g. if all the inputs to a mixer are paused, that mixer still need to operate on its own.

mwasko, 2020-11-02:
> That's certainly a possibility - and a force-in-function design decision to start 'dai pipelines' by a mixer and prevent mixers from being used anywhere else.

@abonislawski will evaluate the possibility to implement the mechanism for zero data generation for mixer scenario. I think there were some attempts in 'buffer' configuration to handle such use-cases.

The design check to prevent mixers from being used anywhere else then on the start of 'dai pipelines' seems like a feature that should be addressed in topology creation tool and not the FW itself.

> the P3 mixer pipeline is the moral equivalent of what an audio server like PulseAudio would do, and these sort of servers never pause their output. Same for a capture pipeline, pausing makes no [...]

@plbossart are there any plans for the SOF driver to distinguish 'host' and 'dai' pipelines instead of handling them as a single entity?

plbossart, 2020-11-02:
> @plbossart are there any plans for the SOF driver to distinguish 'host' and 'dai' pipelines instead of handling them as a single entity?

Not that now of. Why would this help anyways? the pipeline is an abstract grouping of component and an indication to firmware that said components can be scheduled together. The driver has no real use for this information. We actually need to stop thinking in terms of long pipelines, if we start enabling topologies similar to what Windows does with SFX, MFX and EFX we will soon have a hierarchical mixing and pipelines at times reduced to a single component....

mwasko, 2020-11-05:
> Not that now of. Why would this help anyways? the pipeline is an abstract grouping of component and an indication to firmware that said components can be scheduled together. The driver has no real [...]

In current SOF design driver owns pipeline management and in my proposition we would need on P1 pipeline pause keep the P3 pipeline active. We can of course add some logic in FW that in certain circumstances will keep DAI pipeline active but in my opinion it is not a right way to go. If the driver is responsible for pipeline management then the FW should not interfere.

plbossart, 2020-11-05:
> In current SOF design driver owns pipeline management and in my proposition we would need on P1 pipeline pause keep the P3 pipeline active. We can of course add some logic in FW that in certain [...]

The driver only handles PCM devices and pause_push/resume at that level. we don't have information at the PCM level of how things are connected behind and whether there's a mixer, it's done at the DAPM level which doesn't have a notion of pause. 

@ranj063 would you concur?

ranj063, 2020-11-05:
> The driver only handles PCM devices and pause_push/resume at that level. we don't have information at the PCM level of how things are connected behind and whether there's a mixer, it's done at the [...]

@plbossart that is correct. We dont handle pipeline manement in the driver and even if we know the widget connections from DAPM, there is no distinction between a widget that is active or that is paused. This issue is purely has to do with the way the pipeline params/trigger propagation, more so to do with params in fact. I think a simple fix would be (completely untested)
```
diff --git a/src/audio/pipeline.c b/src/audio/pipeline.c
index cd8b4603d..8f68e6db2 100644
--- a/src/audio/pipeline.c
+++ b/src/audio/pipeline.c
@@ -483,7 +483,7 @@ static int pipeline_comp_params(struct comp_dev *current,
        }
 
        /* don't do any params if current is running */
-       if (current->state == COMP_STATE_ACTIVE)
+       if (current->state >= COMP_STATE_PAUSED)
                return 0;
 
        /* do params negotiation with other branches(opposite direction) */
```

ranj063, 2020-11-05:
@mwasko @plbossart we've got more than just pipeline propagation here. The state machine transition for a comp to START is only allowed from PREPARE and not from RELEASE. I think may be we should ease that restriction as well. 

I am thinking maybe keeping the pipeline P3 in active state even when paused might be the right way to do too

abonislawski, 2020-11-10:
Just to have more options - we can also fix this (tested and working) by keeping P3 in active state and adding flag "unrderrun_permitted" to a BUF in P3

plbossart, 2020-11-10:
@abonislawski if you end-up with underruns at the DAI level but keep going, you might end-up with channel swaps. That's not recommended... if it's a 'software underrun' detected on the input of the mixer and you still generate zeroes that might work.
I honestly don't really see the nuance with what was suggested earlier (keep the pipeline active *without* adding the underrun_permitted as you suggested)

abonislawski, 2020-11-10:
@plbossart well, I have also a working version with generating zeroes at mixer level without underrun_permitted flag but it was told me to revert that and use this flag because _generating zeroes_ is a bad approach.
So after all we have few solutions, this is all about decision which should be applied.

plbossart, 2020-11-10:
@abonislawski why is generating zeroes a bad approach? it's what PulseAudio does...

wwittbrx, 2020-11-19:
@mengdonglin PR for this issue https://github.com/thesofproject/sof/pull/3618 has been submitted and can be tested. Looks good so far in slim driver tests.

marc-hb, 2020-11-24:
> At the risk of stating the obvious: don't close this bug until there is a test in sof-test that performs something as (deceptively?) simple as this or triggers the same bug in some other, equally simple way.

`sof-test/test-case/multiple-pause-resume.sh -r 25` has been repeatedly but inconsistently failing. It should probably be made more deterministic.

WYL-12, 2020-11-25:
I rerun the case on sh-byt-mb-nocodec-03 and sh-bdw-wsb-rt286-02, and result is still failed, but dmesg not found ipc error info.

sh-byt-mb-nocodec-03:
```
UTC [INFO] pipeline: Low Latency with arecord
UTC [INFO] pipeline: Media Playback 1 with aplay
UTC [INFO] Check expect exit status 
declare -- cmd="journalctl -k -q --no-pager --utc --output=short-monotonic --no-hostname --since=@1606267272"
UTC [INFO] pkill -TERM sof-logger
UTC [ERROR] sof-logger was already dead
UTC [INFO] nlines=1 /home/ubuntu/sof-test/logs/multiple-pause-resume/2020-11-25-01:20:18-279/slogger.txt

```
sh-bdw-wsb-rt286-02:
```
UTC [INFO] pipeline: Low Latency with aplay
UTC [INFO] pipeline: Media Playback 1 with aplay
UTC [INFO] Check expect exit status
UTC [ERROR] Starting func_exit_handler(), exit status=1, FUNCNAME stack:
UTC [ERROR]  main()  @  /home/ubuntu/sof-test/test-case/multiple-pause-resume.sh
UTC [INFO] Starting /usr/bin/sof-logger  -l /etc/sof/sof-bdw.ldc -o /home/ubuntu/sof-test/logs/multiple-pause-resume/2020-11-25-01:35:26-26974/etrace.txt
error: src hash value from version file (0x608e8d14) differ from src hash version saved in dictionary (0xdb1d7a4f).
UTC [INFO] pkill -TERM sof-logger
UTC [ERROR] sof-logger was already dead
UTC [INFO] nlines=1 /home/ubuntu/sof-test/logs/multiple-pause-resume/2020-11-25-01:35:26-26974/etrace.txt
```

marc-hb, 2020-11-25:
@WYL-12 I suspect the test is failing for another reason - for instance your sof-logger seems broken.

Can you try with the `-s` flag, i.e. `./test-case/multiple-pause-resume.sh -s` ?

WYL-12, 2020-11-25:
I rerun the case(multiple-pause-resume.sh) with the `-s` flag on sh-byt-mb-nocodec-03 and sh-bdw-wsb-rt286-02.

sh-byt-mb-nocodec-03:
rerun more than 5 times, all passed.

sh-bdw-wsb-rt286-02:
rerun more than 5 times, random fail, dmesg not found ipc error info.
fail info:
```
UTC [INFO] pipeline: Low Latency with aplay
UTC [INFO] pipeline: Media Playback 1 with aplay
UTC [INFO] Check expect exit status
UTC [ERROR] Starting func_exit_handler(), exit status=1, FUNCNAME stack:
UTC [ERROR]  main()  @  /home/ubuntu/sof-test/test-case/multiple-pause-resume.sh
```

marc-hb, 2020-11-25:
> dmesg not found ipc error info.

How many lines did you scroll up? On my system the IPC error is about 50 lines away.

Please also try https://github.com/thesofproject/sof-test/pull/534

WYL-12, 2020-11-25:
@marc-hb  rerun test case with #3618，it seems fixed this issue and dmesg not found ipc error info.

marc-hb, 2020-11-25:
> @marc-hb rerun test case with #3618，it seems fixed this issue and dmesg not found ipc error info.

You wrote that the test did not reproduce this bug on your system anyway, so this last run doesn't prove anything.

mengdonglin, 2020-12-01:
@WYL-12 confirmed this issue cannot be reproduced after applying PR#3618. So will wait for PR #3618 is merged and then close this issue.

marc-hb, 2020-12-02:
While not deterministic, @mengdonglin do you trust `multiple-pause-resume.sh` to reproduce frequently enough?

marc-hb, 2021-01-16:
@lgirdwood , @plbossart , @dbaluta , @cujomalainey , others: can we stop github from auto-closing bugs when merging commits with `Fixes:` tags? Auto-closing bugs on merges makes sense when you have:
- super deterministic software
- super deterministic, fully automated and extensive test coverage run on every Pull Request
- no hardware or operating system heavily involved (cause they're not deterministic)
- no other concurrent moving parts

Some projects have all this but we are very often very far from that, especially with an intermittent bug like this one that does not even have a dedicated test reproducing easily.

If this bug (for instance) is not completely fixed and validation or anyone else sees it fail again then they should be able to browse the list of _open_ bugs and find it instantly. Doesn't work if github auto-closed it.

If this bug is completely fixed then it's very little effort to close it after a few days of continuously successful validation. Takes 10 seconds in a bug scrub.

I think closing bugs is too important to be left to a robot.

cujomalainey, 2021-01-16:
> Only manually linked pull requests can be manually unlinked. To unlink an issue that you linked using a keyword, you must edit the pull request description to remove the keyword.

From (here)[https://docs.github.com/en/github/managing-your-work-on-github/linking-a-pull-request-to-an-issue]. So appears the only solution is to no have someone put in the keyword

lgirdwood, 2021-01-18:
@marc-hb this is not an auto-closed bug. @slawblauciak had confirmed fix. 
It sounds like you are suggesting we need an extra step where someone from CI confirms that the test for bug X has passed CI for the past Y days and than can either label the issue or close the issue ? This labelling confirmation could also be automated if we could link bugs to CI in GH somehow ???

marc-hb, 2021-01-18:
> @marc-hb this is not an auto-closed bug. @slawblauciak had confirmed fix.

Sorry, I thought it was auto-closed by github because of the timing and because there is (still) no clear indication of what testing was performed.

> It sounds like you are suggesting we need an extra step where someone from CI confirms that the test for bug X has passed CI for the past Y days and than can either label the issue or close the issue ?

Yes some extra step like this but only for bugs with no deterministic CI reproduction like this one. Simpler bugs can be closed immediately.

> So appears the only solution is to not have someone put in the keyword

OK, then it's never going to work :-( We'll just have to not blindly trust the "closed" status of bugs.

marc-hb, 2021-01-19:
`multiple-pause-resume.sh` failed again in internal nightly validation run 1679 and the failure was linked to this bug. It failed on only one device. It was validating most recent commit e17b90d9b2d6 which includes tentative fix #3618 

There are plenty of logs but I don't know enough about this bug to tell from them if this failure was really it.

lgirdwood, 2021-01-19:
@marc-hb so do you have any thoughts on how we could reference a GH issue against a CI test. We already have an interface where we can tell CI to test a PR via a comment, but I've no idea if this can be done for bugs e.g.
```SOFCI test=id1,id2(cml)```
Which could tell CI that this issue is tested by tests id1 (all platforms) and id2 (on cml) and then CI could report once it was passing for a time period.

marc-hb, 2021-01-19:
I don't know how we would do this and I'm sorry I derailed this (important) bug. I don't think we're struggling and need to some automation to help us close bugs. So far I only observed the opposite: bugs closed  (by github or manually) without process or validation details. Today's example: https://github.com/thesofproject/linux/issues/1449#issuecomment-556109925

cujomalainey, 2021-01-19:
> OK, then it's never going to work :-( We'll just have to not blindly trust the "closed" status of bugs.

You can tell if it was closed automatically by the closing line. E.g.
Automatically `jajanusz closed this in #2148 on Dec 16, 2019`
Manual  `cujomalainey closed this on Dec 3, 2020`

marc-hb, 2021-01-21:
> multiple-pause-resume.sh failed again in internal nightly validation run 1679 and the failure was linked to this bug. It failed on only one device.

This is apparently not #3522 after all but #3714

I still feel like the testing for #3522 is scant but at least it didn't seem to have surfaced recently.

slawblauciak, 2021-02-26:
Is this issue reproducible still?

marc-hb, 2021-02-27:
Hard to tell as long a `multiple-pause-resume.sh` is still not reliable for other reasons like #3714 or #3837.

`multiple-pause-resume.sh` was barely reproducing this bug in the first place and I haven't seen any other test for it, is there?

plbossart, 2021-06-10:
@lgirdwood @marc-hb I haven't seen any failures for pause/resume tests on platforms with a mixer, should we close this?

[2 more replies]

PR #3618 (2021-01-15): mixer: dont stop mixer on pause
audio_stream: add function writing zeros to the buffer without reference to a source buffer
mixer: dont stop mixer on pause and write zeros if all sources are inactive

Fixes #3522
closed 2021-06-10 by plbossart