thesofproject/sof #3523: [FEATURE] sof-logger -C (clear existing traces)

plbossart, 2020-10-14:
**Is your feature request related to a problem? Please describe.**
When running the same test multiple times, the old traces are still visible in the output of sof-logger. It would be nice to have the moral equivalent of 'dmesg -C' to discard all existing traces.

**Describe the solution you'd like**
A command-line option to discard trace and etrace.

**Describe alternatives you've considered**
Test script to only show the output of sof-logger after the first N lines.

**Additional context**
This is for developers who are looping on the same test multiple times (e.g. to flush out random behavior). You really want to see at each trace if something new appears.

lgirdwood, 2020-10-19:
@mwasko @slawblauciak should be simple enough todo, lets add for v1.7. Can you assign someone.

lgirdwood, 2021-01-12:
@plbossart is restarting loger enough to flush the old data out?

mwasko, 2021-01-25:
@bkokoszx will take a look and evaluate if we can deliver it for 1.7

lgirdwood, 2021-05-10:
@marc-hb any of your updates fix this ?

marc-hb, 2021-05-10:
> A command-line option to discard trace and etrace.

The DMA trace has always been working like this, hasn't it? I mean if you quit the logger and run it again then it only shows new traces. So this is an issue only for the shared memory, "error" trace, correct?

> @marc-hb any of your updates fix this ?

Unfortunately not, however some of the timestamps and messages I added should help a bit.

kv2019i, 2023-02-15:
Closing this. Focus on new feature development moved to Zephyr logging infrastructure and no new features planned for sof-logger.

closed 2023-02-15 by kv2019i