thesofproject/sof #3529: [BUG]  Audio Card Did Not Detected on MSI Prestige 15 ALC298

ErnyTech, 2020-10-17:
**Describe the bug**
SOF audio card didn't detected on MSI Prestige 15 (ALC298). I added blacklist snd-hda-intel in alsa-base.conf, input and output audio is not working (there is no audio card detected).

**To Reproduce**
- Install Arch Linux
- Install sof-firmware using pacman
- Add alsa-base.conf on /etc/modprobe.d/
- Reboot

**Reproduction Rate**
Always

**Expected behavior**
Audio Card is detected and Internal Mic is working

**Impact**
Can't use internal mic. if snd-hda-intel blacklisted input and output audio did not work.

**Environment**
Kernel: 5.9.1-arch1-1
SOF: 1.6-1
CPU: i7-10710U (Comet Lake)

**Screenshots or console output**
```
hwinfo --sound
16: PCI 1f.3: 0403 Audio device                                 
  [Created at pci.386]
  Unique ID: nS1_.VN8hBoVpR04
  SysFS ID: /devices/pci0000:00/0000:00:1f.3
  SysFS BusID: 0000:00:1f.3
  Hardware Class: sound
  Device Name: "Onboard - Sound"
  Model: "Intel Audio device"
  Vendor: pci 0x8086 "Intel Corporation"
  Device: pci 0x02c8 
  SubVendor: pci 0x1462 "Micro-Star International Co., Ltd. [MSI]"
  SubDevice: pci 0x129c 
  Memory Range: 0xd5410000-0xd5413fff (rw,non-prefetchable,disabled)
  Memory Range: 0xd5100000-0xd51fffff (rw,non-prefetchable,disabled)
  IRQ: 255 (no events)
  Module Alias: "pci:v00008086d000002C8sv00001462sd0000129Cbc04sc03i00"
  Driver Info #0:
    Driver Status: snd_hda_intel is not active
    Driver Activation Cmd: "modprobe snd_hda_intel"
  Driver Info #1:
    Driver Status: snd_soc_skl is not active
    Driver Activation Cmd: "modprobe snd_soc_skl"
  Driver Info #2:
    Driver Status: snd_sof_pci is active
    Driver Activation Cmd: "modprobe snd_sof_pci"
  Config Status: cfg=new, avail=yes, need=no, active=unknown
```

```
arecord -l
arecord: device_list:274: no soundcards found...
```

```
ls -l /lib/firmware/intel/sof*
lrwxrwxrwx 1 root root   13  3 ott 17.00 /lib/firmware/intel/sof-tplg -> sof-tplg-v1.6

/lib/firmware/intel/sof:
totale 36
lrwxrwxrwx 1 root root  33  3 ott 17.00 sof-apl.ri -> v1.6/intel-signed/sof-apl-v1.6.ri
lrwxrwxrwx 1 root root  20  3 ott 17.00 sof-bdw.ri -> v1.6/sof-bdw-v1.6.ri
lrwxrwxrwx 1 root root  20  3 ott 17.00 sof-byt.ri -> v1.6/sof-byt-v1.6.ri
lrwxrwxrwx 1 root root  33  3 ott 17.00 sof-cfl.ri -> v1.6/intel-signed/sof-cnl-v1.6.ri
lrwxrwxrwx 1 root root  20  3 ott 17.00 sof-cht.ri -> v1.6/sof-cht-v1.6.ri
[... 214 lines cut ...]
snd_sof_pci            24576  0
snd_sof_intel_byt      20480  1 snd_sof_pci
snd_sof_intel_ipc      20480  1 snd_sof_intel_byt
snd_sof_intel_hda_common    90112  1 snd_sof_pci
snd_soc_hdac_hda       24576  1 snd_sof_intel_hda_common
snd_sof_xtensa_dsp     16384  2 snd_sof_intel_hda_common,snd_sof_intel_byt
snd_sof_intel_hda      20480  1 snd_sof_intel_hda_common
snd_sof               135168  4 snd_sof_pci,snd_sof_intel_hda_common,snd_sof_intel_byt,snd_sof_intel_ipc
snd_hda_ext_core       36864  3 snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_sof_intel_hda
snd_soc_acpi_intel_match    45056  2 snd_sof_pci,snd_sof_intel_hda_common
snd_soc_acpi           16384  3 snd_soc_acpi_intel_match,snd_sof_intel_hda_common,snd_sof_intel_byt
snd_intel_dspcfg       24576  2 snd_sof_pci,snd_sof_intel_hda_common
snd_hda_codec         167936  1 snd_soc_hdac_hda
snd_hda_core          106496  5 snd_hda_ext_core,snd_hda_codec,snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_sof_intel_hda
snd_hwdep              16384  1 snd_hda_codec
ledtrig_audio          16384  1 snd_sof
snd_soc_core          327680  3 snd_sof,snd_sof_intel_hda_common,snd_soc_hdac_hda
snd_compress           32768  1 snd_soc_core
ac97_bus               16384  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_pcm               147456  7 snd_hda_codec,snd_sof,snd_sof_intel_hda_common,snd_compress,snd_soc_core,snd_hda_core,snd_pcm_dmaengine
snd_timer              45056  1 snd_pcm
snd                   114688  6 snd_hwdep,snd_hda_codec,snd_timer,snd_compress,snd_soc_core,snd_pcm
soundcore              16384  1 snd
```

plbossart, 2020-10-19:
@ErnyTech would you mind following the suggestions in https://github.com/thesofproject/sof-docs/pull/296/commits/fd1353dfaf93a05a2b26b40e3ea94cf749d35010

We tried to provide some guidance on what we'd need to help you out, the information you posted above doesn't tell us much. 
I realize the documentation is a works-in-progress, but we need to start ironing it out with real examples.

Thanks!

ErnyTech, 2020-10-19:
Sorry, I found that DSP was not enabled by UEFI settings.
Unfortunately, I had not found clear information about DSP before the documentation he provided me.

Thanks, now it works correctly

closed 2020-10-19 by ErnyTech