thesofproject/sof #3534: [BUG] [CHT] Underflow/errors reported in multiple-playback CI tests.

plbossart, 2020-10-19:
**Describe the bug**

https://sof-ci.01.org/linuxpr/PR2500/build4700/devicetest/

reports a failure on Chromebook Cyan on multi-playback cases.

This looks like an underflow condition that isn't well handled.

Filing as a bug to keep a trace of this.

plbossart, 2020-10-19:
Re-running the tests shows no issue reported (https://sof-ci.01.org/linuxpr/PR2500/build4703/devicetest/), so this is a transient error condition.

xiulipan, 2020-10-20:
@plbossart will this be related with the IPC position update delay? We do see such write I/O error on legacy platforms(BYT/CHT/BDW) using IPC position update.
If enable IPC position update for HDA platforms, we will also see such kind of I/O error.

lgirdwood, 2020-10-20:
@plbossart @xiulipan we should not be reading back position via IPC but reading back from SRAM window on CHT

plbossart, 2020-10-20:
@lgirdwood can you double-check the linux driver and let me know if we are doing things the right way? This position report has been a problem for the longest time but only you and @keyonjie understand it.

lgirdwood, 2020-10-20:
@plbossart pointer callback is not defined for BYT/CHT so it's using mailbox and NOT IPC method.
```c
static snd_pcm_uframes_t sof_pcm_pointer(struct snd_soc_component *component,
					 struct snd_pcm_substream *substream)
{
	struct snd_soc_pcm_runtime *rtd = asoc_substream_to_rtd(substream);
	struct snd_sof_dev *sdev = snd_soc_component_get_drvdata(component);
	struct snd_sof_pcm *spcm;
	snd_pcm_uframes_t host, dai;

	/* nothing to do for BE */
	if (rtd->dai_link->no_pcm)
		return 0;

	/* use dsp ops pointer callback directly if set */
	if (sof_ops(sdev)->pcm_pointer)
		return sof_ops(sdev)->pcm_pointer(sdev, substream);

	spcm = snd_sof_find_spcm_dai(component, rtd);
	if (!spcm)
		return -EINVAL;

	/* read position from DSP */
	host = bytes_to_frames(substream->runtime,
			       spcm->stream[substream->stream].posn.host_posn);
	dai = bytes_to_frames(substream->runtime,
			      spcm->stream[substream->stream].posn.dai_posn);

	dev_vdbg(component->dev,
		 "PCM: stream %d dir %d DMA position %lu DAI position %lu\n",
		 spcm->pcm.pcm_id, substream->stream, host, dai);

	return host;
}
```

plbossart, 2020-10-20:
> @plbossart pointer callback is not defined for BYT/CHT so it's using mailbox and NOT IPC method.

ok, so now I have no idea why we only look at the position during an IPC and why we enabled the BATCH option to disable timer-based scheduling in PulseAudio.

lgirdwood, 2020-10-21:
> > @plbossart pointer callback is not defined for BYT/CHT so it's using mailbox and NOT IPC method.
> 
> ok, so now I have no idea why we only look at the position during an IPC and why we enabled the BATCH option to disable timer-based scheduling in PulseAudio.

Might be worthwhile enabling the trace here in the kernel (as FW could be sending garbage at time or sending for wrong stream).

plbossart, 2020-10-21:
@lgirdwood the DMA trace is always enabled in our tests.

lgirdwood, 2020-10-21:
@plbossart sorry I mean from above kernel code (which may be off as it's verbose)
```c
dev_vdbg(component->dev,
		 "PCM: stream %d dir %d DMA position %lu DAI position %lu\n",
		 spcm->pcm.pcm_id, substream->stream, host, dai);

```

mengdonglin, 2022-06-21:
Move CHT platform to maintenance mode and SOF v2.2 will be the last upstream release with support for this platform. There will be no more FW bug fixing effort on this platform.

closed 2022-06-21 by mengdonglin