thesofproject/sof #3541: [TGL] [I2S/SDW][DSM]: sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76

naveen-manohar, 2020-10-21:
Observe following Log randomly, during stress cycling.. Speaker Audio-Functionality post this scenario is yet to be verified as its a cold-reboot stress test.
Log: [system_logs.txt](https://github.com/thesofproject/linux/files/5285932/system_logs.txt)
```
[ 1432.644301] max98373 sdw:1:19f:8373:0:7: ASoC: error at soc_component_read_no_lock on sdw:1:19f:8373:0:7: -16
[ 1432.644318] max98373 sdw:1:19f:8373:0:7: ASoC: error at soc_component_read_no_lock on sdw:1:19f:8373:0:7: -16
[ 1432.644331] max98373 sdw:1:19f:8373:0:7: ASoC: error at soc_component_read_no_lock on sdw:1:19f:8373:0:7: -16
[ 1432.644563] max98373 sdw:1:19f:8373:0:3: ASoC: error at soc_component_read_no_lock on sdw:1:19f:8373:0:3: -16
[ 1432.644576] max98373 sdw:1:19f:8373:0:3: ASoC: error at soc_component_read_no_lock on sdw:1:19f:8373:0:3: -16
[ 1432.644586] max98373 sdw:1:19f:8373:0:3: ASoC: error at soc_component_read_no_lock on sdw:1:19f:8373:0:3: -16
[ 1432.748544] max98373 sdw:1:19f:8373:0:7: ASoC: error at soc_component_read_no_lock on sdw:1:19f:8373:0:7: -16

```
Recipe Based on SoF_V0.8/0.9 Release
Hardware: Volteer Proto Board

naveen-manohar, 2020-10-22:
@RDharageswari shared a patch & DSP-firmware to free-up maxim resource at stop, and with this Issue is not reproducible on SDW SKU 10/10 times.
Log: [dsm_free_fw-dmesg.lo.txt](https://github.com/thesofproject/sof/files/5421070/dsm_free_fw-dmesg.lo.txt) FW:
[sof-tgl_resigned_fix_issue.zip](https://github.com/thesofproject/sof/files/5421078/sof-tgl_resigned_fix_issue.zip)

```
2020-10-22T12:32:31,507438+05:30 sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
2020-10-22T12:32:31,563133+05:30 sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000
2020-10-22T12:32:31,563374+05:30 sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40010000
2020-10-22T12:32:31,563538+05:30 sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x9010f0f successful
2020-10-22T12:32:31,563554+05:30 sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
2020-10-22T12:32:31,563556+05:30 sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
2020-10-22T12:32:31,563611+05:30 sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
2020-10-22T12:32:31,563618+05:30 sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000010 at  00000044
2020-10-22T12:32:31,563623+05:30 sof-audio-pci 0000:00:1f.3: Turning i915 HDAC power 0
2020-10-22T12:32:31,563624+05:30 sof-audio-pci 0000:00:1f.3: Current DSP power state: D3

```
Tried Speaker functionality between suspend-resume & with my limited test, it worked fine..

RanderWang, 2020-10-22:
great!

plbossart, 2020-10-22:
> @RDharageswari shared a patch & DSP-firmware to free-up maxim resource at stop, and with this Issue is not reproducible on SDW SKU 10/10 times.

Was this patch shared @naveen-manohar @RDharageswari ?

naveen-manohar, 2020-10-24:
> > @RDharageswari shared a patch & DSP-firmware to free-up maxim resource at stop, and with this Issue is not reproducible on SDW SKU 10/10 times.
> 
> Was this patch shared @naveen-manohar @RDharageswari ?

Now I notice that @RDharageswari is tracking the DSP FW issue caused due to DSM is being updated here - https://github.com/thesofproject/sof/issues/3538#issuecomment-714709012

lgirdwood, 2020-10-29:
@naveen-manohar @RDharageswari is there a PR for the fix ?

keyonjie, 2020-11-02:
since we have track this on the FW side #3538, let's close this one for driver, @naveen-manohar please re-open it if needed.

closed 2020-11-02 by keyonjie

marc-hb, 2020-11-02:
> since we have track this on the FW side #3538, let's close this one for driver, @naveen-manohar please re-open it if needed.

This bug is located in `sof`, it's not in `linux`.

In #3541 @RDharageswari does not think  they're duplicates.

Even if they were duplicates, this one seems to have much more information.