thesofproject/sof #3546: [BUG] TGL build is not producing ri files via docker

cujomalainey, 2020-10-22:
**Describe the bug**
APL builds fine here is end of build log
```
[100%] Linking C executable sof
Incrementing build number in /home/sof/work/sof.git/build_tgl_gcc/.build
[100%] Built target sof
ls: cannot access 'build_*/*.ri': No such file or directory
ls: cannot access 'build_*/src/arch/xtensa/rom*.bin': No such file or directory
-rwxr-xr-x 1 sof sof 3309296 Oct 22 21:25 build_tgl_gcc/sof
```

Building on Pop-os using docker

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)
`./scripts/docker-run.sh ./scripts/xtensa-build-all.sh tgl`

**Reproduction Rate**
100%

**Expected behavior**
ri files built

**Impact**
showstopper

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: N/A
    * SOF: e04610b19754d0f402b3b76b0a7fb4fe24d9aa18
2) Name of the topology file
    * Topology: N/A
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL

**Screenshots or console output**
see above

cujomalainey, 2020-10-22:
Looks like CI might also be silently passing https://travis-ci.org/github/thesofproject/sof/jobs/738004861#L479

Return code in my test from the build phase was 0, submodules were up to date

juimonen, 2020-10-23:
@cujomalainey currently the tgl builds needs different signing tool, which I understand is not public (meu tool?). Also on most of our tgl devices we have either intel key or debug key, so you can't sign them with the public otc key.

```
export PRIVATE_KEY_OPTION=-DMEU_PRIVATE_KEY=/home/sof/work/sof.git/MEU15_TGL/private.pem
./scripts/docker-run.sh ./scripts/xtensa-build-all.sh -m /home/sof/work/sof.git/MEU15_TGL/Linux64 tgl
```

So I first export the path to the private key and then specify the path to meu tool.

slawblauciak, 2020-10-27:
Fixing this will require openly releasing the MEU tool to the public.
It's currently in progress.

lgirdwood, 2020-10-27:
@slawblauciak we will need to release both Linux and Windows binaries for this tool.

cujomalainey, 2020-12-03:
@lgirdwood this is fixed correct?

lgirdwood, 2020-12-03:
@cujomalainey should be, rimage can now sign for TGL and it's be merged into sof master. Maybe we are missing something in Docker ? @xiulipan ?

cujomalainey, 2020-12-03:
I just figured. haven't tested in a while, will test and close

closed 2020-12-03 by cujomalainey

xiulipan, 2020-12-04:
@lgirdwood @cujomalainey confirm fixed, will also try to refine CI build steps.