thesofproject/sof #3551: [BUG]Firmware info shows compiler is GCC for XCC build binary

xiulipan, 2020-10-23:
**Describe the bug**
`Firmware info: used compiler` shows wrong compiler for XCC build binary now.
Seems to be regression with https://github.com/thesofproject/sof/pull/3496

**To Reproduce**
```
dmesg | grep "Firmware info"
[  777.050607] sof-audio-pci 0000:00:0e.0: Firmware info: used compiler GCC 4:2:0 <RG-2017.8-linux> used optimization flags -O2
[  777.087560] sof-audio-pci 0000:00:0e.0: Firmware info: version 1:5:0-ed219
```

**Reproduction Rate**
3/3

**Expected behavior**
For GCC FW we should have
 
```
dmesg |grep "Firmware info"
[    4.908657] sof-audio-pci 0000:00:0e.0: Firmware info: used compiler GCC 9:2:0 xtensa-apl-elf used optimization flags -O2
[    4.908692] sof-audio-pci 0000:00:0e.0: Firmware info: version 1:5:0-ed219
```
For XCC FW we should have
```
[  777.050607] sof-audio-pci 0000:00:0e.0: Firmware info: used compiler XCC 12:0:8 <RG-2017.8-linux> used optimization flags -O2
[  777.087560] sof-audio-pci 0000:00:0e.0: Firmware info: version 1:5:0-ed219
```

**Impact**
The firmware info shows in dmesg maybe mislead.

**Environment**
Test with ed21921878fc171253a4dfaa0f303d02cf17fffd FW on an APL platform

xiulipan, 2020-10-23:
You can check some latest CI result for PR: https://sof-ci.01.org/sofpr/PR3544/build7509/devicetest/

slawblauciak, 2020-10-27:
This should fix it: https://github.com/thesofproject/sof/pull/3555

closed 2020-10-29 by lgirdwood

mengdonglin, 2020-11-02:
@lgirdwood Do you want to merge this fix for SOF 1.6 release?