thesofproject/sof #3554: [BUG] TGL: WOV power regression in V0.10 release with  97377c3757e9 smart_amp: fix smart amplifier tplg

jairaj-arava, 2020-10-27:
**Describe the bug**
With V0.10 release the WOV power consumption is high with  97377c3757e9 smart_amp: fix smart amplifier tplg. 

Without reverting the patch:
   SOC Power= 39.6 mw
   Total system power=  54.5 mw

With reverting the patch:
          SOC Power= 19.3mw
         Total system power=  29.2 mw

**To Reproduce**
1. Run the WOV and keep the system in S0IX and measure the powers.

**Reproduction Rate**
Always

**Expected behavior**
Power consumption should be within the range.

**Impact**
High power consumption by the system with WOV, hence impact on the battery life.

jairaj-arava, 2020-10-27:
@keyonjie and @ranj063, helped us to narrow down the issue to the motioned patch in the topology. 
We didn't see any impact in the speaker and other end points functionalities but the power consumption is higher (~24mw) with this patch during S0IX with WOV. However, by reverting the patch, the value dropped down to 29.2mw.

@abonislawski, has given a test version but we see the same behavior.

plbossart, 2020-10-27:
well d'uh. if the 24.576MHz PLL is used of course there will be more power burned than if an integer division of the clock is used. This PLL was never supposed to be used in S0i1 IIRC.

I wonder if we could still keep it, but modify the clock sources when the SSP is stopped. Currently we set the clock sources once and that means the PLL will remain on forever. Or we'd need a real clock driver where the clock is disabled when there's no need for it.

lgirdwood, 2020-10-29:
> I wonder if we could still keep it, but modify the clock sources when the SSP is stopped. Currently we set the clock sources once and that means the PLL will remain on forever. Or we'd need a real clock driver where the clock is disabled when there's no need for it.

Ack - ASoc does this for all codecs. Switches any fractional PLL or FLL off when not needed.

mwasko, 2020-10-29:
> Ack - ASoc does this for all codecs. Switches any fractional PLL or FLL off when not needed.

The issue is that with current SOF design with statically created pipelines we keep the SSP clocks up & running. If we would dynamically create & destroy pipelines then there will be no problem.

ranj063, 2020-10-29:
> The issue is that with current SOF design with statically created pipelines we keep the SSP clocks up & running. If we would dynamically create & destroy pipelines then there will be no problem.

@mwasko I have a PR with dynamic pipeline loading https://github.com/thesofproject/linux/pull/2371. It is still under local test on my part but do you think this will help?

plbossart, 2020-10-29:
> The issue is that with current SOF design with statically created pipelines we keep the SSP clocks up & running. If we would dynamically create & destroy pipelines then there will be no problem.

I don't think it has to do with fixed/dynamic pipelines, rather that we don't stop the clocks when the SSP is no longer used. Destroying the pipeline is a sledgehammer way of doing the same thing. It's like removing a module to test pm_runtime :-)

mwasko, 2020-11-02:
> @mwasko I have a PR with dynamic pipeline loading [thesofproject/linux#2371](https://github.com/thesofproject/linux/pull/2371). It is still under local test on my part but do you think this will help?

Yes, this should resolve the problem with SSP PLL draining power. However since it is quite a big design change it may be risky to integrate it for current TGL release.  

Let us first investigate option to move SSP clock enable to component PREPARE state, which possibly should address the regression as well.

mengdonglin, 2020-11-30:
Fixed by release branch https://github.com/thesofproject/sof/tree/tgl-011-drop-stable, RC4

closed 2020-11-30 by mengdonglin