thesofproject/sof #3570: [BUG][APL] Can't build SOF APL with probes.

zrombel, 2020-10-30:
**Describe the bug**
Can not build SOF for APL platform with PROBES configurated due to linker error.

**To Reproduce**
Build APL target with apollolake_defconfig and additional options:
CONFIG_DEBUG=y
CONFIG_PROBE=y
CONFIG_PROBE_POINTS_MAX=16
CONFIG_PROBE_DMA_MAX=4

**Reproduction Rate**
100%

**Environment**
SOF Branch: Master, Hash: 43124daf39a961fa015761ad59ba3d6f19be7dfd and later.

**Screenshots or console output**
Linker error:
RG-2017.8-linux/XtensaTools/bin/xt-ld: sof section `.bss' will not fit in region `sof_fw'

slawblauciak, 2020-11-03:
@jdrzaszcz please check if you can find the FW change that introduced this regression in CI tests.
@lyakh please try to remove excess XT OS files, which are not needed for this build

lyakh, 2020-11-03:
> @lyakh please try to remove excess XT OS files, which are not needed for this build

@slawblauciak I meant #3557 so maybe someone could check whether it gets us any memory back or I can check it myself later

jdrzaszcz, 2020-11-03:
@slawblauciak  On CI, the last debug build for APL platform was done on commit:7709924f16accd45fc2b159d983d84f6e994afd6, in the next build (commit: e04610b19754d0f402b3b76b0a7fb4fe24d9aa18) FW was not compiled.
When I tried to build it by myself, last commit where I can build FW for APL with such a config was commit:901e7b494512b64da80b30c3cd2f2e617fbf7732

lgirdwood, 2020-11-03:
My expectation is that new features may outgrow older platforms. The safest option is to use less probe points on APL or disable it in kconfig for APL.

closed 2025-07-04 by github-actions[bot]