thesofproject/sof #3588: [BUG] [SDW] [TGL] MX98373 feedback capture should follow VIVI order

yongzhi1, 2020-11-05:
**Describe the bug**
Run echo ref capture with MAX98373, and examine the recorded data from Audacity. It shows the feedback data follows RLLR/IVIV instead of LLRR/VIVI.

(Same pattern shown in probes-captured data from BUF4.1 before entering MEXDEMUX4.0)

**To Reproduce**

    Play the following URL with speaker:
    https://www.youtube.com/watch?v=6TWJaFD6R2s
    arecord -Dhw:0,3 -f S32_LE -r48000 -c4 -d100 /home/LR-test.wav -vvv

**Reproduction Rate**
100%

**Expected behavior**
The feedback data is expected to adhere V-I-V-I pattern with L group come first. (my understanding based on SSP)   

**Environment**
        Kernel: https://chromium.googlesource.com/chromiumos/third_party/kernel/+/ceab26b0e4f27f44465d88799119dc4572cefe2b
        SOF: e0d648b (tag: releases/tgl/v10.0-rc1) codec_adapter: make certain .copy logs verbose
        Topology: sof-tgl-sdw-max98373-rt5682.m4
        Platform: TGL

**Screenshots or console output**

[image]

plbossart, 2020-11-05:
@yongzhi1 how do you know it's a firmware issue and not a codec configuration issue? What is the configuration on the SoundWire TX ports?

yongzhi1, 2020-11-05:
Hi @plbossart 

Have to admit I do not know whether it's codec config related or now, here is the related code on port 3 programming

```
	} else {
		/* only IV are supported by capture */
		if (direction == SDW_DATA_DIR_TX)
			stream_config.ch_count = 2;
		else
			stream_config.ch_count = params_channels(params);

		port_config.ch_mask = GENMASK((int)stream_config.ch_count - 1, 0);
	}
```
(from https://github.com/thesofproject/linux/blob/topic/sof-dev/sound/soc/codecs/max98373-sdw.c#L550)

cc @ryans-lee

slawblauciak, 2020-11-06:
Please verify codec configuration in alsamixer. You might be able to change it to LLRR there.

RanderWang, 2020-11-09:
[rander_amixer.txt](https://github.com/thesofproject/sof/files/5509174/rander_amixer.txt)
DSM works on my device, please check my amixer setting.  @yongzhi1 you changed channel number from 4 to 2 in topology why reported a bug with 4 channel ?

yongzhi1, 2020-11-09:
Thanks @slawblauciak and @RanderWang for the helpful advices.

Sorry for the confusion about SMART_REF_CH_NUM  I should mention that the issue was observed for SMART_FB_CHANNELS after https://github.com/thesofproject/sof/commit/2de506025f8564dd5421ecee0511b6a51dcead got merged.

My amixer settings do not have 'SELECTOR13.0 SELECTOR', there is no other critical differences to my eyes.

Hi, @RanderWang , can you clarify that the feedback data was captured in VIVI order with default 'MUXDEMUX4.0 DEMUX' TLV data?

RanderWang, 2020-11-10:
> Thanks @slawblauciak and @RanderWang for the helpful advices.
> 
> Sorry for the confusion about SMART_REF_CH_NUM I should mention that the issue was observed for SMART_FB_CHANNELS or BUF4.1 after [2de5060](https://github.com/thesofproject/sof/commit/2de506025f8564dd5421ecee0511b6a51dcead) got merged.
> 
> My amixer settings do not have 'SELECTOR13.0 SELECTOR' and KWD, there is no other critical differences to my eyes.
> 
> Hi, @RanderWang , can you clarify that the feedback data was captured in VIVI order with default 'MUXDEMUX4.0 DEMUX' TLV data?

@yongzhi1 I never set 'MUXDEMUX4.0 DEMUX'. I sent a email and you can check it by yourself.

yongzhi1, 2020-11-10:
Hi, @RanderWang , thanks for sharing your setup, great appreciated.

I re-tested based on tgl-010-hot-fix-1 , commit: 83cb82d, the problem is no longer reproducible. 

When testing Maxim DSM with https://www.youtube.com/watch?v=6TWJaFD6R2s, it seems the VIVI order issue is related to data format, previously it's hardcoded to S32, now I can use S16 with https://github.com/thesofproject/sof/commit/717354e9e5f46754c694060a3e3d5c2e7b5851a9.

Thanks all for your support!!

closed 2020-11-10 by yongzhi1