thesofproject/sof #3590: [BUG][SSP][TGL] Channel swapped with trigger pause/stop/start on capture pipline.

jdrzaszcz, 2020-11-06:
**Describe the bug**
When running simultaneous playback and capture on external loopback (same SSP) with pause, stop and start operation are done on capture stream there are swapped channel after trigger start on captured wave file.

This commit: 242be345ca9f87db29b9cb4b12b91438d932814d fixed swapped channels with pause, stop and start operation on capture stream with dsm component on TGL platform, but problem is still occurs specially on Frame formats where: 
SAMPLE_SIZE=24 bits
CONTAINER_SIZE = 32 bits.

TGL results:
[image]

Example wave:
[image]

**Topology**
```
                pipe_plb
                +--------------------+
+----------+    | +------+   +-----+ |   +----+
| Host plb |----->|Buffer|-->| Dai |---->|SSPx|
+----------+    | +------+   +-----+ |   +----+
                +--------------------+    |  |
                pipe_cap                  |  |
                +--------------------+    |  |
+----------+    | +------+   +-----+ |   +----+
| Host cap |<-----|Buffer|<--| Dai |---->|SSPx|
+----------+    | +------+   +-----+ |   +----+
                +--------------------+
```
**To Reproduce**
Slim Driver python test:
test_20_ptp_cap_stream_stop_start_operations.py with --playback_iterations>=10

**Reproduction Rate**
On with Frame format 24/32 - 80%
On other Frame formats - 20%

**Expected behavior**
Channels shouldn't be swapped on recorded files.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
TGL RVP platforms.
SOF master, commit: 8a5059a8c0b1e2d67b404efe26e1b79bb6175b5b

**Example logs**
For TGL:
[02_20_TestPtpCaptureStreamStopStartOperation48000Hz24b32b2ch.zip](https://github.com/thesofproject/sof/files/5383867/02_20_TestPtpCaptureStreamStopStartOperation48000Hz24b32b2ch.zip)

keyonjie, 2021-06-11:
this should already be closed by https://github.com/thesofproject/sof/commit/60dfaa7ee30189c4e23fd392ae5f8f9b7bb21113

@jdrzaszcz @kkarask can you double check if we can close this?

kkarask, 2021-06-14:
@keyonjie I double check and problem still exists. It occurs with different Frame formats all with high reproduction rate ~80%. It happens together with glitches in wav file.