thesofproject/sof #3603: [BUG] tplg load failure on TGL Chromebook w/ Maxim codec in I2S mode

keqiaozhang, 2020-11-13:
**Describe the bug**
We set up a TGL chromebook with Maxin codec in I2S mode, but tplg failed to load.
error message:DSP failed to add widget id 0 type 33 name : SMART_AMP1.0 stream none reply 0

**To Reproduce**
boot the system and check the dmesg

**Reproduction Rate**
10/10

**Expected behavior**
tplg can be loaded successfully

**Impact**
TPLG failed to load and no pipelines can be created.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev/65a176da
    * SOF: master/324681ec
2) Name of the topology file
    * Topology: sof-tgl-max98373-rt5682.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL Chromebook

**Screenshots or console output**
```
[    6.144274] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:18:1 Kernel ABI 3:18:0
[    6.144276] sof-audio-pci 0000:00:1f.3: warn: topology ABI is more recent than kernel
[    6.144282] sof-audio-pci 0000:00:1f.3: tplg: ready widget id 0 pipe 1 type 23 name : PCM0P stream Smart Amplifier Playback 0
[    6.144285] sof-audio-pci 0000:00:1f.3: loaded host PCM0P
[    6.144287] sof-audio-pci 0000:00:1f.3:  config: periods snk 2 src 0 fmt 0
[    6.144293] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    6.152849] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    6.152907] sof-audio-pci 0000:00:1f.3: tplg: load control type 2 name : Smart_amp Config
[    6.152931] sof-audio-pci 0000:00:1f.3: tplg: load kcontrol index 1 chans 0
[    6.152958] sof-audio-pci 0000:00:1f.3: tplg: load control type 2 name : Smart_amp Model
[    6.153258] sof-audio-pci 0000:00:1f.3: tplg: load kcontrol index 1 chans 0
[    6.153285] sof-audio-pci 0000:00:1f.3: tplg: load control type 2 name : Smart_amp Model_Get_params
[    6.153582] sof-audio-pci 0000:00:1f.3: tplg: load kcontrol index 1 chans 0
[    6.153622] sof-audio-pci 0000:00:1f.3: tplg: ready widget id 1 pipe 1 type 33 name : SMART_AMP1.0 stream none
[    6.153718] sof-audio-pci 0000:00:1f.3:  config: periods snk 2 src 2 fmt 0
[    6.153780] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    6.158097] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x30010000 size 20
[    6.158564] sof-audio-pci 0000:00:1f.3: error: create process failed
[    6.159022] sof-audio-pci 0000:00:1f.3: error: process loading failed
[    6.159410] sof-audio-pci 0000:00:1f.3: error: DSP failed to add widget id 0 type 33 name : SMART_AMP1.0 stream none reply 0
[    6.160127] sof-audio-pci 0000:00:1f.3: ASoC: failed to load widget SMART_AMP1.0
[    6.160544] sof-audio-pci 0000:00:1f.3: ASoC: topology: could not load header: -22
[    6.161062] sof-audio-pci 0000:00:1f.3: error: tplg component load failed -22
[    6.161496] sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -22
[    6.161965] sof-audio-pci 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    6.162478] sof-audio-pci 0000:00:1f.3: ASoC: failed to probe component -22
[    6.163059] sof_rt5682 tgl_max98373_rt5682: ASoC: failed to instantiate card -22
[    6.164546] sof_rt5682: probe of tgl_max98373_rt5682 failed with error -22
```

**dmesg**
[tgl-chrome.txt](https://github.com/thesofproject/sof/files/5534270/tgl-chrome.txt)

ranj063, 2020-11-13:
@keqiaozhang do you have the sof-logger data?

bkokoszx, 2020-11-17:
@RDharageswari 
Could you try to reproduce it on your end?

keyonjie, 2020-11-18:
We need compile with the smart_amp component for this Chromebook, @keqiaozhang and @bkokoszx

bkokoszx, 2020-11-19:
@keyonjie @keqiaozhang 
As I understand issue is not reproducible when we use maxim library. Therefore, can we close it?

mengdonglin, 2020-11-30:
Closed as won't fix as Maxim DSM library will not be upstreamed.

closed 2020-11-30 by mengdonglin