thesofproject/sof #3605: [BUG] TGL Dell XPS 13 9310: Microphone/input does not work

paulmenzel, 2020-11-13:
With the Intel Tiger Lake laptop Dell XPS 13 9310 and Debian Sid/unstable with Linux 5.9.6, SOF *stable-v1.6* and PulseAudio 13.99.3-1, the user reports, that the internal microphone does not work. First the internal device was only detected after plugging in an external headset, but after [selecting `SND_SOC_SOF_TIGERLAKE_SUPPORT`](https://bugs.debian.org/973879) and installing the SOF *stable-v1.6* a multichannel device appears.

[image]

But the user says, nothing is detected/recorded over this microphone. PulseAudio is also version 13.99.3, which I understand has all necessary changes. How can I check that?

Maybe you see something in the [output of `/usr/sbin/alsa-info`](https://alsa-project.org/db/?f=af8b5e4b1e7aef1c4315aa9eee114f6b79ba355d). Also, an excerpt from the [Linux messages](https://github.com/thesofproject/sof/files/5537000/20201110-debian-linux-5.9.6-dmesg.txt):

```
[   10.132876] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[   10.132878] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   10.132888] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[   10.133059] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   11.347094] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   11.353897] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   11.375128] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[   11.375130] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   11.375132] sof-audio-pci 0000:00:1f.3: DMICs detected in NHLT tables: 2
[   11.377203] sof-audio-pci 0000:00:1f.3: firmware: direct-loading firmware intel/sof/sof-tgl.ri
[   11.481279] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:6:0-18fab
[   11.481281] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:17:0 Kernel ABI 3:16:0
[   11.481281] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
[   11.481959] sof-audio-pci 0000:00:1f.3: warning: unknown ext header type 3 size 0x1c
[   11.481975] sof-audio-pci 0000:00:1f.3: warning: unknown ext header type 4 size 0x10
[   11.534359] sof-audio-pci 0000:00:1f.3: firmware: direct-loading firmware intel/sof-tplg/sof-hda-generic-2ch.tplg
[   11.534364] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:17:0 Kernel ABI 3:16:0
[   11.534364] sof-audio-pci 0000:00:1f.3: warn: topology ABI is more recent than kernel
[   11.534367] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[   11.535687] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[   11.535687] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[   11.537332] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[   11.537333] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[   11.538577] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[   11.538578] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[   11.540163] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[   11.540164] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[   11.540165] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[   11.540166] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[   11.540166] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[   11.540167] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[   11.541412] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[   11.541412] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[   11.542717] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[   11.542717] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[   11.544011] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[   11.556552] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
```

plbossart, 2020-11-13:
That platform looks like a standard HDAudio + 2 mics + Intel HDMI. That should work, we have devices like this left and right.

is the alsa-ucm-conf package up-to-date? It should contain the UCM parts needed to show the dmic.

Also please refer to this documentation to help check dmic issues:

https://thesofproject.github.io/latest/getting_started/index.html#debugging-audio-issues-on-intel-platforms

paulmenzel, 2020-11-13:
> That platform looks like a standard HDAudio + 2 mics + Intel HDMI. That should work, we have devices like this left and right.

Where do you see that? There is no HDMI connector on the [Dell XPS 13 9310](https://www.dell.com/de-de/shop/laptops/xps-13-laptop/spd/xps-13-9310-laptop) to my knowledge – only USB 3 Type C. 

> is the alsa-ucm-conf package up-to-date? It should contain the UCM parts needed to show the dmic.

Debian packages *alsa-ucm-conf* 1.2.3-1.

> Also please refer to this documentation to help check dmic issues:
> 
> https://thesofproject.github.io/latest/getting_started/index.html#debugging-audio-issues-on-intel-platforms

I can try that on Monday, if the user allows me remote access. Please note, that my issue contains a reference to the output of `/usr/sbin/alsa-info`. The parts below look strange:

```
!!Aplay/Arecord output
!!--------------------

APLAY

aplay: device_list:274: no soundcards found...

ARECORD

arecord: device_list:274: no soundcards found...

!!Amixer output
!!-------------

!!-------Mixer controls for card sofhdadsp

Invalid card number.
Usage: amixer <options> [command]

Available options:
  -h,--help       this help
```

plbossart, 2020-11-13:
> Where do you see that? There is no HDMI connector on the [Dell XPS 13 9310](https://www.dell.com/de-de/shop/laptops/xps-13-laptop/spd/xps-13-9310-laptop) to my knowledge – only USB 3 Type C.

[   11.375128] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5

Mask = 0x4 means there is a 'Display Audio' connection handled by Intel. This is helpful to detect if there is an Nvidia/AMD external graphics card handling Display Audio separately. The actual format/connector is irrelevant, what matters is whether SOF handles this audio data or not.

> > is the alsa-ucm-conf package up-to-date? It should contain the UCM parts needed to show the dmic.
> 
> Debian packages _alsa-ucm-conf_ 1.2.3-1.

might be worth updating to 1.2.4, but not sure why that would be a problem.
 
> I can try that on Monday, if the user allows me remote access. Please note, that my issue contains a reference to the output of `/usr/sbin/alsa-info`. The parts below look strange:
>
> ``` [...]

Yes that's not good. This can happen if somehow the card creation fails despite the PCI probe happening without issues. The card creation failure is squelched by the device core.

maybe adding the following options in /etc/modprobe.d/sof-dyndbg.conf would help use figure out what happens

````
options snd_sof_intel_byt dyndbg=+p
options snd_sof_intel_bdw dyndbg=+p
options snd_sof_intel_ipc dyndbg=+p
options snd_sof_intel_hda_common dyndbg=+p
options snd_sof_intel_hda dyndbg=+p
options snd_sof dyndbg=+p
options snd_sof_pci dyndbg=+p
options snd_sof_acpi dyndbg=+p
options snd_sof_of dyndbg=+p
options snd_sof_nocodec dyndbg=+p
options soundwire_bus dyndbg=+p
options soundwire_generic_allocation dyndbg=+p
options soundwire_cadence dyndbg=+p
options soundwire_intel_init dyndbg=+p
options soundwire_intel dyndbg=+p
options snd_soc_skl_hda_dsp dyndbg=+p
options snd_intel_dspcfg dyndbg=+p
````

kv2019i, 2020-11-23:
@paulmenzel I have a 9310 and can confirm the digital microphone works with SOF without issues. Can you capture also the kernel config that is used?

It seems this is an issue with kernel probe. All components of the SOF PCI card do not get initialized and we are stuck in
[   11.556552] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred

... this could:
 - some problem with HDMI/DP codec driver (the kernel config might reveal something), I don't see any traces in the log
          - could be fixed by https://lore.kernel.org/r/20200921100841.2882662-1-kai.vehmanen@linux.intel.c 
 - bug in backwards compatibility (bug happens with older kernel and newer topology)

In any case, looks like a kernel bug, so moving to kernel and assigning to myself.