thesofproject/sof #3613: [BUG] wasted memory in DMA buffer allocations

lyakh, 2020-11-16:
#2425 changed DMA buffer size on all platforms from 0x180 to 0x100, which wastes 0x80 = 128 bytes every time a 0x180 buffer is allocated, as explained in detail in https://github.com/thesofproject/sof/pull/2425#issuecomment-727966533

plbossart, 2020-11-16:
@lyakh why would 256 be better than 384 for the block size? You would waste memory as well in other configurations.
Or put differently why not use 128 for the block size?

lyakh, 2020-11-17:
> @lyakh why would 256 be better than 384 for the block size? You would waste memory as well in other configurations.
> Or put differently why not use 128 for the block size?

@plbossart that's exactly my question, or to be more precise the essence of this bug. #2425 changed the block size from 384 to 256 and thus started wasting memory. My #3614 does exactly what you propose - use 128 byte blocks, but that fails to build on BDW and HSW - BSS size is exceeded, I guess an array of buffer descriptors or something exceeds the BSS now. So, maybe we have to go back to 384, i.e. basically revert https://github.com/thesofproject/sof/pull/2425/commits/c920e41b801c0167142a8847dd646976326bdd7a

plbossart, 2020-11-17:
@lyakh what prevents you from using 128 for all HDaudio since we know it's a required alignment, and keep 256 for legacy platforms? we don't need to have a one-size-fits-all approach.
I don't know why changing the block allocation would result in a BSS size exceeded btw,  BSS as I understand it is for statically allocated stuff, and this is dynamic memory?

lgirdwood, 2020-11-18:
@lyakh we may not want to spend too much time on this since it does not seem to be a blocker currently. Plus the Zephyr allocator does not have this issue ?

lyakh, 2020-11-18:
> @lyakh we may not want to spend too much time on this since it does not seem to be a blocker currently. Plus the Zephyr allocator does not have this issue ?

@lgirdwood I'm fine either way: it's obviously not top priority, so I'm not wasting too much time on it (it's enough that we're wasting memory already :-) ). If we are sure we're dropping this allocator anyway soon, maybe we don't need to fix this. Also *strictly speaking* we'd need some hard evidence in what scenarios and how much RAM is wasted. If indeed my observation is correct and a vast majority of allocations happen in chunks of 0x180 or 0x300 bytes, then maybe returning to 0x180 for legacy platforms and using 0x80 on the rest would be the best solution.