thesofproject/sof #3632: [FEATURE]OpenAMP

hongshui3000, 2020-11-22:
Is there any plan to support OpenAMP in the sof project?

lgirdwood, 2020-11-23:
@hongshui3000 SOF now supports Zephyr as it's RTOS and Zephyr supports OpenAMP. What are the use cases or features you have in mind here ?

hongshui3000, 2020-11-24:
I want to use openamp for communication between NXP RT600 core0 (M33) and core1 (dsp hifi4), M33 uses zephyr, HIFI4 uses zephyr and sof

lgirdwood, 2020-11-24:
@hongshui3000 ok, this should be doable today but only with some Intel DSP platforms (that support Zephyr). Zephyr support for NXP platforms is planned. @dbaluta can provide more context on the NXP platforms.

dbaluta, 2020-11-25:
@hongshui3000 are you doing this as a fun project or planning to use SOF in a real product?

Our plan is to add support for Zephyr with SOF in Q2/Q3 next year for i.MX8 platforms. Also,  we plan to add support for i.MX RT as our bandwidth allows in the next summer.

lenghonglin, 2021-09-28:
@dbaluta @lgirdwood @hongshui3000  i.MX RT support sof in hifi4 now？

dbaluta, 2021-09-28:
@lenghonglin no support for i.MX RT for now as we are different teams at NXP. Anyhow, this is in my plan for the future with Zephyr getting support for i.MX8 boards.

i.MX RT is a beast of different nature. Having only hifi4 and some M4 cores. This means that mostly we need to make SOF run in standalone mode.

lenghonglin, 2021-10-08:
> @lenghonglin no support for i.MX RT for now as we are different teams at NXP. We managed to add support for SOF with Zephyr for i.MX8 and i.MX8X.
> 
> You can try that first as they have M4 cores and hifi4. Support for i.MX RT with zephyr is a good to have feature but this is a different team in NXP and I'm not very familiar with RT family.

Fine, i want to use sof with M4 cores and hifi4 soc,  but with the offical docs, the host is linux , not rtos .  how can i port with rtos host ?

dbaluta, 2021-10-08:
Yes, NXP BSP uses Linux not an RTOS for the host. Please ask for support here: https://community.nxp.com/

lenghonglin, 2021-11-12:
> Yes, NXP BSP uses Linux not an RTOS for the host. Please ask for support here: https://community.nxp.com/

How is it going now? I would like to know the progress for IMX RT for sof.  Could you ask for the situation to i.MX RT team, thanks very much!

dbaluta, 2021-11-12:
@lenghonglin there is no plan to support OpenAMP with SOF. Anyhow, patches are welcome.

lenghonglin, 2021-11-12:
> @lenghonglin there is no plan to support OpenAMP with SOF. Anyhow, patches are welcome.

I wonder know the plan about imx rt600 support for sof project ,  the OpenAMP is not important for me ,

dbaluta, 2021-11-12:
As a personal project i have sof on rt600 on my todo list. But I don't have any experience with RT series and I would appreciate that I wont have time for this in the near future.

@lenghonglin what is your usecase for this? Are you an NXP customer?

lenghonglin, 2021-11-12:
[image]

yes! and  i think if imx rt support  sof project , the audio application will easy to develope~