thesofproject/sof #3633: [BUG] [TGL] HDA controller reset fails in reboot tests

sathya-nujella, 2020-11-22:
**[Bug cloned from]**
https://github.com/thesofproject/linux/issues/2571

DUT: TGL Chrome DUT with Max98373_RT5682 **I2S**
BIOS: Coreboot

**Issue:**
Sound card creation failed.

TEST SCENARIO:
Random scenarios.
one such scenario is:
Boot DUT -> mount file system -> reboot few times to see error.

****More details:****
**ISSUE REPRODUCED WITH:** 
SOF branch: tgl-011-drop-stable   top commit:  27aab4f83c187701bc17b037b39e9453ea0731d0

**After reverting the below patch, issue is not seen in 100 reboot cycles (need to continue further tests):**
tgl: enable WOVCRO clock
Reverted the patch:  https://github.com/thesofproject/sof/commit/75561a99642c50265ffb9f94a760e52e6a2e292b

kv2019i, 2020-11-22:
Hit on Ubuntu Linux device as well. FW 010-hotfix.

@abonislawski Would it be possible to change to a higher clock whenever DSP is stopped? It would seem that if clock is left at WOVCRO and system is reset, HDA controller will not come out of reset anymore (with high repro rate).

keqiaozhang, 2020-11-23:
we also observed this issue when testing sof-011-drop-stable branch,  I tried to test more on TGL-RVP-HDA this morning and found that the reproduction rate is about about 30%。

mwasko, 2020-11-23:
Is the FW even loaded when this reboot issue occurs?

mengdonglin, 2020-11-23:
Thank you @kv2019i and @keqiaozhang 

@abonislawski @mwasko Please note this issue can also be reproduced on branch tgl-010-drop-stable  and tgl-010-hot-fix-1, it's not a branch specific issue for tgl-011-drop-stable

lgirdwood, 2020-11-23:
@sathya-nujella @kv2019i I assume FW with reverted https://github.com/thesofproject/sof/commit/75561a99642c50265ffb9f94a760e52e6a2e292b has booted at least once (without a power OFF - due to the reproduction rates)  during your testing.
@mwasko it does seem that FW is booting, but I wonder if weed to force the clock in our FW D3 code to be sure (since it looks like the clock setting may survive a DSP D3).

abonislawski, 2020-11-23:
@kv2019i @lgirdwood do you think disabling wovcro in ipc_pm_context_save() (pm_prepare_D3) would be a good idea to test?

lgirdwood, 2020-11-23:
> @kv2019i @lgirdwood do you think disabling wovcro in ipc_pm_context_save() (pm_prepare_D3) would be a good idea to test?
> EDIT: but we already did on wakeup

@abonislawski it's probably worth trying (and remember any delay).

kv2019i, 2020-11-23:
@lgirdwood wrote:
> @sathya-nujella @kv2019i I assume FW with reverted [75561a9](https://github.com/thesofproject/sof/commit/75561a99642c50265ffb9f94a760e52e6a2e292b) has booted at least once (without a power OFF - due to the reproduction rates) during your testing.

Correct. DSP is loaded at least once, and then a sw-reset (system reboot) is performed.

mengdonglin, 2020-11-24:
Link the original PR to enable WOVCRO clock of @abonislawski:  https://github.com/thesofproject/sof/pull/3418

keyonjie, 2020-11-24:
> This issue cannot be reproduced with cold reboot(power-cycle), only happens with warm reboot.

Thanks, this is useful information. @kv2019i you are right, it is reproduced with the sw-reset only.

mengdonglin, 2020-11-24:
PR #3641 is trying to fix this blocker issue before an official BIOS fix for WOVCRO clock is available. Will perform stress test on this PR #3641

mengdonglin, 2020-11-25:
@keyonjie also submitted an extra kernel fix https://github.com/thesofproject/linux/pull/2581, which sets DSP back to D0 before reboot if it is at D0i3, to fix the boot fail issue at the consequent rebooting.

abonislawski, 2020-11-25:
@keyonjie @mengdonglin is PR #3641 still needed? Looks like this kernel fix will prevent from boot fail (forced hpro before reboot) and we can fully use wovcro as previously, is that right?

keyonjie, 2020-11-26:
> @keyonjie @mengdonglin is PR #3641 still needed? Looks like this kernel fix will prevent from boot fail (forced hpro before reboot) and we can fully use wovcro as previously, is that right?

@abonislawski as the IP owner mentioned, it is recommend to use WOVCRO for WoV only, or even for WoV in S0Ix only, so I would suggest to keep PR #3641 also.

slawblauciak, 2021-02-26:
@sathya-nujella is the issue still reproducible on the latest tgl-011-drop-stable branch?

sathya-nujella, 2021-02-26:
> @sathya-nujella is the issue still reproducible on the latest tgl-011-drop-stable branch?

Hi @slawblauciak ,

we have taken this PR : inux/pull/2581 to chrome kernel. With this we don't have issue [Issue is not reported in past 3~4 BKC cycles].

If kernel does not have the change, even with latest tgl-011 branch: issue is reproducible if reboot happens with wov stream open.

keyonjie, 2021-03-01:
yes, @slawblauciak @sathya-nujella the fixes are already merged to mainline kernel, let's close the issue. @sathya-nujella please reopen it if needed.

closed 2021-03-01 by keyonjie